lvgl/lvgl #872: Cycling focus takes hidden objects into account

manison, 2019-02-20:
I have a visualization with several objects, some of them hidden depending on the state. All objects are added to a group to enable keyboard navigation between them. Now it turns out that when cycling through the objects using `LV_GROUP_KEY_NEXT`/`PREV` all objects are taken into account, even the hidden ones. Is this intentional behavior or is it a bug? IMHO it does not make sense to focus a hidden object.

kisvegabor, 2019-02-20:
I wouldn't say it's a bug but a not intentional feature. :smile: 

It'd be possible to update `lv_group_focus_next/prev` to ignore hidden objects but it can cause an infinite loop when all objects are hidden.

I suggest adding a `focus_cb` to the group where you can check whether the object is hidden or not and jump to the next/prev object is required. You still need to know the last "direction" to decide `lv_group_focus_next` or `lv_group_focus_prev` should be called.

manison, 2019-02-21:
> I wouldn't say it's a bug but a not intentional feature. 😄

OK, let's call it that way 😄 But it's still unintentional as you say, so I would like to modify the behavior of `lv_group_focus_next`/`prev` if you don't mind. The infinite loop can be detected and iteration can be aborted in such case. The `focus_cb` is a mere workaround and there is a complication with missing direction information.

kisvegabor, 2019-02-21:
Okay, go ahead and send a PR please!

manison, 2019-02-22:
Submitted #879. Merged by @embeddedt .

closed 2019-02-22 by manison

kisvegabor, 2019-02-22:
Great! Thank you!