lvgl/lvgl #874: Developing GUI on STM32F746 MCU(not dev kit)

tomtristan, 2019-02-20:
Not particulary an issue, looking to develop a GUI for a student formula car on self-developed PCB based on the STM32F746 disco. kit.
So my problem is, that the pinout on my PCB's MCU(the same as the disco kit) is a little different from the discovery kit.
I have some experience in C programming, but I am not experienced for sure.
I've come to the conclusion that lvgl is the option with the smallest learning curve, compared to other GUI developing programs/libraries. It also seems to support external buttons and rotary encoders for required navigation.
I am looking for all sorts of tips/tutorials/advice on how to make lvgl compatible with my PCB and at the same time be able to receive (and send) data via CAN and sensors, save controller settings to flash etc.
So far i have worked through the lvgl documentation, analyzed demo codes/programs, and am hoping to start testing first GUI elements on the PCB next week.
Should I start editing(fix the pinout) and adding pieces of this [lvgl ported discovery kit](https://github.com/littlevgl/stm32f746_disco_no_os_sw4stm32) code or download the full lvgl repository and start porting it myself?

embeddedt, 2019-02-20:
@tomtristan It would depend on how similar your PCB is to the discovery kit. The F746 discovery code assumes that it's a framebuffer-based setup.

seyyah, 2019-02-21:
I prefer to start you may use CubeMX and Atollic. You can solve pinout problems with **alternate pin** [1,2].

For example,

| Label      | From | To   |
|------------|------|------|
| LTDC_B3    | PG11 | PJ15 |
| LTDC_B4    | PI4  | PG12 |
| LTDC_B5    | PI5  | PK4  |
| ... | | |

## :clipboard: Reference
1. https://community.st.com/s/question/0D50X00009XkX9sSAF/nucleo-f303re-alternate-function-mapping
2. https://electronics.stackexchange.com/questions/118102/gpio-control-of-alternate-function-pin

tomtristan, 2019-02-21:
@embeddedt The PCB has the same display(Newhaven 480x272 TFT-LCD, connected as ARGB888) as the discovery kit and has the same SDRAM. I left out all of the parts that are unnecessary for me - sdcard, uart, touchpanel plug etc.
So it should be pretty similar to the discovery kit, only difference being the pinout on the MCU.
If I were to go through the whole porting process myself, do you recon there would be many issues with the CubeMX generated HAL libraries and other drivers? Or is this process rather straight-forward and user feedback rather states that it's intuitive if one has at least some understanding of embedded systems programming and C language?  

@seyyah thank you for the tip! I wasnt aware of that, even though this isn't my first PCB that I've designed using CubeMX. I have designed, had it manufactured, and tested the first revision of the PCB, so perhaps this could come handy in designing the second revision.

embeddedt, 2019-02-21:
> If I were to go through the whole porting process myself, do you recon there would be many issues with the CubeMX generated HAL libraries and other drivers? Or is this process rather [...]

I don't think the drivers are generated, but rather the configuration code for them. The HAL drivers are relatively easy to use based on my experience, but I've never had to configure the pinouts, clocking, etc. from scratch (I usually borrow most of that code from a demo).

tomtristan, 2019-02-27:
Fortunately, using the project [stm32f746_disco_no_os_atollic](https://github.com/littlevgl/stm32f746_disco_no_os_atollic) made porting this library quite easy, since it is not directly connected to the mcu's pins, but rather uses ltdc initing pin configuration instead(which is not dependant on the pinout).

closed 2019-02-27 by tomtristan