lvgl/lvgl #877: Button toggle issue

jocelynmass, 2019-02-21:
Hi there,

I'm trying to get a hardware button working with LVGL but I'm facing a bug I can't understand for now.

Here is my code:

```c
static  lv_res_t btn_rel_action(lv_obj_t * btn);

static  lv_res_t btn_rel_action(lv_obj_t * btn)
{
   printf("btn_rel_action cb\n");
}

int32_t lvgl_init(void)
{
    static lv_obj_t *btn;
    static lv_point_t points_array[] = { { 220, 120 } };
    static lv_indev_t *indev;
    static lv_indev_drv_t indev_drv;
    static lv_disp_drv_t disp_drv;
    static lv_obj_t * label;

    lv_init();  
    lv_disp_drv_init(&disp_drv);
    disp_drv.disp_flush = ex_disp_flush;
    lv_disp_drv_register(&disp_drv);

    lv_indev_drv_init(&indev_drv);  
    indev_drv.type = LV_INDEV_TYPE_BUTTON;
    indev_drv.read = btn1_read;
    indev = lv_indev_drv_register(&indev_drv); 
    lv_indev_set_button_points(indev, points_array);

    btn = lv_btn_create(lv_scr_act(), NULL);  
    lv_obj_set_size(btn, 100, 50);
    lv_obj_set_pos(btn, 200, 100);
    lv_obj_set_free_num(btn, 1);
    lv_btn_set_action(btn, LV_BTN_ACTION_CLICK, btn_rel_action);

    label = lv_label_create(btn, NULL);
    lv_label_set_text(label, "Start");
    
    return 0;
}
```

With this code the button is correctly displayed and when I push the button I correctly enter the callback **btn_rel_action**. However the button is not toggled, instead, I'm seeing a sort of blue banner at the top of the screen changing of color (from blue to light blue), like the toggled button was not displayed at the right position. Anyone experienced this issue before?

[image]

Thanks for your help,
Jocelyn

kisvegabor, 2019-02-21:
Hi, 

How does `btn1_read` looks like?

seyyah, 2019-02-21:
Problem's source maybe btn1_read. See also "Hardware button usage" blog entry [1].

## :clipboard: Reference
1. https://blog.littlevgl.com/2019-01-08/hardware-button

chaos4284, 2019-02-21:
you add lv_btn_set_toggle(..,..).

jocelynmass, 2019-02-21:
Thanks for your replies. Here is btn1_read function:

```c
bool btn1_read(lv_indev_data_t*data)
{
    static uint32_t last_btn = 0;   /*Store the last pressed button*/ 
    int btn_pr = gpio.read(&btn1);     /*Get the ID (0,1,2...) of the pressed button*/

    if(btn_pr >= 0) {               /*Is there a button press?*/  
       last_btn = btn_pr;           /*Save the ID of the pressed button*/
       data->state = LV_INDEV_STATE_PR;  /*Set the pressed state*/
    } else {
       data->state = LV_INDEV_STATE_REL; /*Set the released state*/
    }

    data->btn = last_btn;            /*Set the last button*/         

    return false;                    /*No buffering so no more data read*/
}
```

kisvegabor, 2019-02-21:
`lv_btn_set_toggle()` is really missing as @chaos4284 stated.

@jocelynmass The GUI button is shown as pressed while you are pressing the button?

jocelynmass, 2019-02-21:
@kisvegabor no the button remains unpressed even if I pressed on the hard buttong.

Sorry I think I didn't use the right wording. What I meant is that the button is never displayed pressed even if I pressed the button. Please see the photos below, you can see the button is displayed and when I press the hard button a blue banner is toggling from blue to light blue. Any idea?

[image]
[image]

Thanks

embeddedt, 2019-02-21:
@jocelynmass Something's going wrong with your display driver. That banner shouldn't be there. Can you show your `disp_flush` function?

jocelynmass, 2019-02-21:
@embeddedt My PCB uses an SDRAM to store the frame buffer. The frame buffer address is also set in the LCD interface driver so its content is automatically refreshed on the screen without calling any function, this is why I'm not doing anything in ex_disp_flush function. The MCU I'm using is a STM32H753 and I'm setting the frame buffer address this way:

``` c
LTDC_LayerCfgTypeDef layer;
layer.FBStartAdress = lcdif->fb_addr; // Frame buffer located in the SDRAM
```
Please see below the disp_flush function:

``` c
/* Flush the content of the internal buffer the specific area on the display
 * You can use DMA or any hardware acceleration to do this operation in the background but
 * 'lv_flush_ready()' has to be called when finished
 * This function is required only when LV_VDB_SIZE != 0 in lv_conf.h*/
static void ex_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
    lv_flush_ready();
}
```

embeddedt, 2019-02-21:
@jocelynmass The `flush` function still needs to put data on the screen regardless of how your framebuffer works.

jocelynmass, 2019-02-21:
@embeddedt ok just trying to understand. Is LVGL writing directly to the framebuffer?

embeddedt, 2019-02-21:
@jocelynmass No. Everything has to be done by the `flush` and `map` functions.

jocelynmass, 2019-02-21:
@embeddedt ok thanks, I'm gonna change this. Just curious how I got the button displayed then if LVGL is not writing anything to the framebuffer?

embeddedt, 2019-02-21:
@jocelynmass It gets done by your `map` and `flush` functions. That is LittlevGL's only interface to your screen.

jocelynmass, 2019-02-21:
@embeddedt but I didn't implement the `map` function and the `flush` is only calling `lv_flush_ready()`. I'm just trying to understand how the button got displayed on my screen ;-)

embeddedt, 2019-02-21:
@jocelynmass That's impossible; you must be registering a different display driver.

kisvegabor, 2019-02-22:
@embeddedt 
If `LV_VB_SIZE > 0` you only need `disp_flush`. `disp_map` and `disp_fill` are used only in unbuffered mode (`LV_VDB_SIZE == 0`). (The very rarely used unbuffered will be removed in `v6.0` to make the interface clearer.)

@jocelynmass 
Did you set the `LV_VDB_ADR` to the frame buffer's address? If yes, it won't work. LittlevGL always writes to the beginning of the VDB buffer and passes this to `disp_flush` to copy it to the correct coordinates. That's why you see the "button is the first lines."

I suggest creating a few lines of VDB (`LV_VDB_SIZE (LV_HOR_RES * 30)` and leave `LV_VDB_ADR  0`) because:
- LittlevGL works this way :)
- It faster to draw into the internal RAM then external SRAM. This way you need to deal with the external RAM once/pixel when you copy the rendered image.
- This way you will have buffered drawing without flickering. In your case, you would see the process of drawing on the screen (background first + button body + button border + label)

jocelynmass, 2019-02-22:
Thanks @kisvegabor and @embeddedt this is perfectly clear now. I'll implement the disp_flush function and I'll let you know.

jocelynmass, 2019-02-23:
Fixed! Again thanks your help

closed 2019-02-23 by jocelynmass