lvgl/lvgl #878: Related to mouse input coordinates.

chaos4284, 2019-02-21:
I am trying to represent a region of area2 that is large than area1 in fixed area area1.

That is, the horizontal and vertical areas exceeding area 1 are displayed with a scroll bar as follows.
[image]

embeddedt, 2019-02-21:
@chaos4284 Here's a working example with `dev-6.0`:

```c
        lv_theme_set_current(lv_theme_night_init(210, NULL));
	lv_obj_t *area1 = lv_page_create(lv_scr_act(), NULL);
	lv_cont_set_fit(lv_page_get_scrl(area1), true, true);
	lv_page_set_sb_mode(area1, LV_SB_MODE_ON);
	lv_obj_set_size(area1, LV_HOR_RES, LV_VER_RES);
	lv_obj_t *area2 = lv_obj_create(area1, NULL);
	lv_obj_set_size(area2, 800, 400);
	lv_obj_set_click(area2, false);
```

kisvegabor, 2019-02-22:
The returned point is absolute i.e. not relative to the clicked object. If you click the same point of the screen you will get the same point regardless to the objects coordinates. 

You can calculate the relative coordinates like this:
```c
lv_coord_t x = pos.x - obj_property->coords.x1;
lv_coord_t y = pos.y - obj_property->coords.y1;
```

chaos4284, 2019-02-23:
code is well operate. Thanks.

closed 2019-02-23 by chaos4284