lvgl/lvgl #880: cant scroll on window  with checkboxes

pescatore89, 2019-02-21:
Hi there
Iam using an small OLED and trying to create a window, which has several checkboxes.
The point is, if I want to create more then 5 checkboxes, it would become necessary to make the window scrollable. The desired functionality should be like:
- when I step down from one cb to the next, the window should automatically scroll.
Having the following code, the window does not scroll :(

I really would be thankful for some help.

Sorry for the bad quality of the image, but I hope its clear what I mean :)
[image]

`/*Create a scroll bar style*/
	static lv_style_t style_sb;
	lv_style_copy(&style_sb, &lv_style_plain);
	style_sb.body.main_color = LV_COLOR_BLACK;
	style_sb.body.grad_color = LV_COLOR_BLACK;
	style_sb.body.border.color = LV_COLOR_WHITE;
	style_sb.body.border.width = 1;
	style_sb.body.border.opa = LV_OPA_70;
	style_sb.body.radius = LV_RADIUS_CIRCLE;
	style_sb.body.opa = LV_OPA_60;

	/*Create a window*/

	win = lv_win_create(lv_scr_act(), NULL);		/* object for window */
	lv_win_set_title(win, "Pollenarten");		// set title
	lv_win_set_style(win, LV_WIN_STYLE_SB, &style_sb);              /*Set the scroll bar style*/

	/* Create close Window Button*/
	lv_obj_t *closeBtn;
	closeBtn = lv_win_add_btn(win, SYMBOL_CLOSE, win_close_action);
	GUI_GroupPush();
	GUI_AddObjToGroup(closeBtn);		// add close button
	lv_group_focus_obj(closeBtn);

	 /*Create check boxes*/
	int x = 0;
	int nOFnames = getQuantity();	//  get number of pollen to initialize
	char ** name = getNamelist();	//  get names of the pollen
	uint16_t posX = 0;
	uint16_t posY = 0;
	lv_obj_t * cb;
	uint8_t counter = 0;

	while (x < nOFnames) {			// Initializes the desired pollen

		cb = lv_cb_create(win, NULL);
		lv_obj_set_pos(cb, posX, posY);
		GUI_AddObjToGroup(cb);
		lv_cb_set_text(cb, (name[x]));
		lv_cb_set_action(cb, cb_release_action);		//define what happen when checkbox pushed/released
		updateCheckbox(cb,counter);
		counter++;
		x++;
		posY = posY + 20;

	}

	/* Create button to step to next page when checkboxes clicked (playlist created) */
	lv_obj_t *label;
	lv_obj_t *btn_play;

	btn_play = lv_btn_create(win, NULL);
	lv_obj_set_pos(btn_play, 90, 80);
	lv_btn_set_action(btn_play, LV_BTN_ACTION_CLICK, btn_play_click_action);
	label = lv_label_create(btn_play, NULL);
	lv_label_set_text(label, SYMBOL_PLAY);
	GUI_AddObjToGroup(btn_play);`

embeddedt, 2019-02-21:
@pescatore89 You're correct, it appears that the window doesn't scroll.

@kisvegabor Here's a version of the code that can be pasted into `pc_simulator`:

```c
lv_group_t *group;
lv_obj_t *win;
static void test_group(void)
{
		/*Create a scroll bar style*/
	static lv_style_t style_sb;
	lv_style_copy(&style_sb, &lv_style_plain);
	style_sb.body.main_color = LV_COLOR_BLACK;
	style_sb.body.grad_color = LV_COLOR_BLACK;
	style_sb.body.border.color = LV_COLOR_WHITE;
	style_sb.body.border.width = 1;
	style_sb.body.border.opa = LV_OPA_70;
	style_sb.body.radius = LV_RADIUS_CIRCLE;
	style_sb.body.opa = LV_OPA_60;
	group = lv_group_create();
#if LV_EX_KEYBOARD
    lv_indev_drv_t rael_kb_drv;
    rael_kb_drv.type = LV_INDEV_TYPE_KEYPAD;
    rael_kb_drv.read = keyboard_read;
    lv_indev_t * real_kb_indev = lv_indev_drv_register(&rael_kb_drv);
    lv_indev_set_group(real_kb_indev, group);
#endif

#if LV_EX_MOUSEWHEEL
    lv_indev_drv_t enc_drv;
    enc_drv.type = LV_INDEV_TYPE_ENCODER;
    enc_drv.read = mousewheel_read;
    lv_indev_t * enc_indev = lv_indev_drv_register(&enc_drv);
    lv_indev_set_group(enc_indev, group);
#endif
	/*Create a window*/

	win = lv_win_create(lv_scr_act(), NULL);		/* object for window */
	lv_win_set_title(win, "Pollenarten");		// set title
	lv_win_set_style(win, LV_WIN_STYLE_SB, &style_sb);              /*Set the scroll bar style*/

	/* Create close Window Button*/
	lv_obj_t *closeBtn;
	closeBtn = lv_win_add_btn(win, SYMBOL_CLOSE, NULL);

	lv_group_add_obj(group, closeBtn);		// add close button
	lv_group_focus_obj(closeBtn);

	 /*Create check boxes*/
	int x = 0;
	int nOFnames = 20;	//  get number of pollen to initialize
	uint16_t posX = 0;
	uint16_t posY = 0;
	lv_obj_t * cb;
	uint8_t counter = 0;

	while (x < nOFnames) {			// Initializes the desired pollen

		cb = lv_cb_create(win, NULL);
		lv_obj_set_pos(cb, posX, posY);
		lv_group_add_obj(group, cb);
		lv_cb_set_text(cb, "Name");
		lv_cb_set_action(cb, NULL);		//define what happen when checkbox pushed/released
		//updateCheckbox(cb,counter);
		counter++;
		x++;
		posY = posY + 20;

	}

	/* Create button to step to next page when checkboxes clicked (playlist created) */
	lv_obj_t *label;
	lv_obj_t *btn_play;

	btn_play = lv_btn_create(win, NULL);
	lv_obj_set_pos(btn_play, 90, 80);
	lv_btn_set_action(btn_play, LV_BTN_ACTION_CLICK, NULL);
	label = lv_label_create(btn_play, NULL);
	lv_label_set_text(label, SYMBOL_PLAY);
	lv_group_add_obj(group, btn_play);
}
```

BrianPugh, 2019-02-21:
I'm having a similar bug where my `lv_list` no longer scrolls. I'll try and do a bit of debugging later today, but my schedule has been hectic.

pescatore89, 2019-02-22:
hmm, you are right, due to the description it should make my checkboxes scrollable. Unfortunately it doesnt :-(

pescatore89, 2019-02-22:
Its probably because I got a window and not a page

kisvegabor, 2019-02-22:
Have a look at how it is solved here: https://github.com/littlevgl/lv_examples/blob/master/lv_tests/lv_test_group/lv_test_group.c#L92

You need to register a `focus_cb` to scroll the window when the focused object changes.

```c
static void group_focus_cb(lv_group_t * group)
{
    lv_obj_t * f = lv_group_get_focused(group);
   
    /*Get the wondows content object */
    lv_obj_t * par =  lv_obj_get_parent(f);  /*The content object is page so first get scrollable object*/
    if(par) par = lv_obj_get_parent(par);    /*Then get the page itself*/

    /*If the focused object is on the window then scrol lteh window to make it visible*/
    if(par == lv_win_get_content(win)) {
        lv_win_focus(win, f, 200);   
    }
}
...

    group = lv_group_create();
    lv_group_set_focus_cb(group, group_focus_cb);
```

[image]

pescatore89, 2019-02-23:
Thanks, I had to modify some stuff since I got multiple Layers but your Input helped a lot :)

kisvegabor, 2019-02-23:
Great! :)

closed 2019-02-23 by kisvegabor