lvgl/lvgl #882: Creat obj solid

autoprog, 2019-02-21:
Hi,
how to creat simple obj solid with border, and change color in real time.

[image]

thanks,
Carlos.

seyyah, 2019-02-21:
You may use `lv_canvas` [1,2].

Example: `lv_test_canvas_2()`

[image]

## :clipboard: Reference
1. https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_canvas.c
2. https://github.com/littlevgl/lv_examples/blob/master/lv_tests/lv_test_objx/lv_test_canvas/lv_test_canvas.c

embeddedt, 2019-02-21:
@autoprog Another option is to keep adjusting the style of the object and then set it again using `lv_obj_set_style`.

autoprog, 2019-02-28:
Hi, 
testd lv_creat obj + lv_set_style = ok

thanks,
Carlos.

embeddedt, 2019-02-28:
@autoprog If the issue is solved please close it.

closed 2019-02-28 by autoprog