lvgl/lvgl #885: Issues with lv_led alignment and centering

kaszeta, 2019-02-21:
I'm trying to do a simple front panel with a series of LEDs, each with a label above and below the LED (the top one identifying it, and the bottom one being an explanatory state), and while the LED appears to line up with the top label, I can't get the bottom label to line up with the LED.

Here's an example:

~~~~
lv_obj_t * sensor2_label = lv_label_create(sensor_container, NULL);
lv_obj_align(sensor2_label, sensor1_label, LV_ALIGN_OUT_RIGHT_MID, 15, 00);
lv_label_set_text(sensor2_label, "Sensor 2");
lv_label_set_align(sensor2_label, LV_LABEL_ALIGN_LEFT);
lv_label_set_style(sensor2_label, &style_speedbox_label);

lv_obj_t * sensor2_led = lv_led_create(sensor_container, NULL);
lv_obj_set_size(sensor2_led, led_size, led_size);
lv_obj_align(sensor2_led, sensor2_label, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
lv_obj_set_style(sensor2_led, &style_led);
lv_led_on(sensor2_led);
		
lv_obj_t * sensor2_status = lv_label_create(sensor_container, NULL);
lv_label_set_align(sensor2_status, LV_LABEL_ALIGN_CENTER);
lv_obj_align(sensor2_status, sensor2_led, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
lv_label_set_text(sensor2_status, "OK");
lv_label_set_style(sensor2_status, &style_speedbox_label);
~~~~

sensor2_led is centered below sensor2_label, but sensor2_status aligns well left of center.

embeddedt, 2019-02-22:
@kaszeta Can you send a complete code snippet that can be reproduced independently on the PC simulator? The styles and `sensor1_label` are missing.

kaszeta, 2019-02-25:
Let me see if I can make a minimal version of it.   

In the meantime, 

```
static lv_style_t style_speedbox_label;
lv_style_copy(&style_speedbox_label, &lv_style_pretty);
style_speedbox_label.text.font = &lv_font_dejavu_20;

static lv_style_t style_led;
lv_style_copy(&style_led, &lv_style_pretty_color);
style_led.body.radius = 12;
style_led.body.main_color = LV_COLOR_MAKE(0xff, 0x3f, 0x3f);
style_led.body.grad_color = LV_COLOR_MAKE(0xa0, 0x50, 0x50);
style_led.body.border.color = LV_COLOR_MAKE(0xfa, 0x0f, 0x00);
style_led.body.border.width = 2;

lv_obj_t * sensor1_label = lv_label_create(sensor_container, NULL);
lv_obj_align(sensor1_label, sensor_label, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 10);
lv_label_set_text(sensor1_label, "Sensor 1");
lv_label_set_align(sensor1_label, LV_LABEL_ALIGN_CENTER);
lv_label_set_style(sensor1_label, &style_speedbox_label);

```

kaszeta, 2019-02-25:
Okay, here's a minimal case that I've run in the pc simulator:

```
/*Create a style for the LEDs*/
        int led_size = 24;
	static lv_style_t style_led;
	lv_style_copy(&style_led, &lv_style_pretty_color);
	style_led.body.radius = 12;
	style_led.body.main_color = LV_COLOR_MAKE(0xff, 0x3f, 0x3f);
	style_led.body.grad_color = LV_COLOR_MAKE(0xa0, 0x50, 0x50);
	style_led.body.border.color = LV_COLOR_MAKE(0xfa, 0x0f, 0x00);
	style_led.body.border.width = 2;
	
	lv_obj_t * sensor1_label = lv_label_create(lv_scr_act(), NULL);
	lv_obj_align(sensor1_label, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 10);
	lv_label_set_text(sensor1_label, "Sensor 1");
        lv_label_set_align(sensor1_label, LV_LABEL_ALIGN_CENTER);
        lv_label_set_style(sensor1_label, &style_speedbox_label);

	lv_obj_t * sensor1_led = lv_led_create(lv_scr_act(), NULL);
	lv_obj_set_size(sensor1_led, led_size, led_size);
        lv_obj_align(sensor1_led, sensor1_label, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_obj_set_style(sensor1_led, &style_led);
	lv_led_on(sensor1_led);
	
	lv_obj_t * sensor1_status = lv_label_create(lv_scr_act(), NULL);
	lv_obj_align(sensor1_status, sensor1_led, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_label_set_text(sensor1_status, "OK");
        lv_label_set_align(sensor1_status, LV_LABEL_ALIGN_CENTER);
        lv_label_set_style(sensor1_status, &style_speedbox_label);
		
        lv_obj_t * sensor2_label = lv_label_create(lv_scr_act(), NULL);
	lv_obj_align(sensor2_label, sensor1_label, LV_ALIGN_OUT_RIGHT_MID, 15, 00);
	lv_label_set_text(sensor2_label, "Sensor 2");
        lv_label_set_align(sensor2_label, LV_LABEL_ALIGN_LEFT);
        lv_label_set_style(sensor2_label, &style_speedbox_label);

	lv_obj_t * sensor2_led = lv_led_create(lv_scr_act(), NULL);
	lv_obj_set_size(sensor2_led, led_size, led_size);
        lv_obj_align(sensor2_led, sensor2_label, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_obj_set_style(sensor2_led, &style_led);
	lv_led_on(sensor2_led);
		
	lv_obj_t * sensor2_status = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_align(sensor2_status, LV_LABEL_ALIGN_CENTER);
        lv_obj_align(sensor2_status, sensor2_led, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_label_set_text(sensor2_status, "OK");
        lv_label_set_style(sensor2_status, &style_speedbox_label);

	lv_obj_t * sensor3_label = lv_label_create(lv_scr_act(), NULL);
	lv_obj_align(sensor3_label, sensor2_label, LV_ALIGN_OUT_RIGHT_MID, 15, 00);
	lv_label_set_text(sensor3_label, "Sensor 3");
        lv_label_set_align(sensor3_label, LV_LABEL_ALIGN_LEFT);
        lv_label_set_style(sensor3_label, &style_speedbox_label);

	lv_obj_t * sensor3_led = lv_led_create(lv_scr_act(), NULL);
	lv_obj_set_size(sensor3_led, led_size, led_size);
        lv_obj_align(sensor3_led, sensor3_label, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_obj_set_style(sensor3_led, &style_led);
	lv_led_on(sensor3_led);
		
	lv_obj_t * sensor3_status = lv_label_create(lv_scr_act(), NULL);
	lv_obj_align(sensor3_status, sensor3_led, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_label_set_text(sensor3_status, "OK");
        lv_label_set_align(sensor3_status, LV_LABEL_ALIGN_CENTER);
        lv_label_set_style(sensor3_status, &style_speedbox_label);
```
Goal is to have have essentially a 3x3 array, top row is labels (static, unchanging), middle row is LEDs, bottom row is dynamic labels.  Ideally, all centered.

Here's how it renders:"
[image]

kisvegabor, 2019-03-03:
Just change the order:
```c
lv_label_set_text(sensor1_status, "OK");
lv_obj_align(sensor1_status, sensor1_led, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
```
If you align first then the default text (`"Text"`) will be aligned. As `"Ok"` is shorter and you don't realing the label, the position will be wrong.

---

In the past few days [LittlevGL's Forum](https://forum.littlevgl.com) has launched to discuss questions there. See this [Announcment](https://forum.littlevgl.com/viewtopic.php?f=3&t=5).

If the issue still persists please open a topic for it on the Forum.

closed 2019-03-03 by kisvegabor