lvgl/lvgl #886: Can you provide an example of `log. h`?

3038922, 2019-02-21:
I looked for the document. You did not see how to use the built-in `lv_log_xx`.

embeddedt, 2019-02-21:
@3038922 `lv_conf_templ.h` gives enough information to get started:

https://github.com/littlevgl/lvgl/blob/374dfcd60a12ef0127a25a0a8bae4a5fbacf8bd4/lv_conf_templ.h#L151-L164

3038922, 2019-02-21:
> @3038922 `lv_conf_templ.h` gives enough information to get started:
> 
> [lvgl/lv_conf_templ.h](https://github.com/littlevgl/lvgl/blob/374dfcd60a12ef0127a25a0a8bae4a5fbacf8bd4/lv_conf_templ.h#L151-L164)
> 
> Lines 151 to 164 in [374dfcd](/littlevgl/lvgl/commit/374dfcd60a12ef0127a25a0a8bae4a5fbacf8bd4)
> 
>  /*Log settings*/ 
>  #define USE_LV_LOG      1   /*Enable/disable the log module*/ 
>  #if USE_LV_LOG 
>  /* How important log should be added: 
>   * LV_LOG_LEVEL_TRACE       A lot of logs to give detailed information 
>   * LV_LOG_LEVEL_INFO        Log important events 
>   * LV_LOG_LEVEL_WARN        Log if something unwanted happened but didn't caused problem 
>   * LV_LOG_LEVEL_ERROR       Only critical issue, when the system may fail 
>   */ 
>  #  define LV_LOG_LEVEL    LV_LOG_LEVEL_WARN 
>  /* 1: Print the log with 'printf'; 0: user need to register a callback*/ 
>   
>  #  define LV_LOG_PRINTF   0 
>  #endif  /*USE_LV_LOG*/

Is there any specific writing?
···
LV_LOG_TRACE（“helloworld”）; //Would that be all right? Is the path of output file not initialized and not configured?
···

embeddedt, 2019-02-21:
@3038922 If you enable `LV_LOG_PRINTF`, it calls `printf` behind the scenes and you can just do that. Otherwise you have to register a callback with `lv_log_register_print`.

3038922, 2019-02-21:
> @3038922 If you enable `LV_LOG_PRINTF`, it calls `printf` behind the scenes and you can just do that. Otherwise you have to register a callback with `lv_log_register_print`.
What if it's std:: court?

embeddedt, 2019-02-21:
@3038922 Then register a callback that prints the values to `std::cout`. The documentation of `lv_log_register_print` should be self-explanatory:

https://github.com/littlevgl/lvgl/blob/374dfcd60a12ef0127a25a0a8bae4a5fbacf8bd4/lv_misc/lv_log.c#L40-L48

> I set up a `label` to read `log.txt` records. Does he automatically distinguish by color?

If you mean the recoloring system, you have to enable it with `lv_label_set_recolor(label, true)`.

3038922, 2019-02-21:
> @3038922 Then register a callback that prints the values to `std::cout`. The documentation of `lv_log_register_print` should be self-explanatory:
> 
> [lvgl/lv_misc/lv_log.c](https://github.com/littlevgl/lvgl/blob/374dfcd60a12ef0127a25a0a8bae4a5fbacf8bd4/lv_misc/lv_log.c#L40-L48)
> 
> Lines 40 to 48 in [374dfcd](/littlevgl/lvgl/commit/374dfcd60a12ef0127a25a0a8bae4a5fbacf8bd4)
> 
>  /** 
>   * Register custom print (or anything else) function to call when log is added 
>   * @param f a function pointer: 
>   *          `void my_print (lv_log_level_t level, const char * file, uint32_t line, const char * dsc)` 
>   */ 
>  void lv_log_register_print(void f(lv_log_level_t, const char *, uint32_t,  const char *)) 
>  { 
>      print_cb = f; 
>  } 
> > I set up a `label` to read `log.txt` records. Does he automatically distinguish by color?
> 
> If you mean the recoloring system, you have to enable it with `lv_label_set_recolor(label, true)`.

I'll do the experiment first. Thank you.

closed 2019-02-22 by 3038922