lvgl/lvgl #890: C99 issues

embeddedt, 2019-02-22:
@kisvegabor There is a portability issue I'm seeing while testing how LittlevGL compiles with very strict GCC warnings. ISO C99 doesn't support unnamed structs/unions.

Example:

https://github.com/littlevgl/lvgl/blob/a374baa37602288081a5ad770a875089a95acfbf/lv_hal/lv_hal_indev.h#L73-L91

The `union` itself is unnamed and contains two unnamed structs.

C47D, 2019-02-22:
Anonymous enums or structs are available since C11 [1].

[1] https://www.geeksforgeeks.org/g-fact-38-anonymous-union-and-structure/

embeddedt, 2019-02-23:
@C47D That's true, but we're supposed to have C99 compatibility.

kisvegabor, 2019-02-24:
Thank you for pointing this out. 

These structs will be updated by the muli-display update. So let's postpone it until it's merged. (probably next week)

Have you found other unnamed structs or enums too?

embeddedt, 2019-02-26:
@kisvegabor I think there may have been others, but this one was generating a lot of the warnings, since it's included by most of the files. I will report other ones if I see them.

kisvegabor, 2019-02-26:
I named these struct and unions in `dev-6.0`.

embeddedt, 2019-02-26:
@kisvegabor Great, I'll try it again as soon as possible.

embeddedt, 2019-02-27:
@kisvegabor These unions are still unnamed, as well as the ones in the `lv_colorX_t` types in `lv_misc/lv_color.h`:

https://github.com/littlevgl/lvgl/blob/551eadc1d5fc6c20bab9c88ff7dcd1d5f0ae37e4/lv_hal/lv_hal_indev.h#L59-L69

https://github.com/littlevgl/lvgl/blob/551eadc1d5fc6c20bab9c88ff7dcd1d5f0ae37e4/lv_hal/lv_hal_indev.h#L125-L136

Additionally, a new problem was introduced with multi-display support.
```
./lvgl/lv_conf_checker.h:198:16: warning: redefinition of typedef 'lv_disp_drv_user_data_t' [-Wpedantic]
```
Redefining types is not allowed in C, nor is there a facility to check whether the type is already defined.

kisvegabor, 2019-02-28:
Thank you!

I've updated the colors. I'll do the rest soon.

kisvegabor, 2019-03-02:
@embeddedt 
Do you have an idea about what to do with the `typedefs`? It can't be checked by `lv_conf_checker.h` if a `typedef` is exists or not. 

I can't imagine a better solution then simply not checking `typedef`s in `lv_conf_checker.h`. What do you think?

embeddedt, 2019-03-02:
@kisvegabor I agree. Let's remove the typedefs from `lv_conf_checker.h`.

kisvegabor, 2019-03-04:
Ok, I'll do this way.
I fixed these issues in the `event` branch which will be merged soon into `dev-6.0`

beibean, 2019-03-05:
I am having problems when using next functions:

```C
uint8_t (*lv_txt_encoded_size)(const char *) =                  lv_txt_utf8_size;
uint32_t (*lv_txt_unicode_to_encoded)(uint32_t) =               lv_txt_unicode_to_utf8;
uint32_t (*lv_txt_encoded_conv_wc)(uint32_t) =                  lv_txt_utf8_conv_wc;
uint32_t (*lv_txt_encoded_next)(const char *, uint32_t *) =    lv_txt_utf8_next;
uint32_t (*lv_txt_encoded_prev)(const char *, uint32_t *) =   lv_txt_utf8_prev;
uint32_t (*lv_txt_encoded_get_byte_id)(const char *, uint32_t) = lv_txt_utf8_get_byte_id;
uint32_t (*lv_encoded_get_char_id)(const char *, uint32_t) =   lv_txt_utf8_get_char_id;
uint32_t (*lv_txt_get_encoded_length)(const char *) =          lv_txt_utf8_get_length;
```

The question here is, Is this type of initialization legal in C?

What I thought is that a global variable could only be initialized to a constant or a literal since it is done in compiling time.

When assigning a function pointer, like it's the case, its address might not be allocated yet.

embeddedt, 2019-03-05:
@beibean What is the exact error you are getting?

embeddedt, 2019-03-05:
@beibean In that case it sounds like an issue with your device specifically. I think that there is no undefined behavior used here.

beibean, 2019-03-05:
@embeddedt Ok, thanks! I will use proper initialization functions myself.

beibean, 2019-03-06:
@embeddedt What about this define within lv_conf?

```C
#define LV_COMPILER_NON_CONST_INIT_SUPPORTED 1  /* 1: Initialization with non constant values are supported */
```

embeddedt, 2019-03-06:
@beibean You can try configuring that according to how your compiler setup works.

kisvegabor, 2019-03-06:
What is the difference between:
```c
/*Working*/
static void (*px_fp)(lv_coord_t x, lv_coord_t y, const lv_area_t * mask, lv_color_t color, lv_opa_t opa) = lv_vpx;
```

and 

```c
/*Not working*/
uint8_t (*lv_txt_encoded_size)(const char *) =     lv_txt_utf8_size;
```

It seems the same to me: assign a function pointer to a global function pointer variable.

beibean, 2019-03-07:
There is a big difference:

Static variable is initialized when declared while extern variable is not.

Just for you to know, this variable within `lv_draw_aa_get_opa` did not initialize properly neither:

```C
    static const lv_opa_t * seg_map[] = {seg1, seg2, seg3, seg4,
                                         seg5, seg6, seg7, seg8
                                        };
```

kisvegabor, 2019-03-08:
> Static variable is initialized when declared while extern variable is not.

AFAIK all `static` and `global` variables are initialized in the `startup` code.

So these are equivalent for the compiler:
```c
uint8_t global_x = 1;
static uint8_t static global_x = 1;

int main(void) {
  static uint8_t static_local = 1;

}

All these variables are mapped to specific RAM addresses and initialized before `main` is called

```

Anyway, I believe it works correctly because in the past few years nobody reported any issue about it.

beibean, 2019-03-08:
> AFAIK all static and global variables are initialized in the startup code.

That is what I thought, but somehow my compiler treats them in a different manner.

> Anyway, I believe it works correctly because in the past few years nobody reported any issue about it.

No bother, I have already created my own init functions.

I do not have much time these days, but I will investigate more deeply this issue and let you know what is happening with my compiler, just for the case someone could have the same matter.

kisvegabor, 2019-03-08:
@beibean I'm also curious what can cause this issue.

@embeddedt I merged a lot of things to `dev-6.0` today. It also fixes the issues mentioned above. Can you check what is still missing?

embeddedt, 2019-03-15:
@kisvegabor There were a few remaining problems; I fixed them in #948.

kisvegabor, 2019-03-17:
Merged, thank you very much!

PR #948 (2019-03-17): Fix remaining C99 errors
This should resolve #890.
closed 2019-03-17 by kisvegabor