lvgl/lvgl #893: how set label's txt's color?

3038922, 2019-02-24:
```
#define RESET "\033[0m"
#define BLACK "\033[30m"              /* Black */
#define RED "\033[31m"                /* Red */
#define GREEN "\033[32m"              /* Green */
#define YELLOW "\033[33m"             /* Yellow */
#define BLUE "\033[34m"               /* Blue */
#define MAGENTA "\033[35m"            /* Magenta */
#define CYAN "\033[36m"               /* Cyan */
#define WHITE "\033[37m"              /* White */
#define BOLDBLACK "\033[1m\033[30m"   /* Bold Black */
#define BOLDRED "\033[1m\033[31m"     /* Bold Red */
#define BOLDGREEN "\033[1m\033[32m"   /* Bold Green */
#define BOLDYELLOW "\033[1m\033[33m"  /* Bold Yellow */
#define BOLDBLUE "\033[1m\033[34m"    /* Bold Blue */
#define BOLDMAGENTA "\033[1m\033[35m" /* Bold Magenta */
#define BOLDCYAN "\033[1m\033[36m"    /* Bold Cyan */
#define BOLDWHITE "\033[1m\033[37m"   /* Bold White */
```

```
  fprintf(_data, "%s%s\n", GREEN , "hellowrold");//saveTo txt
```
it saved like this.
[image]

i use fopen, it can be red and have colors.
[image]

```
lv_obj_t * debugLab=lv_label_create(parent,NULL);
 lv_label_set_text(debugLab, logData);//logData is my saved txt file.
 lv_label_set_recolor(debugLab, true);
```
He can't parse escaped characters directly
[image]

BesitzeRuf, 2019-02-24:
Hi @3038922, 
it is terminal (or your application) job to parse input text and when escaped character \033 was found, apply coorect color to the following text.

3038922, 2019-02-24:
> Hi @3038922,
> it is terminal (or your application) job to parse input text and when escaped character \033 was found, apply coorect color to the following text.

How can I create a label that automatically parses and configures colors based on escaped characters in a simple way?

BesitzeRuf, 2019-02-24:
@3038922, I think, with a single label it is not possible... First you need to parse your input text and dynamically create labels for each substring you will parse between escape sequences.

embeddedt, 2019-02-24:
@3038922 @BesitzeRuf You need to use the recoloring interface. The label documentation explains it here: https://docs.littlevgl.com/#Label

kisvegabor, 2019-02-24:
> You need to use the recoloring interface.

I also think it should be used to convert your color codes to embedded text coloring for LittlevGL labels.

3038922, 2019-02-25:
how can i know the color;'s code?
```
   lv_color_t mycolor = LV_COLOR_RED;
    std::cout << mycolor.full << std::endl;
```
terminal result:
```
4294901760
```
it seem's wrong.

3038922, 2019-02-25:
It's easier for me to make a console page `DEBUG'.

closed 2019-02-25 by 3038922