lvgl/lvgl #894: Display driver not working correctly after a while?

TimLeijten, 2019-02-24:
Hi,

I'm new to lvgl, and just wrote a driver for my ILI9341 screen. I thought I was finally done and it was working, but it turns out it doesn't. When items are initally drawn, everything is fine. But once they are updated by changing their value or pressing them, it gets weird:
Before pressing button:
[image]

After pressing button:
[image]

This is my driver:
```
void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)
{
  /*Return if the area is out the screen*/
    if(x2 < 0) return;
    if(y2 < 0) return;
    if(x1 > 320 - 1) return;
    if(y1 > 240 - 1) return;

    /*Truncate the area to the screen*/
    int32_t act_x1 = x1 < 0 ? 0 : x1;
    int32_t act_y1 = y1 < 0 ? 0 : y1;
    int32_t act_x2 = x2 > 320 - 1 ? 320 - 1 : x2;
    int32_t act_y2 = y2 > 240 - 1 ? 240 - 1 : y2;

  uint16_t full_w = act_x2 - act_x1 + 1;
  tft.startWrite();
  tft.setAddrWindow(act_x1, act_y1, act_x2+1, act_y2+1);

  int xx = y1;
  for(unsigned int i = 0; i <= act_y2-act_y1; i++)
  {
    tft.writePixels((uint16_t*)color_array, act_x2 - act_x1 + 1);
    color_array+=full_w;
  }
  tft.endWrite();
  /*Tell the flushing is ready*/
  lv_flush_ready();
}
```
I have no idea what's wrong. Do you have any idea what might be wrong? It could be my driver, but I'm not sure as the initial drawing of the items is fine.

Thanks!

embeddedt, 2019-02-24:
@TimLeijten I think it's your driver. I would double check that `act_x2 - act_x1 + 1` always equals `full_w`. Chances are it works the first time because we rewrite the whole screen at the beginning.

TimLeijten, 2019-02-24:
> @TimLeijten I think it's your driver. I would double check that `act_x2 - act_x1 + 1` always equals `full_w`. Chances are it works the first time because we rewrite the whole screen at the beginning.

Any idea what might be wrong with it? I'm not sure what to check.

embeddedt, 2019-02-24:
@TimLeijten I gave a suggestion in my comment.

kisvegabor, 2019-02-24:
I suspect `tft.setAddrWindow(act_x1, act_y1, act_x2+1, act_y2+1);`  
Try `tft.setAddrWindow(act_x1, act_y1, act_x2, act_y2);`

TimLeijten, 2019-02-25:
> I suspect `tft.setAddrWindow(act_x1, act_y1, act_x2+1, act_y2+1);`
> Try `tft.setAddrWindow(act_x1, act_y1, act_x2, act_y2);`

Didn't work either. But I managed to fix it by using the buildin tft.drawRGBBitmap function. Didn't even know it existed.

I have one more question though, not related to this. Would it be possible to use a joystick to navigate littlevgl instead of just the tab button? And how would I implement that without a full keyboard?(I can only find the keyboard driver, no joystick driver in the examples)

Thanks!

kisvegabor, 2019-02-25:
> Didn't work either. But I managed to fix it by using the buildin tft.drawRGBBitmap function. Didn't even know it existed. Thanks for your help though!

Great!

> I have one more question though, not related to this. Would it be possible to use a joystick to navigate littlevgl instead of just the tab button? And how would I implement that without a full keyboard?(I can only find the keyboard driver, no joystick driver in the examples)

Have look at this keyboard driver: https://github.com/littlevgl/lv_drivers/blob/master/indev/keyboard.c#L93

It translates the pressed key to LittlevGL control keys.

As this topic is not related to the original, if you have further questions please open a new issue for it.

closed 2019-02-25 by kisvegabor

vitormhenrique, 2019-02-25:
@TimLeijten can you share your disp_flush with the drawRGBBitmap function?

vitormhenrique, 2019-02-25:
I got my ILI9341 working using, I;m using TFT_eSPI as a driver as it has better performance over Adafruit GFX

<pre><code>void disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_p)
{
  tft.pushImage(x1, y1, (x2 - x1 + 1), (y2 - y1 + 1), (uint16_t*) color_p);
  lv_flush_ready();
}</code></pre>