lvgl/lvgl #895: lv_group_set_style_mod_cb: Can I use animations?

ederguidsa, 2019-02-24:
Hi, I'm using the `lv_group_set_style_mod_cb` functionality in order to change the style of a given focused object in a group. 
I was wondering if it would be possible to apply style animations to the focused object using this callback.

embeddedt, 2019-02-27:
@kisvegabor

kisvegabor, 2019-02-28:
Unfortunately, you couldn't. The `style_mod` callback changes the style on the fly when it's required.

What you can do is to get the current time with `lv_tick_get()` in your `style_mod` callback. and set some parameters accordingly. E.g. set the background color dynamically. But in this case, you need to create an `lv_task` which continuously invalidates the focused object `lv_obj_invalidate(...)` to refresh it with the new style.

kisvegabor, 2019-03-02:
In the past few days [LittlevGL's Forum](https://forum.littlevgl.com) has launched to discuss questions there. See this [Announcment](https://forum.littlevgl.com/viewtopic.php?f=3&t=5).

If the issue still persists please open a topic for it on the Forum.

closed 2019-03-02 by kisvegabor