lvgl/lvgl #898: Possible to flip the screen 180°?

harteware07, 2019-02-26:
Is there a quick and easy way to flip the screen 180°? Could do it in the flushing driver but the touchpad coordinates wont be correct anymore.
Any ideas?

embeddedt, 2019-02-26:
@harteware07 I think that feature is being developed for `v6.0`. In the meantime why not just flip it in both the touch and display drivers?

harteware07, 2019-02-26:
> and display drivers
not so easy after all. I use DMA to flush the buffer and DMA can only increment on copy. Correct me if I am wrong but I would have to change all the places in code where lvgl draws something on the buffer?

embeddedt, 2019-02-26:
@harteware07 If you're using DMA then it might be better to wait until rotation support is implemented.

harteware07, 2019-02-26:
@embeddedt  Thanks a lot for your time!

closed 2019-02-26 by harteware07

puzrin, 2019-02-28:
@embeddedt

> I think that feature is being developed for `v6.0`. In the meantime why not just flip it in both the touch and display drivers?

GUI use internal buffer for partial render. This buffer should be ready for easy DMA transfer. Imagine, you have portrait orientation of display (90 degrees). In this case GUI should select a proper tiling sequence, to allow "long transfers".

Note, all possible rotations & mirroring should be considered. Probably, mirroring can be in driver, but rotation should be in GUI.