lvgl/lvgl #899: Unicode Bidirectional Algorithm

beibean, 2019-02-26:
Hi there,

How about include [Unicode Bidirectional Algorithm](https://www.w3.org/International/articles/inline-bidi-markup/uba-basics) to LittlevGL? Since:

- It is supossed to be used all over the world.
- It supports unicode characters.
- It supports different languages handling.

It seems hard to implement it, but doing a quick research there is a library that could do it for us --> [GNU FriBidi](https://github.com/fribidi/fribidi).

I have never work with Unicode Bidirectional Algorithm, but maybe any of you have and could tell us about the possibility of including it for LittlevGL.

embeddedt, 2019-02-26:
@beibean 

1. Is FriBidi designed for embedded systems  (i.e. low memory requirements, small code size)?
2. It's LGPL licensed. We only accept MIT-licensed code into the project. I also don't think relying on incompatibly-licensed external libraries is wise.

@kisvegabor

beibean, 2019-02-26:
As I said, I don't know much about this library and Unicode Bidirectional Algorithm in general.

It was just a suggestion to take into account, using FriBidi, other third party library or implementing by ourselves. 

I really think this will endow LittlevGL with a good feature ;)

kisvegabor, 2019-02-27:
I'm planning to implement a simple and extendible text shaping engine for `v6.0` later. 

My idea is to assign some glyph modifier rules to the fonts.

I leave this issue open as memo.

puzrin, 2019-04-04:
I think it doesn't worth consider each unicode feature separate. Because attempt to add every new feature can change everything in API and implementation. It would be more consistent to inspect existing JS project, pick required features subset and design all at once.

- [FontKit](https://github.com/foliojs/fontkit) - better glyphs combine, limited kerning tables support, no hinting support.
- [opentype.js](https://github.com/opentypejs/opentype.js) - meduim combine, better kerning, hinting
- [text-rendering-tests](https://github.com/unicode-org/text-rendering-tests) - tests for different rendering engines

Fontkit uses [unicode-properties](https://github.com/foliojs/unicode-properties) to access character attributes. Compressed DB size is ~ 12Kb.

kisvegabor, 2019-04-05:
I was thinking about the complexity of this whole thing (text shaping, bidirectional support, kerning etc) and I realized it's a much bigger work (if done well) than it seemed at first look. Knowing that it should be the main feature of a new release later, instead of parot of `dev-6.0`.

pmp-p, 2019-09-03:
Seriously, use freetype+sdl+harfbuzz on less cheaper hardware. 
That is a pandora box one should never open especially knowing there are scripts requiring deferred shaping.

amirgon, 2019-09-03:
> Seriously, use freetype+sdl+harfbuzz on less cheaper hardware.

At some point I will need RTL support, either through standard feature like this one or by some hack.  
I don't care much about compliant unicode bidirectional support, I just need a proper way to display RTL characters, align text to the right and mix RTL characters with digits and other symbols which are LTR characters.  

I don't think it makes sense to switch hardware and software framework to achieve this. It should be simple enough to do in LittlevGL on a cheap MCU, at least for my use case.

pmp-p, 2019-09-03:
@amirgon then i'd suggest a pre-rendering of words via the SDL simulator ( with added harfbuzz+freetype that's easy) and storing that in flash, think of a gettext but with graphical hooks encoded on https://en.wikipedia.org/wiki/Private_Use_Areas

amirgon, 2019-09-03:
> then i'd suggest a pre-rendering of words via the SDL simulator

@pmp-p This is good for text you can pre-render.  
But I'm getting some of the text from a remote server (as text strings), and I must render it on the embedded device.

pmp-p, 2019-09-03:
well if it's RTL only that should be ok, my remark is from past experience and was specifically about "text shaping, bidirectional support, kerning etc" because you need to think callbacks from the start for rendering text and that eats plenty of memory, also you can't get width/cesure until draw is completed => no batching.

kisvegabor, 2019-09-03:
Let's warm up this topic and collect goals and requirements. @amirgon, if I remember well, you live in an Arabic country so you know the details of  RTL.

Things not clear for me:
1. If you use RTL in "real life" your editor is still LTR?
2. How should the LTR `"some word 123"` string rendered in RTL?
3. How should the LTR `"word1 > word2"` string rendered in RTL?

The required features I see:
- Text shaping:
  - Built-in support for very common character replacement (E.g. Arabic)
  - Possible callback to extend it
- RTL support
  - Text level? A text can be RTL or LRT, no mixing
  - With control characters? E.g. `RTL "right to left" LTR "left to right" RTL "rtl again"`
- Vertical writing?
  - I don't think it's required

What do you think?

pmp-p, 2019-09-03:
Here's a good overview https://www.w3.org/TR/alreq

amirgon, 2019-09-03:
> @amirgon, if I remember well, you live in an Arabic country so you know the details of RTL.

Almost! 🤣 
I live in Israel so my main RTL language is Hebrew. I can answer these question in that context. AFAIK Arabic support is a bit harder because one unicode code-point could represent multiple shapes (there is more than one way to write a letter, depends on letters before/after it), and letter shapes are connected to each other. In Hebrew it's simpler - letter shapes are disconnected and each letter has a single form. 

> 1\. If you use RTL in "real life" your editor is still LTR?

When I edit text, if I enter latin characters or numbers, they are typed LTR. If I enter Hebrew characters, they are inserted RTL on the position the cursor is at.

> 2\. How should the LTR `"some word 123"` string rendered in RTL?

איזושהי מילה 123

> 3\. How should the LTR `"word1 > word2"` string rendered in RTL?

מילה1 > מילה2

(Notice the editor actually replaced `>` with `<` automatically while I was typing! I don't think you need to support something like that...)

> RTL support
> * Text level? A text can be RTL or LRT, no mixing

Actually mixing is important because numbers are LTR and text is RTL.

For my purpose, the priority would be:
- Aligning text to the right on label and text area etc.
- Mixing RTL and LTR, at least for display purposes when a string is given.

pmp-p, 2019-09-03:
> For my purpose, the priority would be:
> 
>     * Aligning text to the right on label and text area etc.
> 
>     * Mixing RTL and LTR, at least for display purposes when a string is given.

i think it that is implemented then it should be possible to handle basic arabic with the MS  (very)  simplified arabic fixed font (also known as CP1256 codepage ).

kisvegabor, 2019-09-03:
> Here's a good overview https://www.w3.org/TR/alreq

@pmp-p Thank you for this reference. It seems (even on a basic level) it's far more complicated than simply replacing some characters based on the context. The [diacritics](https://www.w3.org/TR/alreq/#h_diacritics) is quite difficult to handle because (according to the article) they need some hints from the font file itself. Are the diacritics parts of the "real life" text? Or are they special cases? 

@amirgon Thank you, getting clearer now.

pmp-p, 2019-09-03:
@kisvegabor yeah diacritics and ligatures are omnipresent that's why cp1256 is really not the best solution to render whole arabic words.

kisvegabor, 2019-09-04:
> diacritics and ligatures are omnipresent

That's a big problem because we don't have the required data from the font file to properly place the diacritics. I'm a little bit puzzled at this point... 

One thing is still not totally clear. When you type the last complex character shown in [Figure 15](https://www.w3.org/TR/alreq/#h_diacritics) you hit the "base characters" and then the two diacritics (3 key hits in total) and the last character is continuously changing in the editor? And in the memory, there will be 3 characters?

pmp-p, 2019-09-04:
Lot's of ?  :)  but i'm in the same position as you : as i said it was "past" experience unicode was not yet a thing (and i still can't read or write arabic ! ). 

From what i remember of early pyqt arabic support i don't think it's composition ( all char are stored ) and end of word is changing while typing ( and that's why rendering is delayed via a callback ), erasing char is also very funky and need hammering the del key.

kisvegabor, 2019-09-05:
Uh, that's tough. It seems to add diacritics in run time won't work because the placement information is missing. The possible solutions I see:
1.  Add that missing information to the fonts. (looks like a long and complicated development)
2. Somehow store the composed characters. (no standard, difficult to use)
3. Pre-render characters or texts. (not dynamic) 

The RTL support itself, and the character replacement looks also trick but possible.   

To properly join characters coordinates also needs to be returned which implies a more complicated text shaping engine.

@amirgon 
You mentioned only to support RTL. How do see the issues mentioned above?

amirgon, 2019-09-05:
> @amirgon
> You mentioned only to support RTL. How do see the issues mentioned above?

@kisvegabor In Hebrew it's much simpler, as I mentioned above - no diacritics whatsoever, and each code point has a single shape, no special text shaping.  

So the only complications over Latin script (which is currently supported) are
- Aligning everything to the right
- Handling RTL and LTR in the same string. This is important because numbers in Hebrew are written LTR as usual, and it's not uncommon to find an english word in a sentence here and there.

I don't think that's very hard. Here is what I had in mind:  

- Just aligning (with `lv_ta_set_text_align` for example) is not enough because we need the first character of the logical string to be the right-most character visually, so we would need to reverse the order of the characters visually when displaying an RTL string.
- For each code point we would need to identify the Bidi information: "directionality" of each character and whether it's "strong" or "neutral" (punctuation, whitespace). [Some general explanation about Bidirectional Text on Wikipedia](https://en.wikipedia.org/wiki/Bidirectional_Text).  btw there are a few more types such as "weak" and "explicit formatting", but I don't think there's a need to implement these, at least not at first stage.
Specifying Bidi information can be done by either adding it to the font data, or even by some callback that specifies Unicode ranges hard coded. For Hebrew script, there are only a few such ranges.
- According to the Bidi information (character "directionality", "Strong/Neutral"), we can implement a **very simplified** Bidi algorithm that takes a logical string, identifies "runs" of characters with the same directionality, takes into account "strong/neutral" characters and creates the visual representation of the logical string.  

This would allow mixing RTL and LTR characters for *most* cases, in Hebrew, which is good enough for me.   

It *might* also be good enough for the very simplified Arabic script of CP1256 codepage, as @pmp-p [suggested](https://github.com/littlevgl/lvgl/issues/899#issuecomment-527365800), but I'm not completely sure about that.

kisvegabor, 2019-09-06:
@amirgon Thanks for the clarification. I've missed these differences in Hebrew. If I understood well in Hebrew:
- character replacement based on context is still required
- joining characters (change letter space) is not required

I've done some research and found that Hebrew is spoken by few million people while Arabic is spoken by more than a hundred million people. Would it be useful enough to support Hebrew and Arabic in a limited way (with CP1256 or similar)?

amirgon, 2019-09-06:
@kisvegabor 

> * character replacement based on context is still required

No, each code point has one shape, so there's no need to replace characters by context.  
The only additional information you need for each character (or character range) is Bidi information (is this character a strong RTL, a strong LTR, or neutral).

> * joining characters (change letter space) is not required

Correct.

> I've done some research and found that Hebrew is spoken by few million people while Arabic is spoken by more than a hundred million people. Would it be useful enough to support Hebrew and Arabic in a limited way (with CP1256 or similar)?

I don't know how useful a limited CP1256 Arabic script would be. We would need someone with more knowledge of Arabic script to comment on that. My knowledge is limited to Hebrew.
There are ~10 million native Hebrew speakers and ~420 million Arabic speakers, so I can see your motivation to promote Arabic script support in LittlevGL.

I'm using LittlevGL for my own project, where I would need Hebrew support and not Arabic at this point. This is where my motivation to promote Hebrew support in LittlevGL comes from.  
In case Hebrew will not be formally supported in LittlevGL, I'll probably need to fork LittlevGL and hack Hebrew support in it. But first I'm suggesting the option to do it more cleanly on upstream lvgl even though the target audience is smaller.

kisvegabor, 2019-09-08:
I see.
 To support Hebrew (and bidi in general) without text shaping we can store the original text and the bidi formatted text in the label. This way the bidi "translation" needs to be done only once in `lv_label_set_text()`

But word wrapping is still not clear:
The text "HEBREW ARABIC ENGLISH GERMAN" (let's say the "HEBREW ARABIC" words are RTL) should be written as "ENGLISH GERMAN CIBARA WERBEH". It should be wrapped as: 
```
BEH 
WER
ARA
CIB
ENG
LISH
GER
MAN
```

But the LittlevGL will wrap it as:
```
ENG
LISH 
GER
MAN 
CIB
ARA 
WER
BEH
```

So word wrapping seems another challenging topic. What do you think about it? Do I see the issue well?

amirgon, 2019-09-08:
> So word wrapping seems another challenging topic. What do you think about it? Do I see the issue well?

The behavior should be dependent on the bidi state **of the label itself**.  
(or if you like, the default bidi state of the whole text in the label)

I'll try to demonstrate this on PowerPoint with a TextBox.
The text I typed is "עברית ערבית English German"
(Logically, the first two words are Hebrew and the last two words are English)

If the label is set as LTR:

[image]

And if it's set as RTL:

[image]

kisvegabor, 2019-09-08:
Thanks! Playing in PowerPoint is a good idea. 

This point is interesting. 
[image]

It rather begins a new line than put "Eng" in the same line with Hebrew characters.  It seems it's true only if the word itself is wrapped and whole words with different write direction can be in the same line.

I'm thinking about a reasonable approach to implement it.

kisvegabor, 2019-09-08:
If the words are long enough Libre Office Impress can mix LTR and RTL even is the word is wrapped: 
[image]

kisvegabor, 2019-09-20:
I was thinking about it for a while and could come up with a reasonable approach but it has some limitations:
1. No text shaping for now, "only" RTL support
2. Support only LTR base direction
3. In case of line-wrapping, add a line break when the direction changes.
E.g.
```
ENGL
ISH 
GERM
AN      < Line break on direction change
RBEH
A WE
IBAR
   C
```

What do you think??

amirgon, 2019-09-20:
@kisvegabor 

> * No text shaping for now, "only" RTL support

Ok.

> * Support only LTR base direction

Actually I would need RTL base direction and alignment for my application, otherwise it will not be presented correctly (an additional English word would appear on the wrong side of the Hebrew sentence).  
So choosing RTL/LTR base direction is important for me.  

I guess this could be a "global" flag, to make the whole application LTR or RTL, if you think marking this per component is too expensive (although that's just 1 bit). 

> * In case of line-wrapping, add a line break when the direction changes.

I could live with that, I think.

kisvegabor, 2019-09-21:
@amirgon 

I see. In your particular application how important it is to handle line-wrapping? Basically almost all complications come from this. I see a way to implement it, but it would require another layer above "getting Unicode chars from UTF8". A lot of complicated parts of the library would be affected therefore probably some bugs would appear in various places.

I'm not against to add a proper way to handle line wrapping but I'd like to be sure if it's really required.

amirgon, 2019-09-21:
> I'm not against to add a proper way to handle line wrapping but I'd like to be sure if it's really required.

@kisvegabor For the first phase of my application, I don't need line wrapping.   
I think it's a good idea to support line wrapping in general, but we could split the work to phases and initially implement RTL support without line wrapping, as the first phase.  

I do need, however, RTL base direction support (as I explained).

kisvegabor, 2019-09-21:
> .. initially implement RTL support without line wrapping, as the first phase.

I afraid the two things are independent. A "simple" RTL support is quite straight forward: create a new text where the character's order is changed as required. It's a modular thing, basically an extra step in `lv_label_set_text()` like `rtl_process(str_in, str_out)`.
On the other hand, to properly support line-wrap the text processing engine should be extended and reworked.

So it's rather a "quickly have something which is enough in lot of use-cases" approach.

> I do need, however, RTL base direction support (as I explained).

Clear, it's ok.

amirgon, 2019-09-21:
> It's a modular thing, basically an extra step in `lv_label_set_text()` like `rtl_process(str_in, str_out)`.

@kisvegabor This I can do out of the scope of lvgl. Just translate one string to another before setting the text of a label.  
- Is there extra value in implementing a "simple" RTL support in lvgl vs. processing the string before setting the text in lvgl?
- Did you think about of how to implement the base RTL/LTR? Globally? Per component?

kisvegabor, 2019-09-21:
> Is there extra value in implementing a "simple" RTL support in lvgl vs. processing the string before setting the text in lvgl?

It really can be handled outside of lvgl as you described (even with a built-in lvgl function), but in this case, the user needs to take care with the extra buffer. E.g.
```c
const char * txt = "some text";
char * txt_rtl = lv_mem_alloc(strlen(txt));
rtl_process(txt, txt_rtl);

/*Option1*/
lv_label_set_text(label, txt);
lv_mem_free(txt); 

/*Option2*/
lv_label_set_text_static(label, txt);
...
lv_obj_del(label);
lv_mem_free(txt);   

```

Another disadvantage could be that `lv_label_get_text()` will return a "messed-up" string. With internal handling both the original and the changed string could be stored.

On the other hand, this approach would give great flexibility. It'd be easy to pass ready-to-use strings to any object (e.g. `lv_win_set_title()`) without adding API functions for the direction and base direction.

> Did you think about of how to implement the base RTL/LTR? Globally? Per component?

If we find a good way to handle the rtl translation outside of lvgl (like above) then we can have two functions for ltr and rtl base direction.

Else handling it per object could bloat the API. E.g.:
- `lv_win_set_tile`
- `lv_win_set_tile_base_dir` (might be replaced with global config)

What do you think, which approach would be better? Do you know other advantages/disadvantages I've missed?

embeddedt, 2019-09-21:
@kisvegabor @amirgon Is RTL information normally stored inside the string itself (i.e. UTF-16/UTF-8 codes) or encoded separately?

kisvegabor, 2019-09-21:
@embeddedt The RTL information comes from the character code. A sequence of RTL character should be written Right to left.

Take look at this: https://stackoverflow.com/a/14824756/1999969

amirgon, 2019-09-22:
> What do you think, which approach would be better? Do you know other advantages/disadvantages I've missed?

Do you plan to implement a full RTL support in the future? Perhaps with features that are needed for Arabic scripts as well?  
If you do, maybe we can think of an API that would be forward compatible. For now it would only implement the simplistic string processing, but in the future the same API could be used for full RTL support.  
If we can think of such an API in advance, it could prevent breaking the API in the future and waiting for a major version update.  

For my specific purpose, memory allocation for text buffer is automatic when working with Micropython and I rarely use `get_text` functions, so handling it outside lvgl might be OK if only doing string processing.   

> @embeddedt The RTL information comes from the character code. A sequence of RTL character should be written Right to left.
> 
> Take look at this: https://stackoverflow.com/a/14824756/1999969

The RTL information mainly comes from the character code, as @kisvegabor mentioned, **but**:
- Unicode has some special characters to explicitly set RTL and LTR for some special cases, so it can also come from the string itself. (We don't have to support this in my opinion).
- The label itself has RTL "information" in the form of its base direction. The same string is displayed differently on two labels with different base directions.

Also, be aware that the `isRTL` algorithm on the link you provided **is wrong**.  
For example, it does not take into account code points with "neutral" directionality. So an RTL string that terminates with a period would be considered LTR although its RTL.

kisvegabor, 2019-09-22:
> Do you plan to implement a full RTL support in the future?

Let's consider to current no-line-wrap-support approach as an experiment to get more familiar with the BiDi in general. And we will see what to do with line-wrapping.

> Perhaps with features that are needed for Arabic scripts as well?

The problem with supporting Arabic is the advanced text shaping which requires info from the TTF support. I'd say it's out of scope for now.

> If you do, maybe we can think of an API that would be forward compatible. For now it would only implement the simplistic string processing, but in the future the same API could be used for full RTL support.

To ensure forward compatibility  RTL processing should be hidden in `lv_label_set_text`. If the  `LV_TXT_RTL_SUPPORT` is enabled in `lv_conf.h`, `lv_label_set_text()` can automatically do the required processing. 

> Unicode has some special characters to explicitly set RTL and LTR for some special cases, so it can also come from the string itself. (We don't have to support this in my opinion).

I've read some BiDi examples and find that in some cases algorithms are not able to correctly process the BiDi texts, therefore, special control characters are required. At first look, RTL or LTR isolation means only adding non-printable strong  RTL or RTL characters. (Not sure if it's really that simple though.)

> The label itself has RTL "information" in the form of its base direction. The same string is displayed differently on two labels with different base directions.

I have an idea for that. See below.

> For example, it does not take into account code points with "neutral" directionality. So an RTL string that terminates with a period would be considered LTR although its RTL.

I haven't seen such a rule so far. Can you link some docs to read about it?

---

I summarize what I know so far and what are my approaches.
- There are strong (usually letters), weak (usually arithmetic symbols) and neutral (usually white spaces) characters: See [this](https://en.wikipedia.org/wiki/Bidirectional_Text#Unicode_bidi_support).
- The text is split into RTL and LTR runs. 
- One run lasts until a strong char is not found with a different direction.
- Runs of strong RTL characters should be displayed as RTL. Similar for LTR.
- Neutral chars in strong RTL runs should be displayed as they are. (E.g. space between words). Similar for LTR.
-  "Symmetric"  weak characters (such as `()[]{}<>`) should be changed their pair in RTL runs.
- European numbers (e.g. 456) should be displayed as LTR in RTL runs too.
- The meaning of LTR/RTL base direction is to determine the display order of runs.
   - Idea: we can use a control character as the first character of the text to determine the base direction. This way the whole BiDi processing can be hidden in `lv_label_set_text`.

To simply sharing examples I suggest considering uppercase ASCII letters as strong LTR and lower case letters as strong RTL.

Here I've listed some use cases according to my best understanding and described MEM (as in the memory, as they should be read), LTR (LTR base direction), RTL (RTL base direction).
Please verify or correct them.

```
MEM: "abc ABC" 
LTR: "cba ABC"
RTL: "ABC cba"

MEM: "ABC abc" 
LTR: "ABC cba"
RTL: "cba ABC"

MEM: "abc def ABC DEF" 
LTR: "fed cba ABC DEF"
RTL: "ABC DEF fed cba"

MEM: "abc123 def  ABC123 EFG"
LTR: "fed 123cba ABC123 EFG"
RTL: "ABC123 EFG fed 123cba"

MEM: "abc (def) 123 < 456 ABC"
LTR: "456 > 123 (fed) cba ABC"
RTL: "ABC 456 > 123 (fed) cba "

MEM: "I'VE READ war and peace 25 TIMES"
LTR: "I'VE READ  peace dna raw  25 TIMES"
RTL: "TIMES 25 peace dna raw I'VE READ"
```

amirgon, 2019-09-22:
> > Also, be aware that the `isRTL` algorithm on the link you provided is wrong.
> > For example, it does not take into account code points with "neutral" directionality. So an RTL string that terminates with a period would be considered LTR although its RTL.
> 
> I haven't seen such a rule so far. Can you link some docs to read about it?

@kisvegabor 
I just tried out the [playground page](https://codepen.io/vsync/pen/rcwui) provided on that [SO post](https://stackoverflow.com/a/14824756/1999969) and noticed that when I type RTL string and add a period, which is a neutral character, the string is switched to LTR although it should not:

[image]

---

> * Neutral chars in strong RTL runs should be displayed as they are. (E.g. space between words). Similar for LTR.

I think this is wrong. Neutral chars should inherit the directionality of current run. This is especially important for punctuation symbols (`,.;:!?` etc.) so they are positioned correctly.  
Take for example the exclamation mark - it must be positioned after a word. So
```
סימן! קריאה
```
inherits RTL because it's inside an RTL run, whereas -
```
Exclamation! mark
```
inherits LTR.  

An interesting case is the start/ending of a string. Where would the exclamation mark go in such case?  
This depends on the **base direction** of the component.
For example, when the component is LTR:
```
סימן קריאה!
```
This would seem wrong to a Hebrew speaker. To fix this, **the base direction of the component itself** should be set to RTL in order to position the exclamation mark on the correct (left) position.  
After setting base direction to RTL  it would look like this:

[image] 

> * The meaning of LTR/RTL base direction is to determine the display order of runs.

I would say it determines the "default" direction, and especially affects the characters at the beginning/end of the string. If characters at the beginning/end of the string are neutral, they will be considered as surrounded by the base direction from one side.

> * Idea: we can use a control character as the first character of the text to determine the base direction. This way the whole BiDi processing can be hidden in `lv_label_set_text`.

I don't like this idea.  
It means that every string would need to be changed before calling `set_text` to add this control character.   
This special character would only need to be added when setting text of RTL controls, so I would need to add it depending on which control I plan to set this text on.  

I my opinion, it's better to separate the data (string logical content) and view (string visual display parameters).

I have a different suggestion: Can you spare one bit on `lv_obj_t` struct? (only when LV_TXT_RTL_SUPPORT is defined of course)  
Let's say every object can be set to be either LTR or RTL. When creating a child object, this bit would be automatically set according to the parent, and can also be set by the user with a `set` function. So it's very flexible - you can set the whole screen as RTL before you add child objects, or you can set a page, or a single component if you want.

---

Regarding your examples - I've added Hebrew letters that correspond to your examples.
It really helps to validate your examples. I type each string according to your MEM and see what is displayed on the browser text box. With a right-click I can even change the base direction from LTR to RTL and see how it looks in each case.

On those cases I saw something different I marked the places ("WRONG") where I think it's wrong, and added ("CORRECT") what it should be instead. If I didn't specify "WRONG" or "CORRECT" then this is your original example and it's correct

```
"אבג ABC"
MEM: "abc ABC" 
LTR: "cba ABC" 
RTL: "ABC cba"

"ABC אבג"
MEM: "ABC abc" 
LTR: "ABC cba"
RTL: "cba ABC"

"אבג דהו ABC DEF"
MEM: "abc def ABC DEF" 
LTR: "fed cba ABC DEF"
RTL: "ABC DEF fed cba"

"אבג123 דהו ABC123 EFG"
MEM: "abc123 def  ABC123 EFG"
LTR: "fed 123cba ABC123 EFG"
RTL: "ABC123 EFG fed 123cba"

"אבג (דהו) 123 < 456 ABC"
--> WRONG MEM: "abc (def) 123 < 456 ABC"
--> CORRECT MEM: "abc )def( 123 > 456 ABC"
LTR: "456 > 123 (fed) cba ABC"
RTL: "ABC 456 > 123 (fed) cba "

"I'V READ מלחמה ושלום 25 TIMES"
MEM: "I'VE READ war and peace 25 TIMES"
--> WRONG: LTR: "I'VE READ  peace dna raw  25 TIMES"
--> CORRECT: LTR: "I'VE READ 25 ecaep dna raw TIMES" 
--> WRONG: RTL: "TIMES 25 peace dna raw I'VE READ"
--> CORRECT: RTL: "TIMES 25 ecaeo dna raw I'VE READ"
```

embeddedt, 2019-09-22:
@amirgon The last line of your post has `TIMES 25 ecaeo dna raw I'VE READ` instead of `TIMES 25 ecaep dna raw I'VE READ` (note the "p" in peace). Is that a mistake?

amirgon, 2019-09-23:
> @amirgon The last line of your post has `TIMES 25 ecaeo dna raw I'VE READ` instead of `TIMES 25 ecaep dna raw I'VE READ` (note the "p" in peace). Is that a mistake?

Of course this is a mistake. Fixed above.

kisvegabor, 2019-09-23:
@amirgon Thank you for the corrections.

> I just tried out the playground page provided on that SO post and noticed that when I type RTL string and add a period, which is a neutral character, the string is switched to LTR although it should not:

It seems like a bug to me. 

> I think this is wrong. Neutral chars should inherit the directionality of current run. This is especially important for punctuation symbols (,.;:!? etc.) so they are positioned correctly.

True!

> An interesting case is the start/ending of a string. Where would the exclamation mark go in such case?
This depends on the base direction of the component.

Can't the staring weak/neutral chars inherit the direction of the first strong char? It makes more sense to me because they are related to it anyway. E.g with RTL base dir:
```
intention: "! exclamation"
"! " get base dir:  " !exclamation"
"! " get the dir of first strong char: "! exclamation"
```

> In my opinion, it's better to separate the data (string logical content) and view (string visual display parameters).

I've found that in the practice you can't separate data from view even in [CSS/HTML](https://www.w3.org/International/articles/inline-bidi-markup/). You need to manually markup parts of the text because sometimes the BiDi algorithm does not give the expected result.

A typical issue with the algorithm is shown [here](https://www.w3.org/International/articles/inline-bidi-markup/uba-basics#isolation). So it's inevitable to deal with markups in the text. In the most simple case hidden strong chars can be used to isolate/split runs.

Regarding the base direction, once control chars need to be used anyway, I don't find a control char for base dir so evil. 

We can even autodetect base dir based on the first strong char which can be overwritten by a control character. 

I found [this](https://www.w3.org/International/articles/inline-bidi-markup/uba-basics) short and straightforward write-up. From this the most essential rules are (we have already mentioned some of them):
1. A directions run lasts until a strong char with different direction is not found.
2. Neutrals between the same dir strong chars are treated as the dir of the strong chars. 
3. Neutrals between two different dir strongs inherit the base dir.
4. Numbers (on any language) are always displayed as LTR. 

An experiemnt about how a "space" behaves.
```
MEM: "ABC abc"
LTR base dir: 
  - Run1 (LTR): "ABC " (space comes here)
  - Run2 (RTL): "abc" -> swap "cba"
  - Result: "ABC cba"

RTL base dir: 
  - Run1 (LTR): "ABC"
  - Run2 (RTL): " abc" -> swap "cba "   (space comes here)
  - Result: "cba ABC"
```

A question regarding the alignment. Shall we automatically align texts with RTL base dir to the right?

kisvegabor, 2019-09-23:
I've added my [first attempt](https://github.com/littlevgl/lvgl/blob/2c5c4abdcedfd527d0072c7b37ed9fde6cd8a369/src/lv_misc/lv_txt.c#L349-L435) to implement a "get runs" algorithm to `feat-bidi` branch.

Now it just prints the found runs. It can be tested like the following. (Still lowercase is considered as RTL):
```c
 const char * s1 = "abc123 ABC123 aAbbBB";
    char s2[32];
    lv_txt_rtl_proc(s1, s2, LV_TXT_DIR_RTL);  //Last param is the base dir
```

amirgon, 2019-09-23:
@kisvegabor 

> Can't the staring weak/neutral chars inherit the direction of the first strong char?

I'm not an expert, but I don't think this is how it's supposed to work.  
It's very easy to make empiric tests and see how other applications (PowerPoint, the web browser) behave.  

Here, for example, I tried it on LibreOffice.
First paragraph was set as LTR, and the second as RTL:
[image]
As you can see, the period and the colon position is not dependent, as you suggest, on the direction of the first strong char, but on the base direction of the paragraph.  

You'll get the same behavior on other applications, and even on HTML elements (where you set the base direction with the [`dir` attribute](https://www.w3schools.com/tags/att_global_dir.asp) to `rtl` or `ltr`)  

On HTML5 you have the option to use `dir="auto"`, **but it's only recommended if the text direction is unknown**, which is not the case, at least for my application and I think in most other cases.
By the way, HTML5 `dir="auto"` attribute [works as you suggested](https://www.w3.org/International/questions/qa-html-dir#dirauto):
> HTML5 added an auto value to the dir attribute that can help here. The auto value tells the browser to look at the first strongly typed character in the element and work out from that what the base direction of the element should be. If it's a Hebrew (or Arabic, etc.) character, the element will get a direction of rtl. If it's, say, a Latin character, the direction will be ltr.

> There are some corner cases where this may not give the desired outcome, but in the vast majority of cases it should produce the expected result.

I suggest providing a way to mark the base direction of an element, and not always default to this method of automatically detecting the direction.

> I've found that in the practice you can't separate data from view even in [CSS/HTML](https://www.w3.org/International/articles/inline-bidi-markup/). You need to manually markup parts of the text because sometimes the BiDi algorithm does not give the expected result.

I think it's very rare case that a string is specially modified (by adding special RTL/LTR marks in it) in order to display a string correctly.  
I think that in most cases, the solution to [the isolation problem you present](https://www.w3.org/International/articles/inline-bidi-markup/uba-basics#isolation) is to use two labels (or text boxes etc.) each one with the correct base direction.    

I don't agree with your statement that "it's inevitable to deal with markups in the text."  
Instead of markup in the text (the data), the layout should have different components (view) for isolated parts.

> Regarding the base direction, once control chars need to be used anyway, I don't find a control char for base dir so evil.

I think that in practice - control chars are not used.  
Instead, base dir is set on the component itself.  
The reason is that most developers don't have much understanding about RTL/LTR issues, so they don't know how and where to insert the control chars.  Instead, they solve the problem on the view by selecting the correct base direction for each component.

> We can even autodetect base dir based on the first strong char which can be overwritten by a control character.

The disadvantage of autodetecting the base dir is that it's not always correct.
A common example is a string that starts with a number. For example "2 Apples".  
If you detect base direction automatically, you would detect this string as LTR, and the result would be "Apples 2", which is wrong.  
In most cases we know, in advance, that the label is RTL and supposed to show Hebrew sentence. Auto-detecting this as LTR would yield the wrong result. Why prevent us from configuring this explicitly?  

> I found [this](https://www.w3.org/International/articles/inline-bidi-markup/uba-basics) short and straightforward write-up. From this the most essential rules are (we have already mentioned some of [...]

These rules (and the write-up) look good to me!

> A question regarding the alignment. Shall we automatically align texts with RTL base dir to the right?

That would be very helpful.
Probably every component with RTL base direction should be aligned to the right by default.

---

> I've added my [first attempt](https://github.com/littlevgl/lvgl/blob/2c5c4abdcedfd527d0072c7b37ed9fde6cd8a369/src/lv_misc/lv_txt.c#L349-L435) to implement a "get runs" algorithm to `feat-bidi` branch.

https://github.com/littlevgl/lvgl/blob/2c5c4abdcedfd527d0072c7b37ed9fde6cd8a369/src/lv_misc/lv_txt.c#L352

I assume that in the future RTL/LTR indication would be taken from the font data itself? Or did you think to use some hard-coded ranges for RTL unicode characters?

https://github.com/littlevgl/lvgl/blob/2c5c4abdcedfd527d0072c7b37ed9fde6cd8a369/src/lv_misc/lv_txt.c#L356-L364

I think that punctuation marks are neutral, not weak.
For the testing I've added the following:
```
static const char neutrals[] = " \t\n\r.,:";
```

What about marks that should be reversed? (`<` :arrow_right: `>` for example). Maybe they should be defined in a class of their own?

> Now it just prints the found runs. It can be tested like the following. (Still lowercase is considered as RTL)

The easiest way for me to run it is with Micropython!
We can very easily compile this on the js port and have an online web sandbox for trying out your code!  
For now, I'm running it on the unix port:
```
>>> lv.txt_rtl_proc("abc123 ABC123 aAbbBB", s, lv.TXT_DIR.LTR)
RTL: "abc123"
LTR: " ABC123 "
RTL: "a"
LTR: "A"
RTL: "bb"
LTR: "BB"
True
```
What about this? (I've added `.` to `neutrals`):
```
>>> lv.txt_rtl_proc("a.", s, lv.TXT_DIR.LTR)
RTL: "a."
True
>>> lv.txt_rtl_proc("a.", s, lv.TXT_DIR.RTL)
RTL: "a."
True
```
In this case I would expect the base direction to affect the position of the period.
Shouldn't the period be on a different run?

embeddedt, 2019-09-23:
> The easiest way for me to run it is with Micropython!
We can very easily compile this on the js port and have an online web sandbox for trying out your code!

Let me know whenever you want the JS port recompiled and I will rebuild and push it, if it saves you time.

amirgon, 2019-09-24:
> Let me know whenever you want the JS port recompiled and I will rebuild and push it, if it saves you time.

Thanks @embeddedt.
I'm ok with the unix port for now, but perhaps @kisvegabor would want this at some point.

kisvegabor, 2019-09-24:
Meanwhile, I've added `rtl_reverse` and moved the bidi related things to `lv_misc/lv_bidi.c/h`. The API has been changed accordingly:

```c
    const char * s1 = "abc123 ABC";
    char s2[32];
    lv_bidi_process(s1, s2, LV_BIDI_DIR_RTL);
```

```
RTL: "abc123 " -> " 123cba"
LTR: "ABC"
```

amirgon, 2019-09-24:
@kisvegabor 

> * Add `set_base_dir` API object's which have texts.
> * Add an `lv_obj_t` attribute (as you suggested). It should have 3 possible values: RTL, LTR, NONE (means inherit from the parent).

Why not both? Why 3 possibilities?
How about always inheriting from parent, and in addition provide `set_base_dir` so the user could call  it with an object and change its the direction?  
Practically, in most cases, users would want to set a single direction to the whole application, or to an entire page in a consistent way, so I guess `set_base_dir` would usually be used on screen or page objects, and very rarely on specific label.  
Having only two options (LTR, RTL) saves a little because it would consume only one bit on `lv_obj_t`.

I think it would be a good idea to set the initial screen direction according to a configuration option on `lv_conf.h`, since it's the parent of all other objects and therefore would be the initial (default) direction of all objects.  
For example, LV_LABEL_ALIGN_LTR means that every object that is created without a parent would be set to LTR.

> We can introduce `LV_LABEL_ALIGN_AUTO`. It can be the default value which sets the alignment according to the base dir.

If you want to support automatic alignment, we really need 3 possibilities (LTR, RTL, AUTO).  
But another option is to support this as a global configuration option on `lv_conf.h`:   If `LV_LABEL_ALIGN_AUTO` is defined, we can avoid adding the direction bit in `lv_obj_t` and always set it automatically.  
I assume this could be a default for users who don't want to build an "RTL" layout. (When designing RTL application, it's not just the directionality of the text, it's also how components are arranged and aligned on screen one vs. the other)

> There is no such data in the fonts right now. I meant to use hard-coded ranges. Do you see any problem with that?

No special problem, I think. There are relatively few ranges if we only consider Hebrew.  
On the general case, however, there are many ranges (See [ISO 15924](https://en.wikipedia.org/wiki/ISO_15924) and [also this](https://www.compart.com/en/unicode/bidiclass), but I don't think we need to support the general case).
Maybe those ranges could be configurable on `lv_conf.h`?

> > I think that punctuation marks are neutral, not weak.
> 
> I've found this [here](https://unicode.org/reports/tr9/#Boundary_Neutrals):
> 
> > The bidirectional types associated with numbers are called weak types, and characters of those types are called weak directional characters. With the exception of the directional formatting characters, the remaining bidirectional types and characters are called neutral.

[Here](https://www.w3.org/International/articles/inline-bidi-markup/uba-basics#neutralcharacters) I found this comment:

> Characters are usually classified as 'weak' when they are associated with numbers. A small number of punctuation characters are initially classed as weak, but in a non-numeric context are treated like neutrals. In consequence, in this article we will refer to all punctuation as neutral characters.

So it seems that some punctuation marks are sometimes considered as weak and some other times as neutrals...  

Regarding weak characters, weak characters are discussed [here](https://www.w3.org/International/articles/inline-bidi-markup/uba-basics#numbers):

> Because it is weakly typed, the number is seen as part of the preceding Arabic text, so the two Arabic words that surround the number are treated as part of the same directional run - even though the sequence of digits runs LTR on screen.

If I understand correctly, the sequence of numbers are considered LTR but the surrounding RTL characters are considered a single run, so their order is not switched.

Here is an example:
```
"אחת 123 שתיים"
"אחת abc שתיים"
```

So the order of `אחת` and `שתיים` depends on whether between them there's a **weak** run `123` or a **strong** run `abc`. But both weak and strong runs are LTR, **so their internal order is reversed** (as opposed to neutral, where the order is kept and everything would be the same RTL run).

An example of a number-related character is `$`.
For example:
```
אחת 100$ שתיים
אחת 100! שתיים
```
I think that what's happening here is that the `$` behaves as weak character, and is presented LTR, while the `!` is neutral and presented RTL. (in both cases, the logical "MEM" order is: `1` `0` `0` `!/$`

> Meanwhile, I've added `rtl_reverse` and moved the bidi related things to `lv_misc/lv_bidi.c/h`.

Maybe it's time to generate a font with Hebrew characters... I never created a font yet.
A few questions:
- Does the font generation tool work on Linux, or do I need a Window machine?
- Can I load a font on runtime and provide a pointer to it (that is preferred for Micropython), or do I need to build the project with the font embedded in the program?

kisvegabor, 2019-09-25:
I think we are on the same page with `base_dir` and `align` but maybe I wasn't exact enough. So here is my approach

**lv_obj_set_base_dir**
Sets the base direction of the writing. Has no effect on the label's position.
Can have 4 values:
1. INHERIT: default value
2. LTR
3. RTL
4. AUTO: depends on the first strong char. I've realized it is useful when the user can change between the languages runtime and you don't know the base dir of the objects in advance, however it's usually clear from the text.

You can call `lv_obj_set_base_dir(page/screen/label)` and the clidren will inherit it.

**lv_label_set_align**
Important only with multiline text. E.g. `ddlist`'s options are a long label with multiple lines. 
1. LEFT
2. RIGHT
3. CENTER
4. AUTO: default. Turns to LEFT or RIGHT depending on base dir.

> So the order of אחת and שתיים depends on whether between them there's a weak run 123 or a strong run abc. But both weak and strong runs are LTR, so their internal order is reversed (as opposed to neutral, where the order is kept and everything would be the same RTL run).

The last version should work like this.

> I think that what's happening here is that the $ behaves as weak character, and is presented LTR, while the ! is neutral and presented RTL. (in both cases, the logical "MEM" order is: 1 0 0 !/$

I can extend the algorithm as once it's in an "internal weak run of numbers" consider `$` as numbers. Maybe `.` and `,` should be treated similarly as well. 

> Does the font generation tool work on Linux, or do I need a Window machine?

Platform independent, but fortunately easier on Linux. However the easiest thing is to use the [Online converter](https://littlevgl.com/ttf-font-to-c-array).

> Can I load a font on runtime and provide a pointer to it (that is preferred for Micropython), or do I need to build the project with the font embedded in the program?

If you mean similarly to "bin" images it's not working for fonts. You need to generate C files and build with the project :frowning_face: It was already asked several times so probably it will be added once but needs some time to do.

amirgon, 2019-09-25:
@kisvegabor 

> I think we are on the same page with `base_dir` and `align` but maybe I wasn't exact enough. So here is my approach

Yes, your approach looks good! :+1:

> **lv_label_set_align**
> Important only with multiline text

Could you explain this?  How is multiline related? Why would I want to set align on multiline and not on single line?

> If you mean similarly to "bin" images it's not working for fonts. You need to generate C files and build with the project ☹️ It was already asked several times so probably it will be added once but needs some time to do.

That's unfortunate. It means that a Micropython would need to be built with specific fonts in it, instead of building it only once, and using the same firmware with whatever fonts we need.  

It also means we cannot use the online lvgl simulator to visualize things related to fonts, RTL among them.

What is missing in order to support this? I don't really need lv_fs to load it, I can handle loading it myself, I just need a way to register a new font on runtime by providing a pointer to a binary font file.  
Is it the creation of the binary font file (instead of C file) what is missing?

kisvegabor, 2019-09-25:
> Could you explain this? How is multiline related? Why would I want to set align on multiline and not on single line?

Let's say there label with "Apple" text. The width is automatically set to e.g. 100 px according to the width of the text. As the width is the same as the text's width, the text can be aligned. However, a text like "Apple\nB" also has width = 100px but the lines can be aligned:
Left:
```
Apple
B
```

Center:
```
Apple
  B
```

Right:
```
Apple
    B
```
It's true if the label has `LV_LABEL_LONG_MODE_EXPAND`. E.g. in case of `LV_LABEL_LONG_BREAK` if you specify the width even single lines can be aligned.

 > What is missing in order to support this? I don't really need lv_fs to load it, I can handle loading it myself, I just need a way to register a new font on runtime by providing a pointer to a binary font file.
Is it the creation of the binary font file (instead of C file) what is missing?

The font converter (if run from the command line) can produce bin output (`--format bin`). The details of the binary format are described [here](https://github.com/littlevgl/lv_font_conv/blob/master/doc/font_spec.md).

The task is to write two interface functions to get glyph info and bitmap. Similarly to the [interface functions of the C format](https://github.com/littlevgl/lvgl/blob/master/src/lv_font/lv_font_fmt_txt.c#L48-L104).
Then an `lv_font_t` variable needs to be initialized with the binary data and the interface functions. E.g. [lv_font_roboto_16](https://github.com/littlevgl/lvgl/blob/master/src/lv_font/lv_font_roboto_16.c#L2257-L2263).

The C format is essentially the same as the binary format just represented as structs instead of a blob.

amirgon, 2019-09-26:
@kisvegabor 
I tried a basic "Hello World" test with a real Hebrew font, and I have a few problems.  

First, I changed the RTL range on `lv_bidi_letter_is_rtl` to Hebrew script range:
```c
 bool lv_bidi_letter_is_rtl(uint32_t letter)
 {
-    if(letter >= 'a' && letter <= 'z') return true;
+    if(letter >= 0x5d0 && letter <= 0x5ea) return true;
 
     return false;
 }
```

I created a new font: 
```bash
node lv_font_conv.js --font /usr/share/fonts/truetype/culmus/FrankRuehlCLM-Medium.ttf -r 0x20-0x7F -r 0x5d0-0x5ea --size 16 --format lvgl --bpp 4 --no-compress -o ~/esp/projects/lv_mpy/lib/lv_bindings/lvgl/src/lv_font/lv_font_heb_16.c
```

And added it on `lv_conf.h` with 
```c
#define LV_FONT_CUSTOM_DECLARE LV_FONT_DECLARE(lv_font_heb_16)
```

Then I ran the following:
```python
scr = lv.obj()
btn = lv.btn(scr)
btn.align(lv.scr_act(), lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
s = lv.style_t()
lv.style_copy(s, label.get_style(label.STYLE.MAIN))
s.text.font = lv.font_heb_16
label.set_style(label.STYLE.MAIN, s)
text = u'\u05E9\u05DC\u05D5\u05DD\u0020\u05E2\u05D5\u05DC\u05DD\u0021'
rtl_text = u'.' * (len(text))
lv.bidi_process(text,rtl_text,lv.BIDI_DIR.RTL)
print("text: " + text +"\nrtl_text: " + rtl_text)
label.set_text(rtl_text)
lv.scr_load(scr)
```

And this is what I get:
[image]

And I get this on the console:
```
RTL: "�"
LTR: "�"
RTL: "לום עולם!" -> "!�ם�ל�ו�ע� �ם�ו�ל"
text: שלום עולם!
rtl_text: !�ם�ל�ו
```

The original text was "שלום עולם!" which translates to "Hello World!"

- Why was the first character gone? (the `ש`)
I'm not sure yet if that's lvgl-bidi issue or Micropython issue.
- How do I know the expected string length? Can I assume the length would be the same as the original? (that's true if all you do is reverse sub-strings)  
- I've noticed that if I enable compression (remove `--no-compress`) the font looks garbled.
Is compression supported in lvgl?
In not - do you plan supporting it in the future?

embeddedt, 2019-09-27:
> I've noticed that if I enable compression (remove --no-compress) the font looks garbled.
Is compression supported in lvgl?

Compression is supported in `dev-6.1`, but not in `master`.

kisvegabor, 2019-09-27:
@amirgon 
I've updated the algorithm. Now it prints the strings correctly. 
Note the I still don't compose the whole string from the runs. Therefore the output buffer of `lv_bidi_process` contains only the result from the last run.

amirgon, 2019-09-27:
> @amirgon
> I've updated the algorithm. Now it prints the strings correctly.
> Note the I still don't compose the whole string from the runs. Therefore the output buffer of `lv_bidi_process` contains only the result from the last run.

@kisvegabor Please let me know when you compose the whole string, it would make is easier for me to check your algorithm.  

Here are some problems I see now (all conversions are to RTL):
- `u'\u05d0\u05d7\u05ea'` yields: `RTL: "אחת" -> "ת"`
Only one letter instead of the whole word
- `u'\u05d0\u05d7\u05ea \u0031'` yields one run: `RTL: "אחת 1" -> "1 ת חא"`
There's an extra space in the word "אחת".
- `u'\u05d0\u05dc\u05e3 \u0061 \u05d1\u05d9\u05ea'` yields:
```
RTL: "אלף " -> " ף לא"
LTR: "a"
RTL: " בית" -> "ת"
```
- u`\u0030\u002e\u0031` yields `RTL: "0.1" -> "1.0"`
I would expect it to remain "0.1" because "." in that context should behave like a weak character.

kisvegabor, 2019-09-28:
Have you used the latest version? If so please add the full printf-ed output. It should contain the input string too. It'd be easier for me to reproduce because I can't write `u05e9` in C as it's already Unicode and not UTF-8.

I've added support to currencies:
```
lv_bidi_process("שלום $100 עולם!",  buf, LV_BIDI_DIR_RTL);

> Input str: "שלום $100 עולם!"
> RTL: "שלום $100 עולם!" -> "!םלוע $100 םולש"
```

```
lv_bidi_process("שלום 1.0 עולם!",  buf, LV_BIDI_DIR_RTL);

> Input str: "שלום 1.0 עולם!"
> RTL: "שלום 1.0 עולם!" -> "!םלוע 1.0 םולש"
```

It's interesting how the Terminal of Eclipse changes `->` to `<-`. Somehow its autodetects the base dir.

Character replacement (e.g `<` to `>`) still missing too. I try to add it and composing today or at worse case Monday.

kisvegabor, 2019-10-01:
@amirgon 
I've updated the code with
- composing final string
- swap arithmetical symbols (`<` -> `>`)

amirgon, 2019-10-03:
> @amirgon
> I've updated the code with
> 
> * composing final string
> * swap arithmetical symbols (`<` -> `>`)

@kisvegabor I tried it out and it looks good!
I didn't find any problems so far.   

Good job! :100:

kisvegabor, 2019-10-04:
> I tried it out and it looks good!
I didn't find any problems so far.

Awesome, thank you for the feedback! :relaxed: 

I'll continue with the integration to the library:
- LV_LABEL_ALIGN_AUTO
- LV_OBJ_BASE_DIR_LTR/RTL/INHERIT/AUTO

kisvegabor, 2019-10-08:
I've implemented `LV_LABEL_ALIGN_AUTO` and `LV_BIDI_DIR_AUTO/INHERIT`.

However, I'm a little bit puzzled. By default `LV_LABEL_ALIGN_AUTO` is set and `LV_BIDI_BASE_DIR_DEF` sets the default base dir in `lv_conf.h` (AUTO by default).

With labels all clear:
```c
    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_text(label, "hello שלום $200 > $100 עולם! world");
    lv_label_set_long_mode(label, LV_LABEL_LONG_SROLL);
    lv_label_set_body_draw(label, true);
    lv_label_set_style(label, LV_LABEL_STYLE_MAIN, &lv_style_plain_color);
    lv_obj_set_pos(label, 20, 20);
    lv_obj_set_width(label, 300);
    lv_obj_set_base_dir(lv_scr_act(), LV_BIDI_DIR_LTR);
```
[image]

```c
    lv_obj_set_base_dir(lv_scr_act(), LV_BIDI_DIR_RTL);
```
[image]

It becomes more tricky with a ddlist:

```c
    lv_obj_t * ddlist = lv_ddlist_create(lv_scr_act(), NULL);
    lv_ddlist_set_options(ddlist, "1 שלוםםםם\n2 שלום\n3 שלום");
    lv_obj_set_pos(ddlist, 20, 20);
    lv_ddlist_set_fix_width(ddlist, 150);
    lv_obj_set_base_dir(lv_scr_act(), LV_BIDI_DIR_RTL);
```
[image]

Two things:
1. I can't decide if numbers are on the correct end or not. Can you confirm it?
2. The text is forced to be multi-lined with `\n` but it is still treated as a single run. And as it's an RTL run the characters are reversed. So options' order is reversed too. Shall we somehow terminate on `\n`s?

amirgon, 2019-10-08:
@kisvegabor 

> 1. I can't decide if numbers are on the correct end or not. Can you confirm it?

Yes, numbers should appear on the right **on RTL text**.  
For example `"1. שלום \n2. עולם"` should yield:

[image]

If the same text is displayed on LTR, it would yield:
1. שלום 
2. עולם

---

> 2\. The text is forced to be multi-lined with `\n` but it is still treated as a single run. And as it's an RTL run the characters are reversed. So options' order is reversed too. Shall we somehow terminate on `\n`s?

Let's examine the [Unicode Bidi Algorithm specification](https://unicode.org/reports/tr9/) :

On section 2:

> Three types of explicit directional formatting characters are used to modify the standard implicit Unicode Bidirectional Algorithm (UBA). In addition, there are implicit directional formatting characters, the right-to-left and left-to-right marks. **The effects of all of these formatting characters are limited to the current paragraph; thus, they are terminated by a paragraph separator**.

On section 3:

> 3 Basic Display Algorithm
> The Unicode Bidirectional Algorithm (UBA) takes a stream of text as input and proceeds in four main phases:
> - Separation into paragraphs. The rest of the algorithm is applied separately to the text within each paragraph.
...

> **The algorithm reorders text only within a paragraph**; characters in one paragraph have no effect on characters in a different paragraph. Paragraphs are divided by the Paragraph Separator **or appropriate Newline Function** (for guidelines on the handling of CR, LF, and CRLF, see Section 4.4, Directionality, and Section 5.8, Newline Guidelines of [Unicode]).

On section 3.3.1

> P1. Split the text into separate paragraphs. A paragraph separator (type B) is kept with the previous paragraph. Within each paragraph, apply all the other rules of this algorithm.
> P2. In each paragraph, find the first character of type L, AL, or R while skipping over any characters between an isolate initiator and its matching PDI or, if it has no matching PDI, the end of the paragraph.

> Note that:

> Because paragraph separators delimit text in this algorithm, the character found by this rule will generally be the first strong character after a paragraph separator or at the very beginning of the text.

On section 3.3 the first thing that is done is apply rule P1 to split the text into paragraphs, before any other rule.

So - it looks like the first step is to separate a string into paragraphs, where each paragraph is delimited by a [newline character](https://www.unicode.org/versions/Unicode4.0.0/ch05.pdf#G10213) (CR, LF and some others).  
Only after you do that, you can apply the rest of the algorithm, detect runs etc.

[85 more replies]

closed 2019-11-21 by kisvegabor