lvgl/lvgl #900: How to adjust start position for LV_LABEL_LONG_SCROLL?

MichaelVLV, 2019-02-26:
Hi. 

I have some long label, which comes after another label:
[image]

If I set

```C
lv_label_set_long_mode(label, LV_LABEL_LONG_SCROLL);
```

for "List header too long" label, I got overlayed:
[image]

Can I set somehow start position for scrolling animation of  label?

embeddedt, 2019-02-26:
@kisvegabor

kisvegabor, 2019-02-26:
Do you use the `v5.3`?

MichaelVLV, 2019-02-26:
I'm using this version:
https://github.com/littlevgl/pc_simulator_sdl_eclipse

I will try update to latest master and report if issue still occures

MichaelVLV, 2019-02-27:
@kisvegabor I have updated lvgl to latest master (deleted **lvlg** folder and cloned latest master branch in VS project) and results are same, label animation covers wifi label.

kisvegabor, 2019-02-27:
Oh, I got it!

Have look at this: https://docs.littlevgl.com/#Label

`SCROLL` scrolls the whole object ob the parent.  
`ROLL` scrolls only the text on the label

I know it's a little bit misleading. I'm planning to remove `SCROLL` in `v6.0`.

MichaelVLV, 2019-02-28:
@kisvegabor for some reason this setting
```C
lv_label_set_long_mode(label, LV_LABEL_LONG_ROLL);
```
doing nothing - no animations at all.

I have tried all possible options in enum - only _SCROLL_ is working (at least some anumation begins)

```C
static void list_menu_test()
{
	lv_obj_t * scr = lv_obj_create(NULL, NULL);
	lv_scr_load(scr);

	lv_obj_t * label_wifi = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_text(label_wifi, SYMBOL_WIFI);
	lv_obj_align(label_wifi, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 0);

	lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_text(label, "List header too long");
	lv_obj_align(label, NULL, LV_ALIGN_IN_TOP_LEFT, 25, 0); // removed dependence from wifi_label
	lv_obj_set_size(label, 120, 20); // set at least some size
	lv_label_set_long_mode(label, LV_LABEL_LONG_SCROLL); // animation over whole screen
	//lv_label_set_long_mode(label, LV_LABEL_LONG_EXPAND); // nothing
	//lv_label_set_long_mode(label, LV_LABEL_LONG_BREAK); // nothing
	//lv_label_set_long_mode(label, LV_LABEL_LONG_DOT); // nothing
	//lv_label_set_long_mode(label, LV_LABEL_LONG_ROLL); // nothing
	//lv_label_set_long_mode(label, LV_LABEL_LONG_CROP); // nothing
}
```

kisvegabor, 2019-02-28:
It's because after setting `ROLL` you have to set a smaller width to the label. It will be aminated on this smaller area.

MichaelVLV, 2019-02-28:
Oh, now this is working :)

Really, problem was in order. With _ROLL_ its mandatory to set size **after** enabling mode
```C
. . .
	lv_label_set_long_mode(label, LV_LABEL_LONG_ROLL); 
	lv_obj_set_size(label, 50, 20); 
. . . 
```

@kisvegabor 
May I suggest to make note about this in documentation?
As for me, this nuance is not obvious.

kisvegabor, 2019-02-28:
Great! :)

It's stated here: https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_label.h#L139

Mentioning it in the documentation would be really useful. Can you add a PR? 
https://github.com/littlevgl/docs

kisvegabor, 2019-03-03:
Meanwhile, I update the documentation: https://github.com/littlevgl/docs/commit/3a20dc72f23220135da35824f00f7ecd72c5796c#diff-5a17fda50367924e1aaee1b56a4e3713L10

closed 2019-03-03 by kisvegabor

MichaelVLV, 2019-03-04:
Sorry not be able to get it earlier and thanks for update !

kisvegabor, 2019-03-04:
No problem :)