lvgl/lvgl #901: Button matrix, message box: new API proposal

manison, 2019-02-26:
### Button matrix
**New API**
`void lv_btnm_set_pressed(uint16_t btn_id)`
There is `lv_btnm_get_pressed` function but not the setter.

### Message box
**New API**
`lv_obj_t * lv_mbox_get_btnm(lv_obj_t *mbox)`
Returns button matrix object contained in the message box.

### Motivation
To be able to set the default button in message box. E.g. in Yes/No message box I want the 'No' button to be the default, not the first ('Yes') one.

```c
static const char *buttons[] = {"Yes", "No", ""};
lv_mbox_add_btns(mbox, buttons, action);
lv_obj_t *btnm = lv_mbox_get_btnm(mbox);
lv_btnm_set_pressed(btnm, 1); /*Preselect the 'No' button*/
```

I'm going to implement these APIs if there are no objections.

embeddedt, 2019-02-26:
@manison I don't see any issue with this feature. Go ahead! :)

kisvegabor, 2019-02-27:
@manison 
There was a big update on `lv_btnm` here: https://github.com/littlevgl/lvgl/pull/825/
Please be sure to use the last `dev-6.0` branch to avoid conflicts.

The function is really missing, so waiting for the PR :)

puzrin, 2019-02-28:
Could anyone explain, how button matrix fit into Material UI guideline?

Is it a some kind of local grid layout? What is need of separate goupping for buttons, instead of generic one?

embeddedt, 2019-02-28:
@puzrin The button matrix was created to consume less memory for large layouts. If you have a keyboard, you can create one button matrix object instead of creating 30+ buttons. By using styles, there is no visible difference between button matrixes and buttons themselves.

puzrin, 2019-02-28:
Ah, ok. Seems "local hack" with no potential problems for future migrations to UI guideline. Thanks for explanations.

manison, 2019-03-04:
PR #917 merged, closing.

closed 2019-03-04 by manison

kisvegabor, 2019-03-04:
Great, thank you!