lvgl/lvgl #902: lv_ list roll mode: switch last>top and opposite (top>last)

MichaelVLV, 2019-02-26:
Hi.

Is it possible to make automatic switchin in **lv_list** to top position when last is detected?
When press UP, last position should be selected:

[image]

I found similar issue for lv_roller object https://github.com/littlevgl/lvgl/issues/511 and tried add "fake"
two list items on top and bottom, but it looks pretty weird in list, even with small object sizes:

[image]
[image]
[image]

MichaelVLV, 2019-02-26:
I have found reason of huge blank spaces, need to disable buffer completely, not just set it clear:

```C
list_btn = lv_list_add(list, NULL, buf, sel_action);
```

to

```C
list_btn = lv_list_add(list, NULL, NULL, sel_action);
```

embeddedt, 2019-02-26:
@MichaelVLV At the moment, as you discovered in #511, it's not currently supported.

As this looks like the same issue I think it would be better to continue the discussion there.

I'll also ping @kisvegabor; maybe he has an idea.

kisvegabor, 2019-02-26:
@embeddedt 
It's not the same as the roller issue. For the roller, it'd be very difficult to implement infinite scrolling. For the list, it's quite simple.

It can be added here: https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_list.c#L817

IF the next/prev element is `NULL` select the first/last one instead.

MichaelVLV, 2019-02-27:
Thanks for help, I have made little modification and its working :)

closed 2019-02-27 by MichaelVLV