lvgl/lvgl #903: Cannot compile tutorial - undefined reference

hardfire, 2019-02-26:
Hello,
First of all thank you for the awesome project and I hope to help/support in the future (once I start understanding the core :D )

I have worked with C in the past but its been a while. I tried to compile the code in the tutorial at https://blog.littlevgl.com/2018-01-03/linux_fb . Instead of using a IDE, i am using GCC to compile the project. I compile via 
`gcc -I. main.c`
and get the following issues

```
/tmp/ccuZ3Kt8.o: In function `main':
main.c:(.text+0x18): undefined reference to `lv_init'
main.c:(.text+0x1d): undefined reference to `fbdev_init'
main.c:(.text+0x29): undefined reference to `lv_disp_drv_init'
main.c:(.text+0x30): undefined reference to `fbdev_flush'
main.c:(.text+0x40): undefined reference to `lv_disp_drv_register'
main.c:(.text+0x45): undefined reference to `lv_scr_act'
main.c:(.text+0x52): undefined reference to `lv_label_create'
main.c:(.text+0x69): undefined reference to `lv_label_set_text'
main.c:(.text+0x8a): undefined reference to `lv_obj_align'
main.c:(.text+0x94): undefined reference to `lv_tick_inc'
main.c:(.text+0x99): undefined reference to `lv_task_handler'
collect2: error: ld returned 1 exit status
```

I am pretty sure, there is something silly that i am missing and could use some help with this.

I have the following files in the directory
```
$ls
lv_conf.h  lv_drivers  lv_drv_conf.h  lv_examples  lvgl  main.c 
```

Here is the output of gcc in verbose mode

```
/home/avk/Programs/test/framebuffer gcc -I. main.c -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.3.0-27ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.3.0 (Ubuntu 7.3.0-27ubuntu1~18.04) 
COLLECT_GCC_OPTIONS='-I' '.' '-v' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-linux-gnu/7/cc1 -quiet -v -I . -imultiarch x86_64-linux-gnu main.c -quiet -dumpbase main.c -mtune=generic -march=x86-64 -auxbase main -version -fstack-protector-strong -Wformat -Wformat-security -o /tmp/cc6h8IVu.s
GNU C11 (Ubuntu 7.3.0-27ubuntu1~18.04) version 7.3.0 (x86_64-linux-gnu)
	compiled by GNU C version 7.3.0, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version isl-0.19-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/usr/local/include/x86_64-linux-gnu"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/7/../../../../x86_64-linux-gnu/include"
ignoring duplicate directory "."
  as it is a non-system directory that duplicates a system directory
#include "..." search starts here:
#include <...> search starts here:
 /home/avk/Programs/test/framebuffer/
 /usr/lib/gcc/x86_64-linux-gnu/7/include
 /usr/local/include
 /usr/lib/gcc/x86_64-linux-gnu/7/include-fixed
 /usr/include/x86_64-linux-gnu
 /usr/include
End of search list.
GNU C11 (Ubuntu 7.3.0-27ubuntu1~18.04) version 7.3.0 (x86_64-linux-gnu)
	compiled by GNU C version 7.3.0, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version isl-0.19-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: c8081a99abb72bbfd9129549110a350c
COLLECT_GCC_OPTIONS='-I' '.' '-v' '-mtune=generic' '-march=x86-64'
 as -v -I . --64 -o /tmp/ccFnWwQ4.o /tmp/cc6h8IVu.s
GNU assembler version 2.30 (x86_64-linux-gnu) using BFD version (GNU Binutils for Ubuntu) 2.30
COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/7/:/usr/lib/gcc/x86_64-linux-gnu/7/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/7/:/usr/lib/gcc/x86_64-linux-gnu/
LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/7/:/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/7/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-I' '.' '-v' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-linux-gnu/7/collect2 -plugin /usr/lib/gcc/x86_64-linux-gnu/7/liblto_plugin.so -plugin-opt=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper -plugin-opt=-fresolution=/tmp/cctwUjME.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --sysroot=/ --build-id --eh-frame-hdr -m elf_x86_64 --hash-style=gnu --as-needed -dynamic-linker /lib64/ld-linux-x86-64.so.2 -pie -z now -z relro /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/Scrt1.o /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/7/crtbeginS.o -L/usr/lib/gcc/x86_64-linux-gnu/7 -L/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib -L/lib/x86_64-linux-gnu -L/lib/../lib -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib -L/usr/lib/gcc/x86_64-linux-gnu/7/../../.. /tmp/ccFnWwQ4.o -lgcc --push-state --as-needed -lgcc_s --pop-state -lc -lgcc --push-state --as-needed -lgcc_s --pop-state /usr/lib/gcc/x86_64-linux-gnu/7/crtendS.o /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/crtn.o
/tmp/ccFnWwQ4.o: In function `main':
main.c:(.text+0x18): undefined reference to `lv_init'
main.c:(.text+0x1d): undefined reference to `fbdev_init'
main.c:(.text+0x29): undefined reference to `lv_disp_drv_init'
main.c:(.text+0x30): undefined reference to `fbdev_flush'
main.c:(.text+0x40): undefined reference to `lv_disp_drv_register'
main.c:(.text+0x45): undefined reference to `lv_scr_act'
main.c:(.text+0x52): undefined reference to `lv_label_create'
main.c:(.text+0x69): undefined reference to `lv_label_set_text'
main.c:(.text+0x8a): undefined reference to `lv_obj_align'
main.c:(.text+0x94): undefined reference to `lv_tick_inc'
main.c:(.text+0x99): undefined reference to `lv_task_handler'
collect2: error: ld returned 1 exit status
/home/avk/Programs/test/framebuffer 

```

PS: I am on ubuntu and plan to run this on a PC terminal (ctrl + alt + F3 or such) as mentioned in the tutorial.

embeddedt, 2019-02-26:
@hardfire If you're on Ubuntu it would be much easier for you to start with the PC simulator: https://github.com/littlevgl/pc_simulator_sdl_eclipse. You can build it with `make`.

Your problem is that you haven't compiled and linked LittlevGL itself. You only built `main.c`.

kisvegabor, 2019-02-27:
Just to be suere: have you downloaded the submodules too? Are there files in the `lvgl` folder?

hardfire, 2019-02-27:
@kisvegabor yes, the lvgl folder is a clone of https://github.com/littlevgl/lvgl and lv_drivers of https://github.com/littlevgl/lv_drivers
Taking a tip from @embeddedt suggestion I tried to use the following Makefile to build the project - https://github.com/littlevgl/pc_simulator_sdl_eclipse/blob/master/Makefile to build the project and it seems to work now. 
I can see Hello World in the framebuffer.

kisvegabor, 2019-02-27:
Great!

closed 2019-02-27 by kisvegabor

raphael10-collab, 2023-12-27:
> @hardfire If you're on Ubuntu it would be much easier for you to start with the PC simulator: https://github.com/littlevgl/pc_simulator_sdl_eclipse. You can build it with `make`.
> 
> Your problem is that you haven't compiled and linked LittlevGL itself. You only built `main.c`.

Would you be so in explaining me how to compile and build for Ubuntu ?

I git cloned the branch "release/8.3" with all its submodules, and compiled it:
```
raphy@raohy:~$ git clone  -b release/v8.3  --recurse-submodules https://github.com/lvgl/lvgl.git

raphy@raohy:~/lvgl$ cmake -B build

raphy@raohy:~/lvgl$ cmake --build build
[100%] Building C object CMakeFiles/lvgl_demos.dir/demos/widgets/lv_demo_widgets.c.o
In file included from /home/raphy/lvgl/demos/widgets/../../src/misc/lv_log.h:16,
                 from /home/raphy/lvgl/demos/widgets/../../lvgl.h:25,
                 from /home/raphy/lvgl/demos/widgets/../lv_demos.h:16,
                 from /home/raphy/lvgl/demos/widgets/lv_demo_widgets.h:16,
                 from /home/raphy/lvgl/demos/widgets/lv_demo_widgets.c:9:
/home/raphy/lvgl/demos/widgets/../../src/misc/../lv_conf_internal.h:46:17: note: ‘#pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors’
   46 |         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
      |                 ^~~~~~~
[100%] Linking C static library liblvgl_demos.a
[100%] Built target lvgl_demos
``` 

I did the same with the LVGL Drivers:

```
raphy@raohy:~$ git clone  -b release/v8.3  --recurse-submodules https://github.com/lvgl/lv_drivers.git

raphy@raohy:~/lv_drivers$ cmake -B build
```
But I'm getting this error:

```
raphy@raohy:~/lv_drivers$ cmake --build build
[  4%] Building C object CMakeFiles/lv_drivers.dir/CMakeFiles/3.28.1/CompilerIdC/CMakeCCompilerId.c.o
[  8%] Building C object CMakeFiles/lv_drivers.dir/display/GC9A01.c.o
In file included from /home/raphy/lv_drivers/display/GC9A01.c:10:
/home/raphy/lv_drivers/display/GC9A01.h:20:10: fatal error: ../../lv_drv_conf.h: No such file or directory
   20 | #include "../../lv_drv_conf.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/lv_drivers.dir/build.make:90: CMakeFiles/lv_drivers.dir/display/GC9A01.c.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/lv_drivers.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
raphy@raohy:~/lv_drivers$ cd

```

raphael10-collab, 2023-12-27:
> @hardfire If you're on Ubuntu it would be much easier for you to start with the PC simulator: https://github.com/littlevgl/pc_simulator_sdl_eclipse. You can build it with `make`.
> 
> Your problem is that you haven't compiled and linked LittlevGL itself. You only built `main.c`.

@kisvegabor 

Would you be so in explaining me how to compile and build for Ubuntu ?

    I git cloned the branch "release/8.3" with all its submodules, and compiled it:
    ```
    raphy@raohy:~$ git clone  -b release/v8.3  --recurse-submodules https://github.com/lvgl/lvgl.git
    
    raphy@raohy:~/lvgl$ cmake -B build
    
    raphy@raohy:~/lvgl$ cmake --build build
    [100%] Building C object CMakeFiles/lvgl_demos.dir/demos/widgets/lv_demo_widgets.c.o
    In file included from /home/raphy/lvgl/demos/widgets/../../src/misc/lv_log.h:16,
                     from /home/raphy/lvgl/demos/widgets/../../lvgl.h:25,
                     from /home/raphy/lvgl/demos/widgets/../lv_demos.h:16,
                     from /home/raphy/lvgl/demos/widgets/lv_demo_widgets.h:16,
                     from /home/raphy/lvgl/demos/widgets/lv_demo_widgets.c:9:
    /home/raphy/lvgl/demos/widgets/../../src/misc/../lv_conf_internal.h:46:17: note: ‘#pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors’
       46 |         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
          |                 ^~~~~~~
    [100%] Linking C static library liblvgl_demos.a
    [100%] Built target lvgl_demos
    ``` 
    
    I did the same with the LVGL Drivers but got `fatal error: ../../lv_drv_conf.h: No such file or directory`  : 
    
    raphy@raohy:~/lvgl$  git clone  -b release/v8.3  --recurse-submodules https://github.com/lvgl/lv_drivers.git
    Cloning into 'lv_drivers'...
    remote: Enumerating objects: 2042, done.
    remote: Counting objects: 100% (717/717), done.
    remote: Compressing objects: 100% (160/160), done.
    remote: Total 2042 (delta 628), reused 582 (delta 556), pack-reused 1325
    Receiving objects: 100% (2042/2042), 560.56 KiB | 5.78 MiB/s, done.
    Resolving deltas: 100% (1344/1344), done.
    raphy@raohy:~/lvgl$ cd lv_drivers/
    raphy@raohy:~/lvgl/lv_drivers$ cmake -B build
    -- The C compiler identification is GNU 12.3.0
    -- The CXX compiler identification is GNU 12.3.0
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Check for working C compiler: /usr/bin/cc - skipped
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Check for working CXX compiler: /usr/bin/c++ - skipped
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    -- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
    -- Checking for modules 'wayland-client;wayland-cursor;wayland-protocols;xkbcommon'
    --   Found wayland-client, version 1.21.0
    --   Found wayland-cursor, version 1.21.0
    --   Found wayland-protocols, version 1.31
    --   Found xkbcommon, version 1.5.0
    -- Configuring done (0.4s)
    -- Generating done (0.0s)
    -- Build files have been written to: /home/raphy/lvgl/lv_drivers/build
    raphy@raohy:~/lvgl/lv_drivers$ cmake --build build
    [  4%] Building C object CMakeFiles/lv_drivers.dir/CMakeFiles/3.28.1/CompilerIdC/CMakeCCompilerId.c.o
    [  8%] Building C object CMakeFiles/lv_drivers.dir/display/GC9A01.c.o
    In file included from /home/raphy/lvgl/lv_drivers/display/GC9A01.c:10:
    /home/raphy/lvgl/lv_drivers/display/GC9A01.h:20:10: fatal error: ../../lv_drv_conf.h: No such file or directory
       20 | #include "../../lv_drv_conf.h"
          |          ^~~~~~~~~~~~~~~~~~~~~
    compilation terminated.
    gmake[2]: *** [CMakeFiles/lv_drivers.dir/build.make:90: CMakeFiles/lv_drivers.dir/display/GC9A01.c.o] Error 1
    gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/lv_drivers.dir/all] Error 2
    gmake: *** [Makefile:136: all] Error 2

kisvegabor, 2023-12-30:
Hi,

I suggest using this project: https://github.com/lvgl/lv_port_pc_eclipse