lvgl/lvgl #904: Too many objects bug

IvanShugov, 2019-02-26:
Hi! I'm working on a project including GUI and as far as I see this one is really good! Congratulations for your work, it is awesome! 
Well, I experience some problems especially when there are too many objects. To demonstrate the problem I will post an example code in two frame resolutions - 128x160 and 240x320, with digital input - keyboard, no touchscreen. To do this I simply change the frame buffer size and it is actually very well displayed when there are small count of objects. But when the objects are too many some of them disappeared, until focused. Anyway, there are some "ghosts" - you can see it in the labels, this seems to be default values of the rollers. But I have set them with new values, and actually they appear at the right place, even if the second one is visible only when the roller is selected.

This kind of problem exist even when the buffer is set to match actual screen resolution. If needed I will provide another example code for this case. Mostly the problems are invisible object until selected or ghost effects, usually with default values for the object.

I'm using version 5.3 and esp32. The custom styles include mostly transparent body (for timer/power/win panels and the border is invisible, just used as cage for a label and roller), small font, etc.

The screen is empty by default, the theme is "alien", color set to 210.
The roller objects height is set with lv_obj_set_height() to a value a bit higher than 1 row count, just to allow the change of the border to be visible when the object is focused. This is another idea to implement if possible. Else the roller set to 1 row count is focused without change.

Can you, please help a little - how to setup many objects aligned in this manner for example at small area correctly?

static lv_obj_t * top_panel;
static lv_obj_t * frame_panel;

static lv_obj_t * top_label;

static lv_obj_t * timer_panel;
static lv_obj_t * power_panel;
static lv_obj_t * win_panel;

static lv_obj_t * timer_label;
static lv_obj_t * power_label;
static lv_obj_t * win_label;

static lv_obj_t * timer_roller;
static lv_obj_t * power_roller;
static lv_obj_t * win_roller;

void example_construct (lv_obj_t * scr, lv_theme_t * th)
{
	// load current screen
	lv_scr_load(scr);

	// set theme
	lv_theme_set_current(th);

	// create panels
	top_panel = lv_obj_create(lv_scr_act(), NULL);
	lv_obj_set_style(top_panel, &style_header_panel);
	lv_obj_set_size(top_panel, LV_HOR_RES, (LV_VER_RES * 0.20));
	lv_obj_align(top_panel, NULL, LV_ALIGN_IN_TOP_MID, 0, 0);

	frame_panel = lv_obj_create(lv_scr_act(), NULL);
	lv_obj_set_style(frame_panel, &style_frame_panel);
	lv_obj_set_size(frame_panel, LV_HOR_RES, (LV_VER_RES * 0.80));
	lv_obj_align(frame_panel, top_panel, LV_ALIGN_OUT_BOTTOM_MID, 0, 0);

	timer_panel = lv_obj_create(frame_panel, NULL);
	lv_obj_set_style(timer_panel, &style_trans_panel);
	lv_obj_set_size(timer_panel, (LV_HOR_RES), (LV_VER_RES * 0.20));
	lv_obj_align(timer_panel, top_panel, LV_ALIGN_OUT_BOTTOM_MID, 0, 0);

	power_panel = lv_obj_create(frame_panel, NULL);
	lv_obj_set_style(power_panel, &style_trans_panel);
	lv_obj_set_size(power_panel, (LV_HOR_RES), (LV_VER_RES * 0.20));
	lv_obj_align(power_panel, timer_panel, LV_ALIGN_OUT_BOTTOM_MID, 0, 0);

	win_panel = lv_obj_create(frame_panel, NULL);
	lv_obj_set_style(win_panel, &style_trans_panel);
	lv_obj_set_size(win_panel, (LV_HOR_RES), (LV_VER_RES * 0.20));
	lv_obj_align(win_panel, power_panel, LV_ALIGN_OUT_BOTTOM_MID, 0, 0);

	adj_panel = lv_obj_create(frame_panel, NULL);
	lv_obj_set_style(adj_panel, &style_trans_panel);
	lv_obj_set_size(adj_panel, (LV_HOR_RES), (LV_VER_RES * 0.20));
	lv_obj_align(adj_panel, win_panel, LV_ALIGN_OUT_BOTTOM_MID, 0, 0);

	// Header
	top_label = lv_label_create(top_panel, NULL);
	lv_obj_set_style(top_label, &style_info_text);
	lv_label_set_align(top_label, LV_LABEL_ALIGN_LEFT);
	lv_label_set_long_mode(top_label, LV_LABEL_LONG_EXPAND);
	lv_label_set_text(top_label, "Settings");
	lv_obj_align(top_label, NULL, LV_ALIGN_IN_LEFT_MID, 5, 0);

	// Timer 
	timer_label = lv_label_create(timer_panel, NULL);
	lv_obj_set_style(timer_label, &style_info_text);
	lv_label_set_align(timer_label, LV_LABEL_ALIGN_CENTER);
	lv_label_set_long_mode(timer_label, LV_LABEL_LONG_EXPAND);
	lv_label_set_text(timer_label, "Timer");
	lv_obj_align(timer_label, NULL, LV_ALIGN_IN_LEFT_MID, 5, 0);

	timer_roller = lv_roller_create(timer_panel, NULL);
	lv_roller_set_options(timer_roller, "On\nOff");
	lv_roller_set_visible_row_count(timer_roller, 1);
	lv_obj_set_free_num(timer_roller, 0);
	lv_roller_set_hor_fit(timer_roller, false);
	lv_obj_set_height(timer_roller, 16);
	lv_obj_align(timer_roller, NULL, LV_ALIGN_IN_RIGHT_MID, -5, 0);
	lv_roller_set_action(timer_roller, (lv_action_t) roller_action);

	// Power 
	power_label = lv_label_create(power_panel, NULL);
	lv_obj_set_style(power_label, &style_info_text);
	lv_label_set_align(power_label, LV_LABEL_ALIGN_CENTER);
	lv_label_set_long_mode(power_label, LV_LABEL_LONG_EXPAND);
	lv_label_set_text(power_label, "Power");
	lv_obj_align(power_label, NULL, LV_ALIGN_IN_LEFT_MID, 5, 0);

	power_roller = lv_roller_create(power_panel, NULL);
	lv_roller_set_options(power_roller, "1\n2\n3");
	lv_roller_set_visible_row_count(power_roller, 1);
	lv_obj_set_free_num(power_roller, 1);
	lv_roller_set_hor_fit(power_roller, false);
	lv_obj_set_height(power_roller, 16);
	lv_obj_align(power_roller, NULL, LV_ALIGN_IN_RIGHT_MID, -5, 0);
	lv_roller_set_action(power_roller, (lv_action_t) roller_action);

	// Window 
	win_label = lv_label_create(win_panel, NULL);
	lv_obj_set_style(win_label, &style_info_text);
	lv_label_set_align(win_label, LV_LABEL_ALIGN_CENTER);
	lv_label_set_long_mode(win_label, LV_LABEL_LONG_EXPAND);
	lv_label_set_text(win_label, "Window");
	lv_obj_align(win_label, NULL, LV_ALIGN_IN_LEFT_MID, 5, 0);

	win_roller = lv_roller_create(win_panel, NULL);
	lv_roller_set_options(win_roller, "On\nOff");
	lv_roller_set_visible_row_count(win_roller, 1);
	lv_obj_set_free_num(win_roller, 2);
	lv_roller_set_hor_fit(win_roller, false);
	lv_obj_set_height(win_roller, 16);
	lv_obj_align(win_roller, NULL, LV_ALIGN_IN_RIGHT_MID, -5, 0);
	lv_roller_set_action(win_roller, (lv_action_t) roller_action);
}

[image]

[image]

[image]

embeddedt, 2019-02-26:
@IvanShugov I think it's an issue with your display driver. Can you post your `disp_flush` function?

IvanShugov, 2019-02-26:
Well, the littlevgl library is part of esp_iot_solutions, the driver is ILI9341_adapter.cpp.
CONFIG_LVGL_LCD_DRIVER_FRAMEBUFFER_MODE is set and LV_VDB_SIZE != 0, so the display_flush should be:

void ex_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_p)
{
    lcd_obj->drawBitmap((int16_t)x1, (int16_t)y1, (const uint16_t *)color_p, (int16_t)(x2 - x1 + 1), (int16_t)(y2 - y1 + 1));
    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
    lv_flush_ready();
}

embeddedt, 2019-02-26:
@IvanShugov Can you try making this layout on the PC simulator and see if it works properly? If it does then the problem must be within the `drawBitmap` function.

IvanShugov, 2019-02-26:
Ok, I haven't tested it in simulator, so I will install it and try. Thank you for the guidelines!

As far as I see changing the factor in this line
#define LV_VDB_SIZE         (LV_VER_RES * LV_HOR_RES / 10) 
gives me different image every time, if this helps. Even the full size buffer is not working correctly.

IvanShugov, 2019-02-26:
@embeddedt the code is working pretty good in the simulator. I will try to trace drawBitmap function. If you have some idea, please share it. Thank you for the help so far, I really appreciate it.
Anyway, there should be something wrong with the esp-iot port of the library, I guess.

[image]

embeddedt, 2019-02-26:
@IvanShugov Check coordinates, framebuffer addresses, and width/height values in the driver very carefully. Many issues stem from one of those three areas.

IvanShugov, 2019-02-26:
Something is definitely wrong with the framebuffer mode. If I disable it all objects are positioned correctly. Well, ugly edges, tearing, etc, but where it is expected to be.

kisvegabor, 2019-02-27:
Try to fill an array with a color (`lv_color_t test[1000]`) and put it to various places with `lcd_obj->drawBitmap` to see where could be the issue.

IvanShugov, 2019-02-27:
@kisvegabor I tried your proposal with the following code, with enabled frame buffer:

.....
	uint16_t i;
	lv_color_t test_color[160];
	lv_color_t test_clear[160];
	for(i = 0; i < 160; i++)
	{
		test_clear[i] = LV_COLOR_WHITE;
		test_color[i] = LV_COLOR_RED;
	}

	for(i = 0; i < 128; i++)
	{
		if(i > 0)
		{
			board_lcd_blit_area(0, i-1, (uint16_t *) test_clear, 160, 1);
			lv_flush_ready();
		}
		board_lcd_blit_area(0, i, (uint16_t *) test_color, 160, 1);
		lv_flush_ready();
		vTaskDelay(2000 / portTICK_RATE_MS);
	}

	for(i = 0; i < 160; i++)
	{
		if(i > 0)
		{
			board_lcd_blit_area(i-1, 0, (uint16_t *) test_clear, 1, 128);
			lv_flush_ready();
		}

		board_lcd_blit_area(i, 0, (uint16_t *) test_color, 1, 128);
		lv_flush_ready();

		vTaskDelay(2000 / portTICK_RATE_MS);
	}
.....

void board_lcd_blit_area(int16_t x, int16_t y, uint16_t *bitmap, int16_t w, int16_t h)
{
    lcd_obj->drawBitmap(x, y, bitmap, w, h);
}

And the result is exactly as expected - horizontal line moving from position 0 to position 127 followed by vertical line from position 0 to 159 without any artefacts.
Here comes the weird part: if the coordinates are set in some other format the color of the line changes to BLUE, even if the only color I set is RED. For example:
board_lcd_blit_area(75, i, (uint16_t *) test_color, 40, 1); // RED line - OK
board_lcd_blit_area(75, i, (uint16_t *) test_color, 30, 1); // BLUE line - ???
board_lcd_blit_area(75, i, (uint16_t *) test_color, 20, 1); // BLUE line - ???
board_lcd_blit_area(35, i, (uint16_t *) test_color, 20, 1); // BLUE line - ???
board_lcd_blit_area(5, i, (uint16_t *) test_color, 20, 1); // BLUE line - ???
board_lcd_blit_area(5, i, (uint16_t *) test_color, 40, 1); // RED line - OK

But if the line is thicker than 1 pixel the color is RED as set above, for example:
board_lcd_blit_area(5, i, (uint16_t *) test_color, 20, 2);  // RED line

This is quite strange..

embeddedt, 2019-02-27:
@IvanShugov You'll need to figure out what is wrong with the `drawBitmap` call, I'm 99% sure it's the culprit.

IvanShugov, 2019-02-28:
Update: Well, I found something interesting - the communication with the ILI9341 is over SPI. So the default clock speed was 40MHz. When I set it at 20 MHz at the example screen above everything goes at the expected place, no "ghosts". But the test code with the line still failed at the same manner (RED is BLUE), and the more complex screens also failed. So I set the speed at 5MHz. And.. magic!.. so far every screen I have tested seems to be ok. 
There is something I haven't researched yet - what is the connection between constructing the content of the frame buffer, flushing the frame buffer and the actual interface speed w/o DMA. What can goes wrong here?

embeddedt, 2019-02-28:
@IvanShugov Check your display's maximum SPI clock speed. It sounds like you were sending too much data at a time.

MichaelVLV, 2019-02-28:
@IvanShugov 

not all ESP32 pins supports high speed:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#gpio-matrix-and-iomux

If it's possible, suggest to check real clock freq with oscilloscope or with logic analyzer

puzrin, 2019-02-28:
Looks like synchronization fail on hardware bus.

- Check display power first. From stabilizer to display pins. Wires should be "fat enougth", probably with ceramic capacitors on both ends. Bad power is source of huge count of problems, with absolutely random effects. It's useless to continue until sure your power is ok.
- Check signal fronts with oscilloscope. Also make sure SPI 40MHz attached directly to appropriate esp32 pins, without remapping.
- Don't use 40MHz without need (calculate how much do you need for 25-50 FPS, according to screen size and raw data volume)

IvanShugov, 2019-02-28:
@embeddedt It seems to be 100ns, i.e. 10MHz max frequency, so yes 40MHz is far over the limits, you are absolutely right!
@MichaelVLV, I will be able to check the real speed next week, just no access to oscilloscope right now. But I will, when possible. 
@puzrin Thanks for the advices, I will pass them to the hardware team. Right now I'm using esp-wrover-kit and I really hope they have read the same source as you :) 

The only thing so far which is wrong is the RED<->BLUE line swap... I have no problems with the objects, but this color swap.. hm.. strange, it exist even at 5Mhz.

embeddedt, 2019-02-28:
> The only thing so far which is wrong is the RED<->BLUE line swap... I have no problems with the objects, but this color swap.. hm.. strange, it exist even at 5Mhz.

It might be a data alignment issue... check that you're sending the right number of bytes at a time.

kisvegabor, 2019-03-01:
@IvanShugov Try `LV_COLOR_16_SWAP  1` in `lv_conf.h`

kisvegabor, 2019-03-02:
@IvanShugov 
In the past few days [LittlevGL's Forum](https://forum.littlevgl.com) has launched to discuss questions there. See this [Announcment](https://forum.littlevgl.com/viewtopic.php?f=3&t=5).

If the issue still persists please open a topic for it on the Forum.

closed 2019-03-02 by kisvegabor