lvgl/lvgl #905: Documentation unclear around intervals for calling lv_task_handler()

fovea1959, 2019-02-26:
Documentation says:

> Call lv_tick_inc(x) every x milliseconds in a Timer or Task (x should be between 1 and 10)
> ...
> Call lv_task_handler() periodically every few milliseconds in the main while(1) loop, in Timer interrupt or in an Operation system task.

Should lv_task_handler() be called more or less often than lv_tick_inc()? Can I call them in succession? Why do I have to call both (what is the difference)?

embeddedt, 2019-02-26:
@fovea1959 You **must** call both. `lv_task_handler` should be called roughly every 10 milliseconds, but its timing is not critical.

`lv_tick_inc` must be consistently called at an interval between 1 and 10 milliseconds. You have to put it in a timer or separate OS task; it should not be in the main `while(1)` loop with `lv_task_handler`.

kisvegabor, 2019-03-03:
I updated the README to make it clearer.

closed 2019-03-03 by kisvegabor