lvgl/lvgl #906: The Forum has launched

kisvegabor, 2019-02-27:
The forum has opened to separate user questions from development. It should be the place where you can ask about
- getting started with your board/MCU
- PC simulator
- how-to solve something
- something is not working

GitHub issues should be about
- bugs
- new features

Before reporting a bug on GitHub be sure it's really a bug and open a topic here first.

puzrin, 2019-02-28:
Create `.github` with issue/pr configs instead of pinned "PLEASE READ". This will be more effective.

embeddedt, 2019-02-28:
@puzrin If @kisvegabor is alright with it I'll do that as soon as I get a chance.

kisvegabor, 2019-02-28:
@embeddedt It would be great! Thank you!

jedie, 2019-03-01:
URL?

kisvegabor, 2019-03-01:
@jedie Added, thank you!

embeddedt, 2019-03-22:
Not stale.

embeddedt, 2019-03-22:
https://github.com/littlevgl/lvgl/issues/975#issuecomment-475720784

kisvegabor, 2019-03-22:
@embeddedt I think we can close this issue because:
1. almost 1 month has elapsed to let people know that we have a forum
2. the issue template will direct people to the forum

closed 2019-03-23 by embeddedt