lvgl/lvgl #907: Using GUI elements' values for back-end tasks

tomtristan, 2019-02-27:
Is it possible to use GUI elements' values to set necessary variables in the back-end code and vice versa?
I will describe my task as an example for easier understanding of my issue.
I would like to have a tab for a list of controller parameters like radiator fan speed, max regeneration torque(their values are drop-down lists of numbers) etc. 
These controller values need to be sent via CAN and also will be saved to on-board external flash memory. Upon booting the PCB, the GUI values are first reset, and then set according to parameter values from flash.
Is there a function/known method to get and set these GUI values, or should I brutely use strtonum function(stdlib) on my_action to get the value from the list?

embeddedt, 2019-02-27:
@tomtristan All widgets allow you to get their value. Some store it as a string, others will give you an integer. Read the [documentation](https://docs.littlevgl.com) and the header files in the `lv_objx` directory for more information.

tomtristan, 2019-02-28:
Thank you for the guidance! Will have a look at the function descriptions and pay more attention while reading the documentation.

closed 2019-02-28 by tomtristan