lvgl/lvgl #908: BUG about 'kb' obj or 'btnm' obj

as564335sa, 2019-02-27:
Hi,
 I found a strange bug about 'kb' obj or 'btnm' obj ?
the code:
```
lv_res_t action(lv_obj_t *obj)
{
	lv_obj_t * kb1 = lv_kb_create(lv_scr_act(), NULL);
	lv_obj_align(kb1, NULL, LV_ALIGN_IN_BOTTOM_MID, 0, 0);
	lv_kb_set_ta(kb1, obj);
	return LV_RES_OK;
}
void lv_test_kb_1(void)
{

    lv_obj_t * ta = lv_ta_create(lv_scr_act(), NULL);
    lv_obj_align(ta, NULL, LV_ALIGN_IN_TOP_MID, 0, 30);
    lv_ta_set_action(ta,action);
    /* Default object*/
    lv_obj_t * kb1 = lv_kb_create(lv_scr_act(), NULL);
    lv_obj_align(kb1, NULL, LV_ALIGN_IN_BOTTOM_MID, 0, 0);
    lv_kb_set_ta(kb1, ta);
}
```
the picture:
[image]
[image]
[image]
[image]

you can see,the normal pictures are the first run of the program, then press the symbol 'X' to colse the 'kb',click the ‘ta’ obj and reload the 'kb'；
Then it was found that the ‘’#‘’ symbol and only the ‘’#‘’ symbol was not displayed, but there was no error in clicking on the button with the '#'

as564335sa, 2019-02-28:
I found it was due to a problem with the `recolor` command.How to fix it？

embeddedt, 2019-02-28:
@as564335sa It's probably because `#` is the character used for recoloring.

@kisvegabor Please comment.

kisvegabor, 2019-02-28:
I examined it and found the issue. The recolor feature was not initialized in `lv_btnm_craete` therefore it got random value (from memory garbage).

Fixed in `master`. It it working now?

as564335sa, 2019-02-28:
Great! It is working now.

closed 2019-02-28 by as564335sa