lvgl/lvgl #909: Change 'lang_id' Type to Accept ISO 639-1

beibean, 2019-02-27:
Hi,

When talking about classifying languages, one very common standard is [ISO 639-1](https://en.wikipedia.org/wiki/List_of_ISO_639-1_codes) which basically assigns two ASCII characters to each language.

So, I suggest to change lang_id and related functions to uint16_t instead of uint8_t:

```C
@file lv_lang.h
void lv_lang_set(uint8_t lang_id);   ---> void lv_lang_set(uint16_t lang_id);
uint8_t lv_lang_act(void);   ---> uint16_t lv_lang_act(void);

@file lv_lang.c
static uint8_t lang_act = 0;   ---> static uint16_t lang_act = 0;
void lv_lang_set(uint8_t lang_id)   ---> void lv_lang_set(uint16_t lang_id)
uint8_t lv_lang_act(void)   ---> uint16_t lv_lang_act(void)
```

If there are no objections I will do this little changes and PR.

embeddedt, 2019-02-27:
@beibean Work is being done on a new language translation system in https://github.com/littlevgl/lv_i18n.

beibean, 2019-02-28:
Oh, I didn't know about this. 

Thanks for sharing this info. ;)

closed 2019-02-28 by beibean