lvgl/lvgl #910: Could system clock speed be responsible for crash after loading first frame?

tomtristan, 2019-02-27:
In the documentation it is said that 16MHz is a requirement for this library to be used. Is this a minimal requirement or the MCU's system's clock (SYSCLK) has to be set at 16MHz? The system clock at the moment is set at 216MHz, could this be the reason why the screen crashes(not the mcu, the infinite while(1) is running, but task handler seems to do no change to the screen).

embeddedt, 2019-02-27:
@tomtristan It's a minimum requirement. I think very few people are actually using MCUs running at 16MHz. I run it on the Discovery board at 216MHz so that can't be the problem. Are you calling `lv_tick_inc` in a separate thread and calling `lv_task_handler` every 1-10 ms?

tomtristan, 2019-02-28:
Thank you so much!!! I had forgot to call 'lv_tick_inc', and now the tab screen is refreshing as it should, it is smooth. Your support is very much appreciated! Slowly beginning to understand how this library works :) Thanks again!

embeddedt, 2019-02-28:
Great!

closed 2019-02-28 by embeddedt