lvgl/lvgl #911: [feature] animated slider on group input

BrianPugh, 2019-02-27:
Currently the `lv_slider` object just snaps to discrete positions on button input. However, it'd be nice if it did this with an animation. Adding the animation functionality is trivial (just change the `lv_slider_set_value` to `lv_slider_set_value_anim`), I'm just making this issue to discuss how to implement the API. Should it just be animated by default, and setting animation time to 0 would result in the current behavior? Should there be a function like `lv_slider_enable_anim` and `lv_slider_disable_anim` to set a flag? I can make a pull request, just curious about how it should be approached.

embeddedt, 2019-02-27:
@BrianPugh I would implement it the same way I animated `lv_sw`. But let's get @kisvegabor's opinion too.

BrianPugh, 2019-02-27:
so that was my initial though, but lv_sw is a bit different since you have to manually toggle the switch in your callback, while in `lv_slider` its automatic.

embeddedt, 2019-02-27:
@BrianPugh Are you sure? It should be automatic with `lv_sw` as well.

BrianPugh, 2019-02-27:
You're right; I'll make a pull request for it to behave similar to a `lv_sw`. I think it's a good solution and it's good to be consistent.

kisvegabor, 2019-02-28:
@BrianPugh Good idea to animate the slider if used via keyboard. A similar interface to `lv_sw` seems good to me:
- `lv_slider_set_anim_time`.
- `lv_slider_set_value(slider, 23)`
- `lv_slider_set_value_anim(slider, 23)`  (no anim time specified here)

 `lv_sw` can use `lv_slider`'s `anim_time` attribute.

BrianPugh, 2019-03-01:
closing to continue conversation in #912

closed 2019-03-01 by BrianPugh