lvgl/lvgl #913: [dev-6.0] SIGSEGV with LV_FIT_FILL

embeddedt, 2019-02-28:
I've noticed that the `lv_test_theme_1` layout changed after the `lv_fit_t` system was added.

[image]

I tried to solve it by changing [this line](https://github.com/littlevgl/lv_examples/blob/aa8528597489eb1f5d9c27085c5d6c8a3bb7df3d/lv_tests/lv_test_theme/lv_test_theme_1.c#L81) from `LV_FIT_TIGHT` to `LV_FIT_FILL` but this results in LittlevGL crashing with SIGSEGV.

The problem appears to be recursion. Backtrace is below:

```
#0  0x000000000044b379 in lv_obj_invalidate (obj=0x74d6f8 <work_mem_int+7288>) at ../lvgl/lv_core/lv_obj.c:415
#1  0x000000000044b6ab in lv_obj_set_pos (obj=0x74d6f8 <work_mem_int+7288>, x=41, y=92) at ../lvgl/lv_core/lv_obj.c:524
#2  0x000000000044c003 in lv_obj_align (obj=0x74d6f8 <work_mem_int+7288>, base=0x74d658 <work_mem_int+7128>, align=2 '\002', x_mod=0, 
    y_mod=16) at ../lvgl/lv_core/lv_obj.c:772
#3  0x0000000000417c11 in lv_cont_layout_col (cont=0x74d658 <work_mem_int+7128>) at ../lvgl/lv_objx/lv_cont.c:366
#4  0x0000000000417acc in lv_cont_refr_layout (cont=0x74d658 <work_mem_int+7128>) at ../lvgl/lv_objx/lv_cont.c:314
#5  0x00000000004179e4 in lv_cont_signal (cont=0x74d658 <work_mem_int+7128>, sign=2 '\002', param=0x7fffff7ff200)
    at ../lvgl/lv_objx/lv_cont.c:278
#6  0x0000000000418b63 in lv_cont_refr_autofit (cont=0x74d658 <work_mem_int+7128>) at ../lvgl/lv_objx/lv_cont.c:708
#7  0x000000000041798f in lv_cont_signal (cont=0x74d658 <work_mem_int+7128>, sign=1 '\001', param=0x74d6f8 <work_mem_int+7288>)
    at ../lvgl/lv_objx/lv_cont.c:274
(repeats thousands of times)
```

kisvegabor, 2019-02-28:
Thank you for the report, I'm working on it.

kisvegabor, 2019-03-08:
Should be fixed in dev-6.0. Can you verify it?

embeddedt, 2019-03-11:
@kisvegabor Sorry for the late reply. It looks like it works properly now. Thanks!

kisvegabor, 2019-03-12:
No problem, thank you for testing! :)

embeddedt, 2019-03-19:
@kisvegabor There's no crash this time but the layout has changed again.

[image]

kisvegabor, 2019-03-20:
What's the problem with the layout? It seems the horizontal resolution is just too small for an other column.

embeddedt, 2019-03-20:
@kisvegabor It's the same horizontal resolution as before (480).

kisvegabor, 2019-03-20:
With these settings, it looks like this for me. What is different in your config?
[image]

embeddedt, 2019-03-20:
@kisvegabor

> not working properly with the *default theme*.

kisvegabor, 2019-03-20:
oh, sorry with the default theme only the first column is visible.

kisvegabor, 2019-03-20:
I've found it. It was because of the styles of the `table` was not initialized and paddings were too big with the default styles.

[image]

embeddedt, 2019-03-20:
@kisvegabor Thanks! I'll test it later today but I'm sure it's working now. :smiley:

closed 2019-03-20 by embeddedt

kisvegabor, 2019-03-20:
I hope so :)