lvgl/lvgl #914: Reduce Glyph Storing Size

beibean, 2019-02-28:
Hi,

I just saw how the Font Glyphs/Symbols are stored and I can see that there is loads of Flash memory it's been wasted.

This is because for every Glyph/Symbol we store every pixel within its height. No matters if it is useful information or not:

[image]

When I have been working with glyph storing, **baseline** concept has been taken into account. Where baseline stands for the difference between the _height 0_ and the first row with any non-zero pixel starting to count from bottom.

I think LittlevGL should include this concept and get rid of loads of _0x00 useless bytes_. So, for the two characters shown above, they would be stored like this:

[image]

If this new concept is included several changes should be done:

- Tools to get fonts.
- Update font files.
- Add new baseline parameter to structure lv_font_glyph_dsc_t.
- Writing glyphs/symbols functions.

Let me know what you think about this idea!

kisvegabor, 2019-02-28:
Hi,

What a coincidence! @puzrin is planning to rewrite the converters in `node.js` and we were just discussing this improvement too. So probably it will be in `v6.0`.

beibean, 2019-02-28:
Awesome!

I venture to say that zillions of bytes will be saved all over the world with this improvement 😃.

puzrin, 2019-02-28:
Let me start with some buzz, as usual :). 

## Goals and estimates

First, we work in "resource-constrained" mode (we have limited time to spend, and should distribute it for most valuable things).

- Current sample describe principle, but does not give estimate of total saving for all font glyphs (`^` is very edge case). Total saving for real projects should be calculated to decide if it worth spend time for optimizations.
- Personally, i DON'T like to spend my time for writing new format spec. But if someone provide such document/example - it will be supported in font convertor. If not - i'm ok with existing format.

## Format spec

- DO NOT reinvent new formats. Check existing ones first.
  - For example, one from u2g8 (it's not single, but the most obvious).
- Check, that new format can be rendered effectively.
- Check, that new format supports important features
  - kerning table
  - grayscale (for antialiasing)
  - (?) check that "holed" sequences (multiple single glyphs with non-sequent charcodes) are supported well (both store and invoke)

---

Is someone wish to spend time for spec development - keep me in touch, i'll try to help.

BrianPugh, 2019-03-01:
Just to throw fuel into the fire, we could integrate miniz (stripped down zlib) into the library and store each glyph with a very limited dictionary size. This would, however, require more CPU resources. I do think baseline is a good (and common) bitmap glyph solution.

Really, we should just have an engine that accepts standard formats, like BDF:
https://en.wikipedia.org/wiki/Glyph_Bitmap_Distribution_Format

This takes into account a lot of the standard optimizations.

puzrin, 2019-03-01:
@BrianPugh 1 year ago i reviewed a lot of GUI-s and related things, to understand sutiation and select something useable. BDF support it a road to nowhere :)

1. BDF does not support kerning. After this point - nothing to discuss :)
2. Font size has no benefits over u8g2 format
3. Utils are ugly. We will have to write something useable ourself - cost (time) is the same as for custom format.

I also don't beleive to miniz compression. You can check that easy - take binary u8g2 font and try to gzip. You will also significantly increase memory requirements. RAM is much more expensive than FLASH in small MCUs.

Just to clarify - i'm one of core devs of [pako](https://github.com/nodeca/pako) and [fontello](https://github.com/fontello/fontello), and i understand internals of discussed things good enougth.

kisvegabor, 2019-03-01:
**Font format**
I think it's not necessary to use a third party lib for it. Having our own solutions gives more flexibility and will fit the best to LittlevGL. The current format is easy to extend to work with the "baseline optimization".

puzrin, 2019-03-01:
> In addition, in `u8g2` I didn't see anything about anti-aliasing (more bit per pixel)

As i said, `u8g2` is not ideal. But it worth to read prior to invent any kind of optimizations.

beibean, 2019-03-01:
I have taken a look at BDF and u8g2 and IMO, custom optimization as baseline one fits us better:

- As @kisvegabor says, it's really easy to implement in LittlevGL.
- I don't know how the lv_util works, but adaptation to this new _baseline format_ will _just_ need (glyph crop & add bottom crop value as baseline to lv_font_glyph_dsc_t table.
- Big flash saving space is achieved. I would say that, at least as u8g2.

I will do approximate calculations about how much we can save with _baseline optimization_ with a default font just to see in what order of magnitude we are moving. 

Y you need me for any other task I'm willing to help as much as I can.

beibean, 2019-03-01:
I finally calculated the sizes using the font used in the project I am involved in right now:

Font: FreeSans
Font Size: 14
bpp: 2
Alphabets: 
[image]
[image]
[image]
[image]

Bitmap size without baseline format:
- Latin: 2912 bytes
- Latin Extended: 3136 bytes
- Greek & Coptic: 3344 bytes
- Cyrillic: 2416 bytes

Bitmap size with baseline format (crop on top and crop on bottom):
- Latin: 1817 bytes (+ 1 byte * 94 chars)
- Latin Extended: 2135 bytes (+1 byte * 95 chars)
- Greek & Coptic: 2184 bytes (+ 1 byte * 97 chars)
- Cyrillic: 1538 bytes (+ 1 byte * 64 chars)

Remember this are results for 2bpp Font. The greater resolution the more flash saved.

puzrin, 2019-03-01:
32%... nice but not impressive... But if it costs nothing at all...

BrianPugh, 2019-03-01:
I'm all for saving a few KB of flash for a relatively simple change :+1:

kisvegabor, 2019-03-01:
@beibean Thank you for the calculation. Saving 1/3 of flash without sacrificing anything is graet!

BrianPugh, 2019-03-01:
Just for the sake of argument (I think the baseline idea is better), I decided to investigate the zlib/miniz option a little further. An argument could be made that decompressing glyphs doesn't require an additional library if the application already requires PNG decompression (which contains zlib/miniz).

Using the 8bpp `^` dejavu_20 example. Compression is performed using the smallest output buffer window of 256-bytes (smallest zlib in python will let me go) and best compression. This assumes that each character would be independently compressed for random access. Baseline modification assumes the baseline is stored in a `uint8_t`

Uncompressed: 220 bytes

ZLIB Compressed:  58 bytes

Uncompressed + baseline: 56 bytes

ZLIB + baseline: 56 bytes

So no savings here, but lets look at a larger character, like `A`

Uncompressed: 240 bytes

ZLIB Compressed: 119 bytes

Uncompressed + baseline: 157 bytes

ZLIB + baseline: 117 bytes

So, just from these 2 points, we can see a few things:
1. The compression barely benefits from the baseline modification
2. ZLIB + baseline is always equal or outperforms the baseline modification
3. For larger characters, could save an additional 25%

Of course, this is at the expense of CPU cycles (decompression is fast, but obviously slower than just loading from memory) and 256 bytes of RAM for decompression. Overall, probably not worth it, just providing some data.

puzrin, 2019-03-01:
@BrianPugh looks like compare of "unonptimized with imaginated" :). I'd suggest to check metrics of battle-tested implementations first (u8g2). It drops data out of bounding box & has some kind of RLE. That's much more simple than deflate and good enougth (because in big glyphs each line has long repeating sequences).

Also, if you draw huge glyph, you may not be able to allocate memory for full decompression. But it will be enough to access line-by-line (each line should be packed separate). If you draw line-by line, with applying "recolor" & opacity to each pixel, then RLE will cost you nothing.

> So, just from these 2 points, we can see a few things:

I'd suggest to fix metrics first:

- measure full glyphs set instead of separate glyphs
- use real glyph sizes (for example, from matherial ui) and real BPP (i guess, 4 bits of grayscale will be enougth)

BrianPugh, 2019-03-01:
I agree RLE is much more appropriate for this task.

kisvegabor, 2019-03-01:
Can you measure or guess approximately what decompression rate can we achieve with RLE?

puzrin, 2019-03-01:
> Can you measure or guess approximately what decompression rate can we achieve with RLE?

I'd expect something like 2x (additional to bounding box crop) for big scale 4bpp fonts.

kisvegabor, 2019-03-02:
> I'd expect something like 2x (additional to bounding box crop) for big scale 4bpp fonts.

Meanwhile, I made a simple test with on only one character (4bpp, 20 px height, croped):
```c
    0x38, 0xdf, 0xfd, 0x92, 0x00,  //.%@@@@%..
    0xff, 0xff, 0xff, 0xff, 0x50,  //@@@@@@@@+
    0xc6, 0x20, 0x03, 0xaf, 0xd0,  //@+....%@@
    0x00, 0x00, 0x00, 0x1f, 0xe0,  //.......@@
    0x00, 0x00, 0x02, 0xaf, 0x90,  //......%@%
    0x00, 0xff, 0xff, 0xd7, 0x00,  //..@@@@@+.
    0x00, 0xff, 0xff, 0xf8, 0x00,  //..@@@@@%.
    0x00, 0x00, 0x14, 0xcf, 0x80,  //.....+@@%
    0x00, 0x00, 0x00, 0x1f, 0xe0,  //.......@@
    0x00, 0x00, 0x00, 0x2f, 0xf0,  //.......@@
    0xb4, 0x10, 0x14, 0xdf, 0xb0,  //%+...+@@%
    0xff, 0xff, 0xff, 0xfe, 0x20,  //@@@@@@@@.
    0x39, 0xdf, 0xfd, 0x81, 0x00,  //.%@@@@%.
```

The raw data is (65 bytes):  
```
0x38, 0xdf, 0xfd, 0x92, 0x00, 0xff, 0xff, 0xff, 0xff, 0x50, 0xc6, 0x20, 0x03, 0xaf, 0xd0, 0x00, 0x00, 0x00, 0x1f, 0xe0, 0x00, 0x00, 0x02, 0xaf, 0x90, 0x00, 0xff, 0xff, 0xd7, 0x00, 0x00, 0xff, 0xff, 0xf8, 0x00, 0x00, 0x00, 0x14, 0xcf, 0x80, 0x00, 0x00, 0x00, 0x1f, 0xe0, 0x00, 0x00, 0x00, 0x2f, 0xf0, 0xb4, 0x10, 0x14, 0xdf, 0xb0, 0xff, 0xff, 0xff, 0xfe, 0x20, 0x39, 0xdf, 0xfd, 0x81, 0x00, 
```

I used this site the execute RLE: https://eleif.net/rle.html
The result (96 bytes):
```
0x01,0x38,0x01,0xDF,0x01,0xFD,0x01,0x92,0x01,0x00,0x04,0xFF,0x01,0x50,0x01,0xC6,0x01,0x20,0x01,0x03,0x01,0xAF,0x01,0xD0,0x03,0x00,0x01,0x1F,0x01,0xE0,0x02,0x00,0x01,0x02,0x01,0xAF,0x01,0x90,0x01,0x00,0x02,0xFF,0x01,0xD7,0x02,0x00,0x02,0xFF,0x01,0xF8,0x03,0x00,0x01,0x14,0x01,0xCF,0x01,0x80,0x03,0x00,0x01,0x1F,0x01,0xE0,0x03,0x00,0x01,0x2F,0x01,0xF0,0x01,0xB4,0x01,0x10,0x01,0x14,0x01,0xDF,0x01,0xB0,0x03,0xFF,0x01,0xFE,0x01,0x20,0x01,0x39,0x01,0xDF,0x01,0xFD,0x01,0x81,0x02,0x00
```

So RLE made it larger. Am I doing something wrong? Or is it a too some RLE algorith to just count the idientical bytes?

puzrin, 2019-03-02:
There are tons of RLE variations. You tried wrong one - for bytes, but we need for "pixels".

Try one from u8g2. Seems it even has a simplified fixed vocabulary.

puzrin, 2019-03-02:
I've googled a bit, what happens with grayscale compressions. Here is a short summary, if anyone wish to dive into topic:

- Google "grayscale image compression algorythms", there are many interesting modern publications.
- Specially for our case, with high white/black repeats, RLE-family CAN be effective for big sizes & bpp >=4.
- [I3BN](https://www.researchgate.net/publication/305820775_New_Modified_RLE_Algorithms_to_Compress_Grayscale_Images_with_Lossy_and_Lossless_Compression) encoding (optimized RLE) is nice start.
- Next step is switch to variable-length codes (Huffmann).
  - Our case is very predictable (0, ff - max, everything between - few), and we can use predefined tables. That's not principle - only compression time changes (not our case).

There are more advanced methods, but complexity will increase faster than compression ratio.

beibean, 2019-03-02:
We have to be careful with RLE compressions, it is a double-edged sword, since we can reduce storing size for many glyphs but increase for others.

In the project I am involved in I use RLE for icons (grayscale 4bpp, 50x50, 40x40 and 36x36) but just for the background pixels (0x00). This is since I made an study long time ago and this was the best compression level we could achieve.

Just to list some of the points to consider:

- As puzrin says, using RLE for grayscale < 4bpp may be worse. I did the study of RLE with the font set (2bpp) shown some comments above and the size overall increased.
- Font size matters: the bigger the font, the more we can save using RLE. Reaching cases where if we use RLE for small fonts (even >=4bpp) size could be increased.
- As puzrin says as well, in fonts we use, the most used pixels are 0x00 and 0xFF. Therefore, those are the potential ones to apply RLE.

Taking all the points into account, if we finally use RLE to compress:
- In wich pixels will be applied RLE. Only 0x00, only 0xFF or both.
- How lv_utils knows to which char/font should apply RLE.
- Compress by font set: new variable needed to know de decompression algorithm.
- Compress by chars: new variable needed for every char.

kisvegabor, 2019-03-03:
@puzrin 
I've checked the mentioned the I3BN article but it needs some time to understand it in details :)

@beibean 
Selectively applying RLE is a very good idea. It seems easy to decompress if the storage format is `(data, [cnt])` where `cnt` only exists if `data` has specific values (0x00 and 0xFF in or case).

In LittlevGL to fonts' bitmaps are returned by a [callback](https://github.com/littlevgl/lvgl/blob/master/lv_misc/lv_font.h#L56) function. It makes possible to: 
- simply return a pointer from flash (most typical case)
- read the bitmap from external flash into a buffer and return it 
- real-time render the bitmap from a TTF font into a buffer and return it
- **decompress the bitmap into a buffer and return it**

So we can handle it on font level (not character level)

puzrin, 2019-03-03:
> I've checked the mentioned the I3BN article but it needs some time to understand it in details :)

I3BN describes very simple principle of encoding repeating (byte) patterns:

1. Single [ a ] => [ a ], 0 <- bit
2. 2 chars: [ a ], [ a ] => [ a ], 1, 0
3. 3 chars: [ a ], [ a ], [ a ] => [ a ], 1, 1, 0
4. more => [ a ], 1, 1, 1, (count - 3)

kisvegabor, 2019-03-03:
@puzrin I don't know why a simple example is not included in the article. It's clear now, thank you!

As we usually have only a few repeating data it seems a more effective compression. But the decompression is more expensive as we are thinking in bits and not bytes.

puzrin, 2019-03-03:
> @puzrin I don't know why a simple example is not included in the article. It's clear now, thank you!

It's included, but authors are real scientist. I've spent 4 hours to read all related articles, prior to understand what they mean :)

> As we usually have only a few repeating data it seems a more effective compression. But the decompression is more expensive as we are thinking in bits and not bytes.

That depends... I'd say, the most expensive operations are not maths but memory access. From this point of view, RLE even has less data to read.

If you take into account all next operations of software rendering (mix colors/opacity), i'd say decompression expences will be about zero.

> In LittlevGL to fonts' bitmaps are returned by a callback function. It makes possible to:
>
> - simply return a pointer from flash (most typical case)
> - read the bitmap from external flash into a [...]

That will still require significant memory buffer for huge glyphs. In theory, it would be more reasonable to return something like "pixels stream", or "lines stream".

With lines it will be still possible to have effective unpack + be effective with acccelerators (less effective than with rectangles, but still good enougth).

Note, all this is about big fonts only. Small fonts can works different (as now, with inline bitmap unpack).

beibean, 2019-03-03:
> So we can handle it on font level (not character level)

This will actually work for most of the fonts (>=4bpp), but what I was thinking in character level was including the first byte with compression algorithm used. As an example:

- 0x00: no RLE.
- 0x01: RLE used just for 0x00.
- 0x02: RLE used just for 0xFF.
- 0x03: RLE used for 0x00 & 0xFF.

Of course, this will increase a lot time cost of developing lv_utils (should check different algorithms for every glyph and decide which one is better) and I am not sure this effort worths it.

kisvegabor, 2019-03-04:
> That will still require significant memory buffer for huge glyphs. In theory, it would be more reasonable to return something like "pixels stream", or "lines stream".

It not as much as it seems. For example with a 60px X 40px glyph on 4bpp it is 1,2kB RAM. Not a big deal.

> This will actually work for most of the fonts (>=4bpp), but what I was thinking in character level was including the first bitmap byte with compression algorithm used.

It sounds good! Adding this extra byte and the special `bitmap_get` callbacks all can work together.

puzrin, 2019-03-04:
Could you tell, what bpp required for comfortable font look, from your point of view? Is 4bpp ok, or difference still visible with 8bpp?

kisvegabor, 2019-03-04:
It depends on the pixel density of the display too. With higher pixel density you can use smaller bpp.

My experiences:
- I used 8 bpp for very small fonts like 10 px. I'm not sure there is any visible difference from 4bpp. 
- For "normal" sizes like ~20px I usually use 4bpp.
- For large fonts like 80 px I use 2bpp.
- I use 1bpp only for pixel fonts.

beibean, 2019-03-15:
Hi again,

I found that some characters add _blank_ pixels on both sides:

[image]

Apart from the flash saving issue, there is a difference between some characters spacing when represented.

So, IMO lv_utils should _crop_ the gyph on four sides.

What do you think?

puzrin, 2019-03-15:
> What do you think?

- This is covered in u8g2 spec (+ coupled with RLE compression)
- The root question is - can we switch to bit stream instead of "pixels".

IMHO - nothing new to discuss. It's more easy to do all at once.

kisvegabor, 2019-03-17:
@beibean 
> Apart from the flash saving issue, there is a difference between some characters spacing when represented.

It seems like an issue in the converter due to rounding. In the converter, the glyphs are rendered with 8 bpp and then reduced to 1/2/4 if required. The glyphs are cropped with 8 bpp too. Probably the extra columns are there because there was a very light pixel on the sides but it was rounded to zero.

Hopefully, it will work correctly in @puzrin's script. 

@puzrin 
> The root question is - can we switch to bit stream instead of "pixels/bytes".

Actually, we are already on bit-stream. The only constraint is to finish a line with a full byte.

puzrin, 2019-03-17:
> The only constraint is to finish a line with a full byte.

How important is that?

- Can we expect GUI to request always full bitmap unpack for glyph?
  - seems yes for small and medium sizes
  - under question with huge sizes (imagine sliding string with full screen height size)
- As far as i know, lvgl has no interfaces to request glyphs line-by-line. If we design line-aligned compression for big sizes, how difficult is it to add appropriate support to core? Should we care about this at all?

kisvegabor, 2019-03-18:
> How important is that?

LittlevGL draws only the required part of a letter. The "full byte line" is used to make the navigation in the bitmap easier.
 
> Can we expect GUI to request always full bitmap unpack for glyph?

Let's take an extreme example. A glyph with 100x200 size and 4 bpp needs 10kB. Still acceptable for rare and extreme cases.

Do you agree?

puzrin, 2019-03-18:
> The "full byte line" is used to make the navigation in the bitmap easier.

What kind of navigation do you mean, if current api supports only full bitmap render to buffer? For full render consecutive access from start to end is enougth.

> Let's take an extreme example. A glyph with 100x200 size and 4 bpp needs 10kB. Still acceptable for rare and extreme cases.

> Do you agree?

So-so. M0 (stm32f0xxx) usually have 4-8-16К memory.

Of cause, we can always say "for extreme cases, use more powerful MCU-s". But i'd like to understand first, can we solve buffer size problem with minimal efforts or not? Or can we reserve such extendability, without immediate implementation.

If that's difficult - we can drop this feature.

kisvegabor, 2019-03-19:
> What kind of navigation do you mean, if current api supports only full bitmap render to buffer? For full render consecutive access from start to end is enougth.

By "easy navigation" I mean to get the required pixel's value fast when rendering a glyph. It's a small overhead in memory usage but makes font rendering faster and easier.

> So-so. M0 (stm32f0xxx) usually have 4-8-16К memory.

LittlevGL needs at least 10 kB RAM and 80 kB flash. So the entry level is rather STM32F103.

> Of cause, we can always say "for extreme cases, use more powerful MCU-s". But i'd like to understand first, can we solve buffer size problem with minimal efforts or not? Or can we reserve such extendability, without immediate implementation.

I think it could be an issue in very very rare cases. As a comparision, form the `y offset` improvement every user would benefit. So I vote for leaving `get_line` for now.

puzrin, 2019-03-19:
> LittlevGL needs at least 10 kB RAM and 80 kB flash. So the entry level is rather STM32F103.

https://lcsc.com/product-detail/ST-Microelectronics_STMicroelectronics_STM32F072CBT6_STM32F072CBT6_C81720.html 16K + 128K

stm32f072 has interesting feature - USB without crystal, not available in F1 and F3 families.

Also, some questions about memory:

- Is 80K flash requirement from real code size or ineffective fonts store? I'm not familiar with details.
- As far as i understand, minimal buffer size for 1/20 320\*200\*16bpp is ~2K. I thought, rendering buffer is the most aggressive RAM consumer. Who eats the rest of 6K?

puzrin, 2019-03-19:
> As a comparision, form the y offset improvement every user would benefit. So I vote for leaving get_line for now.

As far as i understand, you mean rendering text into cropped area.

- "skip XX pixels/lines in RLE bitstream without render" should be fast enougth.
- if you wish to optimize, it's not necessary to use 1 line precision, you can skip with 3-line blocks, for example.

On other words, that looks similar, but not the same as `get_line`, that's `skip_to_line` and does not require pack each line separate.

kisvegabor, 2019-03-20:
> Is 80K flash requirement from real code size or ineffective fonts store? I'm not familiar with details.

It's for almost all features are disabled and only a small ASCII font used with ASCII range.

> As far as i understand, minimal buffer size for 1/20 320*200*16bpp is ~2K. 

`1/20 x 320 x 200 x 2 (16 bpp in bytes) = 6.25kB`

> I thought, rendering buffer is the most aggressive RAM consumer. Who eats the rest of 6K?

Besides, you need some heap for LittlevGL: `> 2kB`

When I examined the memory usage of LittlevGL I didn't find a tool to print the RAM/ROM usage for each file. Do you know such a tool?

> As far as i understand, you mean rendering text into cropped area.

Mainly yes. If you look at the [letter rendering function](https://github.com/littlevgl/lvgl/blob/master/lv_draw/lv_draw_vbasic.c#L237) you can see it's quite complex even if it takes advantage of "full-byte-rows".

> if you wish to optimize, it's not necessary to use 1 line precision, you can skip with 3-line blocks, for example.

It still would make things complicated because not full lines need to handles and the benefit would be minimal. Let's say a font with: 
- height = 20
- average width = 10
- glyph count = 100
- bpp = 4;

The size is 10kB. If it has 4 bpp, in 50% of cases we cloud save 0,5 in each row which is 
`20 (height) x 100 (glyph cnt) x 0.5 (50%) + 0.5 (saved bytes) = 500 bytes`. `0,5/10 = 5%`. I calcualted with "full-bitstrem". For a so small gain I thing it not worth ti implement it.

puzrin, 2019-03-20:
Thanks for all estimates.

> The size is 10kB. If it has 4 bpp, in 50% of cases we cloud save 0,5 in each row which is
20 (height) x 100 (glyph cnt) x 0.5 (50%) + 0.5 (saved bytes) = 500 bytes. 0,5/10 = 5%. I calcualted with "full-bitstrem". For a so small gain I thing it not worth ti implement it.

I'm 100% agree, 5% of gain is not enougth to spend time. But we can also not gain for current features, but adding new features, not available earlier. May be (not sure) we can fit more high quality (MD?) themes into the same hardware requirements.

- Gain increase with glyph size (significantly).
- Gain also increase with bpp.

I think, next step would be to make test compressors to measure exact savings

beibean, 2019-03-20:
>> When I examined the memory usage of LittlevGL I didn't find a tool to print the RAM/ROM usage for each file.

If you are using ARM, you can take a look a [arm-none-eabi-size](https://manned.org/arm-none-eabi-size/c5dd2ab5). For other compilers I think there should be options within the post-build in order to print size.

kisvegabor, 2019-03-21:
> I'm 100% agree, 5% of gain is not enougth to spend time. But we can consider situation not as gain for current features, but for adding new features, not available earlier.

I don't understand. Which gain do you mean? You just said it's not worth for 5%.  

> I think, next step would be to make test compressors to measure exact savings

I liked @beibean's idea of selecting the best RLE type for each glyph and compressing only 0x00 and 0xFF. @beibean could you make some test with some glyphs?

@beibean 
> If you are using ARM, you can take a look a arm-none-eabi-size. For other compilers I think there should be options within the post-build in order to print size.

Thank you, I meant exactly this. I didn't notice the `-t` flag. In the table, I don't see any extraordinary things at first. Anyway, I'll play with it a little bit later.

beibean, 2019-03-21:
> could you make some test with some glyphs?

Sure I will. The problem is when and how long will it take 😄 

I will be updating info as I get results.

puzrin, 2019-03-21:
@beibean how are you with javascript? Base skeleton of new convertor will be ready in couple of weeks. Then anyone will be able to write test compressions.

beibean, 2019-03-21:
@puzrin I have never work with javascript. I think I will make the tests with python/matlab since I already have the base (use it to compress icons with RLE on 0 always).

@kisvegabor Will it be possible for LittlevGL to change the way it aligns texts? If we add "baseline" concept, when setting the position of a text, the "y" coord should indicate where the baseline lays.

kisvegabor, 2019-03-23:
> I have never work with javascript. I think I will make the tests with python/matlab since I already have the base (use it to compress icons with RLE on 0 always).

As it's "only" an extension it's blocking the developement of the font converter. @puzrin correct me if I'm wrong.

> Will it be possible for LittlevGL to change the way it aligns texts? If we add "baseline" concept, when setting the position of a text, the "y" coord should indicate where the baseline lays.

Do you mean when you use `lv_obj_set_y(label, 10)` the baseline should be at `y = 10`?

puzrin, 2019-03-23:
> As it's "only" an extension it's blocking the developement of the font converter. @puzrin correct me if I'm wrong.

That's not critical part to block everything. To debug renderer and so on, it's enougth to have json/png/lvgl5 backends.

beibean, 2019-03-23:
>Do you mean when you use lv_obj_set_y(label, 10) the baseline should be at y = 10?

That's exactly what I mean. When I have use own graphical library, texts where positioned that way since it's the "point you control". In "live" texts you can't control the absolute height of a changing string.

What do you think?

kisvegabor, 2019-03-24:
`lv_obj_set_y()` is generic and tells where to move the top edge of the object. So we can make it work differently in case of labels. But we can add a function directly for the label: `lv_label_set_y_baseline()`.

>  In "live" texts you can't control the absolute height of a changing string.

In LittlevGL a label with `"."` or `"A"` has the same height which is extracted from the font's `h_px` parameter. And it's not possible to mix fonts with different height in a label so the "point of control" is sill can be the top edge. However, "baseline" would be useful to align two texts with different height next to each other.

It seems storing the baseline / font is appropriate. E.g. in case of a 20 px height font baseline is 14px. Do you agree?

puzrin, 2019-03-24:
> It seems storing the baseline / font is appropriate. E.g. in case of a 20 px height font baseline is 14px. Do you agree?

As far as i understand, you mean `ascent`. I'd suggest to dig fontkit's renderer, because real fonts keep much more info. I don't beleive single `accent` will solve all your "problems".

kisvegabor, 2019-03-24:
> As far as i understand, you mean ascent. I'd suggest to dig fontkit's renderer, because real fonts keep much more info. I don't beleive single accent will solve all your "problems".

What other problems do we have?

puzrin, 2019-03-24:
> What other problems do we have?

I think from another side. Instead of search separate problems, and add params one-by one, i'd suggest to read TTF tables spec, and verify nothing missed.

If accent was not stored, there is high probability to miss something else.

kisvegabor, 2019-03-25:
@puzrin I've check it and ended up on this [Wikipedia page](https://en.wikipedia.org/wiki/Ascender_(typography)). Form this what we need is only the position of the baseline.

beibean, 2019-03-25:
> But we can add a function directly for the label: lv_label_set_y_baseline()

This would be perfect. 

> It seems storing the baseline / font is appropriate. E.g. in case of a 20 px height font baseline is 14px. Do you agree?

It atually depends on the font type, so it should be calculated for every font and besides, every glyph should have either absolute baseline or relative baseline offset to fonts baseline.

One more thing. When aligning a text, lets say `LV_ALIGN_IN_BOTTOM_MID`, what would happen with this two texts?:

`asdklm`

`fjyp`

They should be laying on the same baseline  as you can see here: `asdklm   fjyp`.

kisvegabor, 2019-03-25:
> It atually depends on the font type, so it should be calculated for every font and besides, every glyph should have either absolute baseline or relative baseline offset to its font baseline.

So what is needed to position the glyphs vertically:
- `row_cnt` number of rows in the cropped glyph
-  `baseline_ofs` space between the font's baseline and the bottom edge of the cropped glyph
  
> One more thing. When aligning a text, lets say LV_ALIGN_IN_BOTTOM_MID, what would happen with this two texts?

`lv_obj_align` uses `lv_obj_set_x/y` and know nothing about the object's type. The labels have the same height regardless to the characters inside it. So in your example, the bottom of the labels will be aligned but therefore the baselines will be the same as well. 

But if you have two labels, one with 12 px font and other with 18 px font the baselines will not be aligned. However, you should be able to get the baseline of the fonts and use it as a correction in the `y_mod` paramter of `lv_obj_align`.

beibean, 2019-03-25:
> So what is needed to position the glyphs vertically:
>- row_cnt number of rows in the cropped glyph
>- baseline_ofs space between the font's baseline and the bottom edge of the cropped glyph

That's it. With these parameters and font's baseline we should be able to place every glyph. (Note that if we use relative offset, negative baseline is possible).

> lv_obj_align uses lv_obj_set_x/y and know nothing about the object's type. The labels have the same height regardless to the characters inside it. So in your example, the bottom of the labels will be aligned but therefore the baselines will be the same as well.

Just to make it clear, when introducing baseline concept and aligning texts to buttom, texts belonging to the same font are not going to align like this:

[image]

But, like this:

[image]

Is it?

> But if you have two labels, one with 12 px font and other with 18 px font the baselines will not be aligned. However, you should be able to get the baseline of the fonts and use it as a correction in the y_mod paramter of lv_obj_align.

That's no problem and besides a good enhancement for library 😉

kisvegabor, 2019-03-25:
> Just to make it clear, when introducing baseline concept and aligning texts to buttom, texts belonging to the same font are not going to align like this:

Texts with the same font are aligned well now as well. The problem occurs when you have fonts with different height/baseline. Then you need to use `y_mod` parameter of `lv_obj_align`

kisvegabor, 2019-03-26:
I move this issue to the font converter repo as it's more related to it.