lvgl/lvgl #915: Propossals When Using Sparse Fonts

beibean, 2019-02-28:
When looking to the way a sparse font is implemented I have come up with two propposals I would like to share with you.

**Font tool should generate unicode list**
I think it should be very simple to generate this list since characters are given as an input and this would make it easier for those who use Sparse Fonts.

**Use binary search instead of linear search**
Assuming the glyphs are ordered according to their unicode value, [binary search](https://en.wikipedia.org/wiki/Binary_search_algorithm) ([bsearch in C](https://www.tutorialspoint.com/c_standard_library/c_function_bsearch.htm)) is a way faster than linear search like it is done now.
Looking for glyphs is a task that is done for every character in a label and should be done as faster as possible.

What do you think about this suggestions??

kisvegabor, 2019-02-28:
> Font tool should generate unicode list

Actually, the unicode list is added to the font as `<name>_unicode_list`. 

> Use binary search instead of linear search

It really would be better and I'm planning to update is a long time ago. Would you be interested in improving it and sending a  PR?

beibean, 2019-02-28:
Sorry, I'm just looking the library over and I have not put myself to work seriously on it. So I just saw the default files found under lv_fonts folder.

About including binary search, no problem. I will implement it by next week and send a PR!

kisvegabor, 2019-02-28:
> Sorry, I'm just looking the library over and I have not put myself to work seriously on it. So I just saw the default files found under lv_fonts folder.

No problem :) The default fonts are not "sparse"  therefore all letters between `unicode_first` and `unicode_first` is included. 

> No problem. I will implement it by next week and send a PR!

Great, thank you!

puzrin, 2019-02-28:
See https://github.com/littlevgl/lvgl/issues/914#issuecomment-468266232 and https://github.com/littlevgl/lv_utils/issues/19

All this proposals are about "new font format". It would be better to solve all at once, because incremental format change is ass pain.

Also note, all such suggestions are "well known" (almost everything used in u8g2 fonts, for example). No need to reinvent wheel :). I'd suggest to create issue about new font format spec and keep all things in one place.

puzrin, 2019-02-28:
To be "on the same wave" - see u8g2 font format description:

https://github.com/olikraus/u8g2/wiki/u8g2fontformat

It's not ideal, but worth to know.

kisvegabor, 2019-03-01:
See my comment [here](https://github.com/littlevgl/lvgl/issues/914#issuecomment-468555861).

beibean, 2019-03-01:
Hi,

Just to clarify, I think this proposals are not related with a new font format since the first of them is already implemented 😃.

**Font tool should generate unicode list**
It already does, so no problem with it.

**Use binary search instead of linear search**
It is about the C method used to search an item within a list, not about font format. If no problems I will implement this as soon as I can.

kisvegabor, 2019-03-01:
> It is about the C method used to search an item within a list, not about font format. If no problems I will implement this as soon as I can.

Awesome, thank you in advance! :) Theretically this update should go to `dev-6.0` but it's still continously changing. You can send the PR for `master` and I'll rebase it to `dev-6.0`.

kisvegabor, 2019-03-03:
@beibean CanI assign you to this issue?

beibean, 2019-03-03:
@kisvegabor Sure 😉

kisvegabor, 2019-03-19:
The PR is merged so I close this issue.

Thank you for this improvement!

closed 2019-03-19 by kisvegabor