lvgl/lvgl #920: ta cusor size(height) display not right when set ta's child label style

HinsunRan, 2019-03-02:
1、My littlevgl version is the latest 5.3(2019-03-01 code)
2、The test code is the lv_test_kb_2, i want to set ta fixed size: width 200pix, height 60pix，
   and set the ta's child label style：talabelstyle.text.font = &lv_font_dejavu_40;
   the corresponding effect could be that the height of cursor is 40pix(or 60pix ？)。
   In the result of test in first display，the height of cusor is not right。
[image]
   And the height is right when i input some char by kb.
[image]
   i follow the code to find the height is 20pix。
   my test code is：
```
void lv_test_kb_2(void)
{
	lv_obj_t * ta = lv_ta_create(lv_scr_act(), NULL);
	//lv_ta_set_text(ta, " ");
	lv_obj_align(ta, NULL, LV_ALIGN_IN_TOP_MID, 0, 30);
	lv_obj_set_size(ta, 200, 60);
	static lv_style_t talabelstyle;
	lv_style_copy(&talabelstyle, lv_obj_get_style(lv_ta_get_label(ta)));
	talabelstyle.text.font = &lv_font_dejavu_40;
	lv_obj_set_style((lv_ta_get_label(ta)), &talabelstyle);
        ...
}
```

   how to make the cursor height shows be right in first display ？ 
   or how to make the cursor height be synchronized with ta size ？
   Thank you very much。

embeddedt, 2019-03-02:
@HinsunRan This problem was fixed in `dev-6.0` but not in `master`. I just applied the fix to the `master` branch, can you test that it works now?

HinsunRan, 2019-03-03:
Great! It is working now. The height of cursor is right in first display.
And the height be synchronized with ta size, i will solve it by refering to your method. 
Thank you very much.

closed 2019-03-04 by HinsunRan