lvgl/lvgl #927: [Proposal] Public image decoder API

manison, 2019-03-05:
In related discussion #684 the image rotation API is being designed. Although right now I'm rather interested in image mirroring than arbitrary angle rotation I suppose the transformation functions would all work similarly with raw pixel buffers, i.e. _source buffer_ → _transformation_ → _destination buffer_. The buffers are supposed to work with the _canvas_ object. My problem is that my images are compressed. So now I use the _image_ object which supports decoding images but it can't transform image and it won't provide me access to the raw pixel buffer either.

.|image|canvas
-|-----|------
supports custom decoder|✓|X
supports pixel buffer|X|✓

To solve this problem I would like to decode the compressed image into the pixel buffer, eventually perform the transformation and then assign the buffer to the canvas object. But unless I'm missing something the custom image decoder is only used internally and there is no public surface to use it from user code (sure, I can call lodepng directly, but my point is to use some universal lvgl API with whatever custom decoder registered).

I suggest creating the following API:

```c
lv_img_dsc_t * lv_img_dsc_open(void * src);
void lv_img_dsc_close(lv_img_dsc_t * dsc);
```

The `lv_img_dsc_open`
* If `src` is `LV_IMG_SRC_VARIABLE` and is **not** `LV_IMG_CF_RAW`, returns the `src` directly.
* If `src` is `LV_IMG_SRC_VARIABLE` and is `LV_IMG_CF_RAW`, dynamically allocates the `lv_img_dsc_t`, the buffer and uses custom decoder to decode the raw data.

The `lv_img_dsc_close`
* Frees the `lv_img_dsc_t` if it was allocated dynamically.
* Otherwise does nothing.
One of the bits reserved in the `lv_img_header_t` would be used to flag when the structure is allocated on the heap.

The custom decoder interface would probably need to be changed to decode into user supplied buffer to prevent multiple allocations and memory copying.

What do you think?

embeddedt, 2019-03-05:
cc @kisvegabor: This is architecture-related.

beibean, 2019-03-06:
Hi,

I will use compressed images as well, and this proposal sounds  really good.

In my case images are 4bpp grayscale compressed with RLE algorithm so, instead  of using decoder just when `LV_IMG_CF_RAW`, why don't we indicate it in another way, so every kind of image can get advantage of this custom decoder?

Maybe within lv_conf.h like custom function can be assigned for `tick_inc()`.

manison, 2019-03-06:
> …instead of using decoder just when LV_IMG_CF_RAW, why don't we indicate it in another way, so every kind of image can get advantage of this custom decoder?

I'm OK with `LV_IMG_CF_RAW`. Consider the image is loaded from file or event downloaded from network. In that case you don't know in advance what the format of the image is (well, usually – with file you could possibly use file extension, with HTTP you could use Content-Type header etc). So its job of the decoder to determine what he `LV_IMG_CF_RAW` is actually.

I would rather like to have the possibility to register multiple decoders, so you can have distinct decoder for PNGs, another one for JPEGs etc. Then you have e.g. PNG image and if the first decoder does no recognize the content, it passes it to the next one.

beibean, 2019-03-07:
> [..] So its job of the decoder to determine what he LV_IMG_CF_RAW is actually. I would rather like to have the possibility to register multiple decoders [...]

What I meant by registering a global decoder within lv_conf.h was more like a decoder handler, which could receive, for example, pointer to bitmap and pointer to image info struct. Then, you can have multiple decoders within your app and decide which one to use. IMO this way, more image types can get advantage of custom decoders.

What do you think?

manison, 2019-03-07:
I think we're on the same page.

Still I would rather let the decoder itself decide whether it can handle the data or not without the need of some global umbrella handler. That way you can register decoders as you like without the need of additional configuration.

Consider you need PNG image support in your application. You do

```c
lv_img_decoder_add(&png_decoder);
```

and that's it. Nothing more.
Than you decide you need to add JPEG support. You just add one line to the above code

```c
lv_img_decoder_add(&png_decoder);
lv_img_decoder_add(&jpg_decoder);
```

and that's it. No need to configure a global handler.

The image decoder can implement some `can_handle` callback. When it returns `true`, it will be called again to actually decode the data. If it returns `false`, the next registered decoder will be tried.

```c
bool png_decoder_can_handle(const void *data, size_t len)
{
    return strncmp(data, "PNG", 3) == 0;
}

lv_color_t * png_decoder_decode(const void *data, size_t len)
{
    ...
}
```

```c
bool jpg_decoder_can_handle(const void *data, size_t len)
{
    return strnstr(data, "JFIF", 4) != NULL;
}

lv_color_t * jpg_decoder_decode(const void *data, size_t len)
{
    ...
}
```

kisvegabor, 2019-03-07:
I'm not against the idea of registering the decoders this way. However, I don't see a big benefit. In your global decoder open you can write this:
```c
const uint8_t * my_decoder_open(const void * src, const lv_style_t * style) 
{
    const uint8_t * res;
    res = png_open(src, style);
    if(res != LV_IMG_DECODER_OPEN_FAIL) return res;

    res = jpg_open(src, style);
    if(res != LV_IMG_DECODER_OPEN_FAIL) return res;

    res = bmp_open(src, style);
    if(res != LV_IMG_DECODER_OPEN_FAIL) return res;

    ....
}
``` 

which is still very simple.

If we add this feature the "register functions run-time" approach should be followed. We are working hard to remove hardcoded configs from `lv_conf.h` so we shouldn't introduce new ones. :)

manison, 2019-03-08:
> However, I don't see a big benefit. In your global decoder open you can write this: … which is still very simple.

Simple it is, sure :) Still need to write some boilerplate code though.

> If we add this feature the "register functions run-time" approach should be followed.

Agreed.

Anyway, I would like to discuss my initial proposal making the decoder API public. What do you think about it, @kisvegabor?

kisvegabor, 2019-03-08:
> Anyway, I would like to discuss my initial proposal making the decoder API public. What do you think about it, @kisvegabor?

I didn't immerse in it deeply yet but my initial thoughts are:
1. having a generic open function would assume that the width, height, and other information are stored in the file. Of course, it's true for png, jpg etc but not true in general. You can have a custom image format where you store only the pixel data somehow and set the meta-data in `lv_img_dsc_t`  directly. For example  `LV_IMG_CF_INDEXED_1/2/4/8BIT` has a palette on the beginning of the image data and then only the indexes of the colors. It can be other formats where knowing only the image data is not enough.
2. having a generic open function would assume that you return a pointer to buffer where the image data is stored. It also not true in every case. That's why `lv_img_decoder_read_line` exists. There can be image formats where you can return the required row of the image without decompressing the whole image. It saves time and memory.

So at first look, I think a generic open function does not fit very well - although the idea is logical. I  suggest writing this function out of LittlevGL.

Once you have the decompressed image you can really assign it to a canvas. For `v6.0` the canvas should receive some updates to support some transformation/effect. Rotation and blur are in my mind right now.

manison, 2019-03-11:
Thank you for comment, @kisvegabor.

> For example LV_IMG_CF_INDEXED_1/2/4/8BIT has a palette on the beginning of the image data and then only the indexes of the colors. It can be other formats where knowing only the image data is not enough.

I'm sorry I don't get it. If I'm reading the code directly, the `lv_img_dsc_get_info` reads the `lv_img_header_t` from the beginning of the file (in case of `LV_IMG_SRC_FILE`). Then the palette follows (in case `LV_IMG_CF_INDEXED_x`) and then there are image data. All this is hardcoded behavior for `LV_IMG_SRC_FILE`.

I believe that based on the the image source you have just a few options to get the corresponding `lv_img_header_t` (correct me if I'm wrong):
* the source points directly to `lv_img_header_t`
* the source is file, then you have to extract the `lv_img_header_t` from the file _somehow_ (currently it has to be at the beginning of the file)
* (the source is symbol)

All I want is to make the existing internal infrastructure provided by function `lv_img_decoder_open` public:
* it extracts `lv_img_header_t` (using `lv_img_dsc_get_info`), so the suggested function `lv_img_dsc_open` can return it to the caller
* it extracts the data of the image (possibly delegating the work to the decoder), so the suggested function `lv_img_dsc_open` can return it to the caller

> having a generic open function would assume that you return a pointer to buffer where the image data is stored. It also not true in every case. That's why lv_img_decoder_read_line exists. There can [...]

I know about the read line interface, but I don't think it is useful outside the LittlevGL. I only want public `lv_img_decoder_open`-ish interface. The return value of `NULL` or `LV_IMG_DECODER_OPEN_FAIL` is returned in case of failure.

I hope I explained my intentions in clearer way. I don't want to change the decoder interface (much), I only want to use the existing internal infrastructure and build public API on top of it.

kisvegabor, 2019-03-12:
I examined this issue again (in light of the points you mentioned). If it only means adding two new functions  (`lv_img_dsc_open/close`) which uses the internal static open/close functions then it seems fine.  
However, to avoid unnecessary memory allocation the interface of the open functions should be   
`void * lv_img_dsc_open(lv_img_dsc_t * dsc, void * src);` 

E.g.:
```c
lv_img_dsc_t my_dsc;
void * image_data = lv_img_dsc_open(&my_dsc, "dir/img.png");

...

lv_img_dsc_close(&my_dsc); 
```

What do you think?

manison, 2019-03-13:
> If it only means adding two new functions (lv_img_dsc_open/close) which uses the internal static open/close functions then it seems fine.

That's the plan :) I think the decoder's close callback will need to be changed to
```diff
-void (*lv_img_decoder_close_f_t)(void);
+void (*lv_img_decoder_close_f_t)(void *p);
```

so it can determine what decoded data to free.

> However, to avoid unnecessary memory allocation the interface of the open functions should be
void * lv_img_dsc_open(lv_img_dsc_t * dsc, void * src);

Could be. Since the destination `img_dsc` will contain the pointer to the decoded data (eventually), it is not necessary return that pointer, is it? In that case the return value can be simple `bool`.

kisvegabor, 2019-03-14:
> I think the decoder's close callback will need to be changed to...

As you pointed out below "Since the destination img_dsc will contain the pointer to the decoded data..." `dsc` can work too. I vote for `dsc` because it has some extra info (e.g. size) which might be useful.

However, I've just realized that `dsc` is not required at all in `close` because `lv_img_decoder_close_custom()` (the set custom close function) doesn't have arguments. It needs to close the previously opened image.

> Since the destination img_dsc will contain the pointer to the decoded data (eventually), it is not necessary return that pointer, is it? In that case the return value can be simple bool.

If we leave `read_line` then it really can return true/false.
And the `open` function should copy `decoder_header` to the `dsc` argument.

manison, 2019-03-14:
> However, I've just realized that dsc is not required at all in close because lv_img_decoder_close_custom() (the set custom close function) doesn't have arguments. It needs to close the previously opened image.

I'm lost :) So you are saying the decoder close callback stays as it is (with no argument at all) or you accept my suggestion to modify the close callback to accept the simple pointer? I think it is necessary because now the close callback frees the previous one, which is fine in internal image drawing infrastructure, but not OK once the interface is public because the decoded pointer can outlive another one:

```c
img_dsc_open(&dsc1, &src1);
img_dsc_open(&dsc2, &src2);

img_dsc_close(&dsc1); // which one this really frees?
img_dsc_close(&dsc2);
```

> If we leave read_line then it really can return true/false.
And the open function should copy decoder_header to the dsc argument.

Agreed.

kisvegabor, 2019-03-14:
> So you are saying the decoder close callback stays as it is (with no argument at all) or you accept my suggestion to modify the close callback to accept the simple pointer? I think it is necessary [...]

You are absolutely right with this. However, we need to assume that the opened files and other resources are freed/closed in `img_dsc_open` and only the allocated image data itself should be freed in `img_dsc_close`. Anyway, in general, it can work.

manison, 2019-03-14:
OK, great. I will start implementing it and will submit PR for initial review.
One more thing – I don't like the function names I suggested but can't think of anything better. Maybe just `lv_img_open`/`close`? Any ideas? Also should this go to _lv_draw_img.c_? In fact it does not draw anything…

kisvegabor, 2019-03-15:
> OK, great. I will start implementing it and will submit PR for initial review.

Great!

> One more thing – I don't like the function names I suggested but can't think of anything better. Maybe just lv_img_open/close? Any ideas? 

For first I'd choose `lv_img_src_open/close` to avoid confusion with the functions of `lv_img` object. But we can change it later if got a better idea. 

> Also should this go to lv_draw_img.c? In fact it does not draw anything…

Yes, because the internal open/close functions are static they are there.

manison, 2019-03-15:
I'm on it :)

manison, 2019-03-18:
@kisvegabor 
1. Why does the `lv_img_dsc_get_info` returns dummy header for file images when reading file fails? Wouldn't it be better to indicate the error to the caller?

https://github.com/littlevgl/lvgl/blob/1927376fe571a17fec6a5cca7a2abff63c35980c/src/lv_draw/lv_draw_img.c#L123-L128

2. I need to indicate in the returned `lv_img_dsc_t` that the data were
  * allocated on lvgl heap (when read data from file)
  * returned from the custom decoder

so the `lv_img_src_close` could free the resources appropriately.
There are two reserved bits in the `lv_img_header_t` which could be used to tag the header. Any other idea?

Masi2017, 2019-03-19:
Hi @kisvegabor @manison 
I was wondering if you have any jpeg decoder source code?

kisvegabor, 2019-03-20:
@manison
> Why does the lv_img_dsc_get_info returns dummy header for file images when reading file fails? Wouldn't it be better to indicate the error to the caller?

It has a [return value](https://github.com/littlevgl/lvgl/blob/1927376fe571a17fec6a5cca7a2abff63c35980c/src/lv_draw/lv_draw_img.c#L97) and it really should be used in this case too. The header still needs to be initialized to a known value in case of the user ignores the return value.

> I need to indicate in the returned lv_img_dsc_t that the data were...

If the image source is a file then it's not loaded to RAM but the `read_line` will be used later. 
https://github.com/littlevgl/lvgl/blob/1927376fe571a17fec6a5cca7a2abff63c35980c/src/lv_draw/lv_draw_img.c#L404-L409

So in `open` there are two possible outcomes:
1. A pointer to the data is returned directly because the image is stored in the flash
2. `lv_img_decoder_open_custom` opened the image somehow. It can use `lv_mem_alloc`, normal `malloc`, use a static buffer etc. 

So LittlevGL won't allocate memory to open the image. The close function should know how it was opened the image and free it accordingly. Have I missed something?

@Masi2017 I didn't try a JPEG library yet but should be similar to PNG. [This post](https://blog.littlevgl.com/2018-10-05/png_converter) summarizes how to integrate a PNG library

beibean, 2019-03-20:
Hi,

I just wanted to share with you the modifications I have done to LittlevGL in local to solve a problem I found.

The point is to use RLE for some images but not for every image. I have 4-bit indexed images and 4-bit alpha images (coded with RLE). The problems I found:
- When you use custom decoder LittlevGL expect a pointer to a buffer where the data is fully decoded to the final format. So, when decompressing 4-bit alpha images it expects RGB565 instead of a 4-bit alpha bitmap decompresed (buffer increases a lot).
- For my 4-bit indexed images will have the same problem, I need to decompress them to RGB565.
- LittlevGL is already in charge of the fully decompression depending on image type.

So, what I have added is a new intermediate decoder function pointer:

```C
/**
 * Open an image to convert its bitmap.
 * @param   src     the image source.
 * @param   style   the style of image (maybe it will be required to determine a color or something)
 * @return  pointer to image converted. Both bitmap and data_length could change (in RLE transformation)
 */
typedef const void * (*lv_img_decoder_converter_f_t)(const void * src, const lv_style_t * style);
```

Which can be added like the other decoder custom functions: `lv_img_decoder_set_custom`

And it is called at the very beginning of `lv_img_decoder_open`:

```C
static const uint8_t * lv_img_decoder_open(const void * src, const lv_style_t * style)
{
    decoder_custom = false;

    if (lv_img_convert_bitmap_to_custom) {
      src = lv_img_convert_bitmap_to_custom(src, style);
    }

    /*Try to open with the custom functions first*/
    if(lv_img_decoder_open_custom) {

[...]

[...]

```

Then, the converter_bitmap function works for me as follows:

- If the image type does not need to be converted the same src image pointer is returned.
- If the image needs some type of conversion, i fill a buffer, modify image properties and return a pointer to new modified global `lv_img_dsc_t decodedImage` variable.

This way, I can use different kinds of images with decoder (or not) and get advantage of image rendering of LittlevGL.

This is just FYI, I don't know if it's the best way to do it but works for me (I did it as quick as I was able).

manison, 2019-03-20:
@kisvegabor 
> It has a return value and it really should be used in this case too.

Yes, it has the return value. But in case the file operation fails, the return value is **[true](https://github.com/littlevgl/lvgl/blob/1927376fe571a17fec6a5cca7a2abff63c35980c/src/lv_draw/lv_draw_img.c#L144)**! (Should be `LV_RES_OK`/`INV` in fact.)

> …in case of the user ignores the return value.

If a user ignores an error return value and uses the result anyway then she is on her own and should burn in hell 😃

> If the image source is a file then it's not loaded to RAM but the read_line will be used later.

For drawing purpose sure. For purpose of `lv_img_src_open` the memory for entire image can be allocated, image can be read line by line from the file and then returned in `lv_img_dsc_t`.

> So in open there are two possible outcomes:

Should `lv_img_src_open` support files then three possible outcomes:

3. Memory allocated for the entire file and reading the file into the preallocated buffer.

> The close function should know how it was opened the image and free it accordingly. 

Exactly. How does the `lv_img_src_close` determine the scenario how to free the data provided it only has the `lv_img_dsc_t`?

kisvegabor, 2019-03-21:
@manison 

> Yes, it has the return value. But in case the file operation fails, the return value is true! 

Yes, it should return false. 

> (Should be LV_RES_OK/INV in fact.)

I agree.

> If a user ignores an error return value and uses the result anyway then she is on her own and should burn in hell smiley

Then I think hell is full of developers. If we can save some soul we should do. :smile: Using a default value more deterministic and costs nothing.

> For drawing purpose sure. For purpose of lv_img_src_open the memory for entire image can be allocated, image can be read line by line from the file and then returned in lv_img_dsc_t.

Sure, a PNG decoder will read the entire file and decompress it into image sized buffer. `read_line` is used with the simple internal formats.

> Should lv_img_src_open support files then three possible outcomes: 
3. Memory allocated for the entire file and reading the file into the preallocated buffer.

Once the image decoder decodes the image into an image-sized buffer the file buffer is not required. So the file buffer can be freed in open. So it's the same as 2).

> How does the lv_img_src_close determine the scenario how to free the data provided it only has the lv_img_dsc_t?

In the current situation (internal decoders) it's easy. Only one image can be opened at a time so the decoder can keep track with those single resources. But it's really not trivial if more images can be opened at the same time. A `void * user_data` in `lv_img_dsc_t` could solve it, right?

kisvegabor, 2019-03-21:
@beibean It's a good idea to leverage the internal formats. 
Maybe the prototype of `lv_img_decoder_open` and `lv_img_decoder_open_custom` can be modified to:
```c
lv_res_t lv_img_decoder_open_custom(src_in, src_out, style);
```  
To signal that t`read_line` should be used `src_out->data` can be set to `NULL` instead of returning `NULL` like in the current implementation.

manison, 2019-03-21:
@beibean @kisvegabor 
> Maybe the prototype of lv_img_decoder_open and lv_img_decoder_open_custom can be modified…

Isn't it enough (or shouldn't it) for the decoder to indicate the return format from its `lv_img_decoder_info_f_t`?

@kisvegabor 
> In the current situation (internal decoders) it's easy. Only one image can be opened at a time so the decoder can keep track with those single resources. But it's really not trivial if more images can be opened at the same time. A void * user_data in lv_img_dsc_t could solve it, right?

This would do without breaking anything:

```diff
typedef struct {
    /* The first 8 bit is very important to distinguish the different source types.
     * For more info see `lv_img_get_src_type()` in lv_img.c */
    uint32_t cf           :5;    /* Color format: See `lv_img_color_format_t`*/
    uint32_t always_zero  :3;    /*It the upper bits of the first byte. Always zero to look like a non-printable character*/

-   uint32_t reserved     :2;   /*Reserved to be used later*/
+   uint32_t alloc        :2;   /*How the image data is allocated*/

    uint32_t w:11;              /*Width of the image map*/
    uint32_t h:11;              /*Height of the image map*/
} lv_img_header_t;

+enum
+{
+    LV_IMG_ALLOC_STATIC,
+    LV_IMG_ALLOC_HEAP,
+    LV_IMG_ALLOC_DECODER
+} lv_img_alloc_t;
```

manison, 2019-03-21:
Note to self: https://github.com/littlevgl/lvgl/issues/934#issuecomment-470816814 is interesting idea of decoding images into a buffer without the public decoder API. Disadvantages: complex, big overhead.

beibean, 2019-03-21:
@kisvegabor
> Maybe the prototype of lv_img_decoder_open and lv_img_decoder_open_custom can be modified to [...]

@manison 
> Isn't it enough (or shouldn't it) for the decoder to indicate the return format from its lv_img_decoder_info_f_t?

As I said I just wanted to share with you the info and enhancement could have the library. I did not do it in the best way since I need it as fast as possible 🏃!

Said this, both of your propposals on how to implement this are OK for me. I am not the one to choose one or another since I am not experienced working with images within LittlevGL. Let it on your 🤝!

kisvegabor, 2019-03-22:
@manison 

> Isn't it enough (or shouldn't it) for the decoder to indicate the return format from its 

Decoded image's size should be known after it is opened. It will contain all the information which might be required.

What does `LV_IMG_ALLOC_DECODER` mean? the data can be static or in the heap. What is the third option?

I think a `user_data` would give more flexibility. You can have data in it to store the type of decoder (png, jpg) and anything else you need in your specific situation. 

@beibean 
> As I said I just wanted to share with you the info and enhancement could have the library. I did not do it in the best way since I need it as fast as possible running_man!

Anyway, thank you for the suggestion. We will keep in mind and try to make it easier to use.

manison, 2019-03-25:
@kisvegabor 

> Decoded image's size should be known after it is opened. It will contain all the information which might be required.

Which is in `lv_img_decoder_info_f_t`, isn't it?

> What does LV_IMG_ALLOC_DECODER mean? the data can be static or in the heap. What is the third option?

It means that the data are provided by a custom decoder _somehow_. It might be the LittlevGL heap or it might not. It indicates that the data must be freed by the custom decoder `lv_img_decoder_close_f_t`. For example in my scenario I use [LodePNG](https://github.com/lvandeve/lodepng), but I implement `lodepng_malloc`/`realloc`/`free` in heap separate from lvgl's heap.

> I think a user_data would give more flexibility. You can have data in it to store the type of decoder (png, jpg) and anything else you need in your specific situation.

More flexibility for whom? For the custom decoder maybe (then we should probably support `LV_USE_USER_DATA_SINGLE`/`MULTI` concept). Anyway `lv_img_src_open`/`close` needs to know this information in "fixed" (i.e. non-user) way. Also the majority of `lv_img_dsc_t` is static const so they couldn't take advantage of the user_data.

kisvegabor, 2019-03-26:
I'm starting to get lost little bit :) 
Can we continue in a PR where we could see the suggestions better?

manison, 2019-03-26:
Very well. One line of code is worth a thousand words 😃

kisvegabor, 2019-03-26:
Sometimes yes :smile:

kisvegabor, 2019-04-10:
@manison Do you still planning to add this improvement?  
As it's a more complicated topic as it seemed at first I understand if you just modify the lib as you need.

manison, 2019-04-11:
> Do you still planning to add this improvement?

I'd like to but some other things boosted in priority and I need to put this one on hold.
I even did some initial work but since the recent turbulent changes in the repository they are not mergeable right now.

kisvegabor, 2019-04-11:
I see, no problem. It's not urgent. `v6.0` is planned for June, hopefully, you will fond sometime until that. :)