lvgl/lvgl #928: Please advise how to handle the group objects in the tabview

TMarkivULCRobotics, 2019-03-05:
Could you please review the code and say if its normal way of the handling of tabs. My problem with the code is that:
- the sometime the tabview and some of the objects in the tab are highlighted at the same time.
- every text area have blinking cursor at the begging. After text  area was selected and then deselected, then the cursor is shown only on the text area object that is focused.

```
lv_group_t * group0;
lv_group_t * group1;
lv_group_t * group2;

// keypad_get_group this function returns currect group that is set up in the indev structure.
// keypad_update(group0) function sets the group object in the indev structure.

lv_group_t * current_group;
void keypad_update(lv_group_t * keypad_group)
{
    lv_indev_set_group(kp_indev, keypad_group);
    current_group = keypad_group;
}
lv_group_t *  keypad_get_group(void)
{
    return current_group;
}

lv_res_t tab_loaded(lv_obj_t *tabview, uint16_t tab_id)
{
    switch(tab_id)
    {
    case 0:
    {
        /* Freeze previous group */
        lv_group_focus_freeze(keypad_get_group(), true);
        /* Unfreeze current group */
        lv_group_focus_freeze(group0, false);
        /* Put new group into the indev structure */
        keypad_update(group0);
    }break;
    case 1:
    {
        lv_group_focus_freeze(keypad_get_group(), true);
        lv_group_focus_freeze(group1, false);
        keypad_update(group1);
    }break;
    case 2:
    {
        lv_group_focus_freeze(keypad_get_group(), true);
        lv_group_focus_freeze(group2, false);
        keypad_update(group2);
    }break;
    default: break;
    }

    return LV_RES_OK;
}

{
    /* Create a tabview object */
    lv_obj_t * tabs = lv_tabview_create(lv_scr_act(), NULL);
    lv_tabview_set_anim_time(tabs, 0);
    lv_tabview_set_sliding(tabs, false);
    lv_tabview_set_tab_load_action(tabs, tab_loaded);
    lv_obj_align(tabs, btnm1, LV_ALIGN_OUT_BOTTOM_MID,0,0);
    /* Add 3 tabs into the tabview object */
    lv_obj_t * tab0   = lv_tabview_add_tab(tabs, "tab0");
    lv_obj_t * tab1   = lv_tabview_add_tab(tabs, "tab1");
    lv_obj_t * tab2   = lv_tabview_add_tab(tabs, "tab2");

    /* Create, align and set text for the 3 text areas in each tab */
    lv_obj_t * ta00 = lv_ta_create(tab0,NULL);
    lv_obj_t * ta01 = lv_ta_create(tab0,NULL);
    lv_obj_t * ta02 = lv_ta_create(tab0,NULL);
    lv_obj_align(ta00, NULL, LV_ALIGN_IN_TOP_LEFT,0,0);
    lv_obj_align(ta01, NULL, LV_ALIGN_IN_TOP_MID,0,0);
    lv_obj_align(ta02, NULL, LV_ALIGN_IN_TOP_RIGHT,0,0);
    lv_ta_set_text(ta00,"ta00");
    lv_ta_set_text(ta01,"ta01");
    lv_ta_set_text(ta02,"ta02");

    /* Objects for the second tab */
    lv_obj_t * ta10 = lv_ta_create(tab1,NULL);
    lv_obj_t * ta11 = lv_ta_create(tab1,NULL);
    lv_obj_t * ta12 = lv_ta_create(tab1,NULL);
    lv_obj_align(ta10, NULL, LV_ALIGN_IN_TOP_LEFT,0,0);
    lv_obj_align(ta11, NULL, LV_ALIGN_IN_TOP_MID,0,0);
    lv_obj_align(ta12, NULL, LV_ALIGN_IN_TOP_RIGHT,0,0);
    lv_ta_set_text(ta10,"ta10");
    lv_ta_set_text(ta11,"ta11");
    lv_ta_set_text(ta12,"ta12");

    /* Objects for the third tab */
    lv_obj_t * ta20 = lv_ta_create(tab2,NULL);
    lv_obj_t * ta21 = lv_ta_create(tab2,NULL);
    lv_obj_t * ta22 = lv_ta_create(tab2,NULL);
    lv_obj_align(ta20, NULL, LV_ALIGN_IN_TOP_LEFT,0,0);
    lv_obj_align(ta21, NULL, LV_ALIGN_IN_TOP_MID,0,0);
    lv_obj_align(ta22, NULL, LV_ALIGN_IN_TOP_RIGHT,0,0);
    lv_ta_set_text(ta20,"ta20");
    lv_ta_set_text(ta21,"ta21");
    lv_ta_set_text(ta22,"ta22");

    /* Create group for each tab in the tabview */
    group0 = lv_group_create();
    group1 = lv_group_create();
    group2 = lv_group_create();

    /* Initialize each group with appropriate objects */
    lv_group_add_obj(group0, tabs);
    lv_group_add_obj(group0, ta00);
    lv_group_add_obj(group0, ta01);
    lv_group_add_obj(group0, ta02);

    lv_group_add_obj(group1, tabs);
    lv_group_add_obj(group1, ta10);
    lv_group_add_obj(group1, ta11);
    lv_group_add_obj(group1, ta12);

    lv_group_add_obj(group2, tabs);
    lv_group_add_obj(group2, ta20);
    lv_group_add_obj(group2, ta21);
    lv_group_add_obj(group2, ta22);

    /* This is neccessary bacause in the callback I set the keypad group. */
    lv_tabview_set_tab_act(tabs, 0, false);
}
```

embeddedt, 2019-03-05:
> the sometime the tabview and some of the objects in the tab are highlighted at the same time.

@kisvegabor will know more about this.

> every text area have blinking cursor at the begging. After text area was selected and then deselected, then the cursor is shown only on the text area object that is focused.

Try calling `lv_ta_set_cursor_type(ta, LV_CURSOR_HIDDEN);` on each text area.

TMarkivULCRobotics, 2019-03-05:
Thank you, the following function call does the work i needed:
`lv_ta_set_cursor_type(ta, LV_CURSOR_HIDDEN|LV_CURSOR_LINE);`

kisvegabor, 2019-03-06:
@TMarkivULCRobotics 
I'd kindly check the code but please move your question to the [Forum](https://forum.littlevgl.com/) and continue there.

closed 2019-03-06 by kisvegabor