lvgl/lvgl #929: How can generate font in  same width size

crazyuav, 2019-03-06:
Hello kisvegabor,
Thanks for your great lib!

I try to generate font using [https://littlevgl.com/ttf-font-to-c-array](url),but font width(example:0-9) is diffrent. 

There is any way can generate same width font?

Best Regards,

seyyah, 2019-03-06:
You can use the `Monospace` parameters,

> Optional for monospace fonts specify width [px]

crazyuav, 2019-03-08:
It works.Thanks!

closed 2019-03-08 by crazyuav

AloyseTech, 2020-01-29:
Hi @kisvegabor ,
Sorry to comment on this old issue, but the monospace parameter is not available with the latest online TTF font converter. Is there any plan to add it?

embeddedt, 2020-01-29:
As far as I know, if you convert a monospaced font, it will automatically be handled (and displayed in monospace). I don't think you have manual control over the setting anymore, because treating a regular font as monospace won't look right on a screen.

(cc @puzrin)

puzrin, 2020-01-29:
There is NO way to CORRECTLY convert proportional font to monospace. New convertor produces result exactly as source designed. If you need monospace output - use monospace input.

The only "right" way to get predictable result is to find proper TTF font as source. That's possible, including case of pixel-perfect TTF fonts, designed for specific size.

AloyseTech, 2020-01-29:
I often use regular font to render digits only, but in a monospaced format. Monospaced font with good looking digits are rare IMO.

Thanks anyway :)

embeddedt, 2020-01-29:
It might be possible to modify your regular font so that all the digits have an equal width. That way, it really would be a monospaced font and the converter would treat it as such.

puzrin, 2020-01-29:
By historic reason, font conversion had many kludges, caused by missunderstanid how font renderers work. I understand pain of development process change - nobody likes that. But it really worth to drop all those ancient magic in favour of more modern and correct font generation.

Of cause, you can edit digits in fontforge, but i think finding nice monospase font will take less time.

kisvegabor, 2020-01-30:
@AloyseTech 
I've found cool monospaced digit fonts here: https://www.1001fonts.com/monospaced-fonts.html?page=1&items=10

I hope it helps.