lvgl/lvgl #932: Canvas fill bug

chenshijianworkgit, 2019-03-07:
Hi，
In the lv_test_canvas_2 test function I found that some fill functions do not take effect.

For example, filling a circle in the "b. Draw complex objects" test has no effect. After I exchanged the last two parameters, the fill color will be red. However, the color of the circle filled with lv_canvas_flood_fill in "h. Draw circle" does not take effect.

Then fill in the rectangle color with lv_canvas_fill_polygon in "d. Draw rect/fill" and it doesn't take effect.

The first triangle filled with the lv_canvas_fill_polygon function in "e. Draw triangle/fill" does not take effect, and the second triangle takes effect.

In "f. Draw lines", the triangles enclosed by three points are then connected by lines, and the effect is not filled with the lv_canvas_flood_fill function.

chenshijianworkgit, 2019-03-07:
The problem is in the latest master branch

seyyah, 2019-03-07:
Can you share code running on the pc_simulator_sdl_eclipse?

chenshijianworkgit, 2019-03-08:
> Can you share code running on the pc_simulator_sdl_eclipse?

The code is the official latest master branch code

embeddedt, 2019-03-08:
@chenshijianworkgit Please create a small, reproducable example that can easily be tested.

chenshijianworkgit, 2019-03-08:
@embeddedt 

The test code is as follows, it has been marked in the code which fill does not take effect；

The actual effect chart is as follows：

[image]

`void lv_test_canvas_2(void)
{
    lv_color_t bg_color = LV_COLOR_MAKE(0xec, 0xf0, 0xf1);
    lv_color_t fill_color = LV_COLOR_MAKE(0xbc, 0xa0, 0xa1);

    /*Create a buffer for the canvas set its pixels to light blue*/
    static lv_color_t cbuf[sizeof(lv_color_t) * 100 * 80];
    uint32_t i;
    for(i = 0; i < sizeof(cbuf) / sizeof(cbuf[0]); i++) {
        cbuf[i] = bg_color;
    }

    /*Create the canvas object*/
    lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);

    /*Assign the buffer to the canvas*/
    lv_canvas_set_buffer(canvas, cbuf, 100, 80, LV_IMG_CF_TRUE_COLOR);

    // a. Draw image
    LV_IMG_DECLARE(img_flower_icon);
    lv_canvas_copy_buf(canvas, img_flower_icon.data, img_flower_icon.header.w, img_flower_icon.header.h, 0, 0);

    // b. Draw complex objects
    lv_point_t points5[] = {
      {.x=10, .y=10},
      {.x=40, .y=10},
      {.x=30, .y=40},
      {.x=20, .y=40},
    };

    lv_canvas_draw_polygon(canvas, points5, 4, LV_COLOR_RED);
    lv_canvas_fill_polygon(canvas, points5, 4, LV_COLOR_RED, fill_color);

    lv_canvas_draw_circle(canvas, 25, 54, 10, LV_COLOR_BLACK);
    lv_canvas_flood_fill(canvas, 25, 54, bg_color, LV_COLOR_RED);      **//Fill circle does not take effect**

    lv_canvas_draw_circle(canvas, 25, 54, 5, LV_COLOR_BLACK);
    lv_canvas_draw_circle(canvas, 25, 54, 15, LV_COLOR_BLACK);

    // c. Draw polygon/fill
    lv_point_t points4[] = {
      {.x=10, .y=10},
      {.x=40, .y=10},
      {.x=30, .y=40},
      {.x=20, .y=40},
    };

    lv_canvas_draw_polygon(canvas, points4, 4, LV_COLOR_RED);
    lv_canvas_fill_polygon(canvas, points4, 4, LV_COLOR_RED, fill_color);

    // d. Draw rect/fill
    lv_point_t points3[4] = {
      {.x=10, .y=10},
      {.x=50, .y=10},
      {.x=50, .y=50},
      {.x=10, .y=50}
    };

    lv_canvas_draw_rect(canvas, points3, LV_COLOR_RED);
    lv_canvas_fill_polygon(canvas, points3, 4, LV_COLOR_RED, fill_color);

    // e. Draw triangle/fill   
     **//Filled rectangle does not take effect**
    lv_point_t points1[3] = {
      {.x=50, .y=10},
      {.x=30, .y=50},
      {.x=70, .y=50}
    };

    **//Filling the first triangle does not take effect, filling the second triangle takes effect**
    lv_canvas_draw_triangle(canvas, points1, LV_COLOR_RED);
    lv_canvas_fill_polygon(canvas, points1, 3, LV_COLOR_RED, fill_color);    

    lv_point_t points2[3] = {  
      {.x=80, .y=5},
      {.x=50, .y=30},
      {.x=99, .y=79}
    };
    lv_canvas_draw_triangle(canvas, points2, LV_COLOR_BLUE);
    lv_canvas_fill_polygon(canvas, points2, 3, LV_COLOR_BLUE, fill_color);

    // f. Draw lines 
    lv_point_t point1 = {.x=50, .y=10}, point2 = {.x=30, .y=50}, point3 = {.x=70, .y=50};
    lv_canvas_draw_line(canvas, point1, point2, LV_COLOR_RED);
    lv_canvas_draw_line(canvas, point1, point3, LV_COLOR_RED);
    lv_canvas_draw_line(canvas, point3, point2, LV_COLOR_RED);
    lv_canvas_flood_fill(canvas, 50, 30, LV_COLOR_RED, bg_color);    **//Fill does not take effect**

    // h. Draw circle
    lv_canvas_draw_circle(canvas, 50, 40, 15, LV_COLOR_BLACK);
    lv_canvas_flood_fill(canvas, 50, 40, LV_COLOR_RED, bg_color);    **//Fill does not take effect**

    lv_canvas_draw_circle(canvas, 50, 40,  5, LV_COLOR_BLACK);
    lv_canvas_draw_circle(canvas, 50, 40, 10, LV_COLOR_BLACK);
    lv_canvas_draw_circle(canvas, 50, 40, 20, LV_COLOR_BLACK);
    lv_canvas_draw_circle(canvas, 50, 40, 25, LV_COLOR_BLACK);
}`

embeddedt, 2019-03-11:
cc @kisvegabor

seyyah, 2019-03-12:
Can you test only this code part? @chenshijianworkgit 

```c
void lv_test_canvas_2(void)
{
  lv_color_t bg_color = LV_COLOR_MAKE(0xec, 0xf0, 0xf1);
  lv_color_t fill_color = LV_COLOR_MAKE(0xbc, 0xa0, 0xa1);

  /*Create a buffer for the canvas set its pixels to light blue*/
  static lv_color_t cbuf[sizeof(lv_color_t) * 100 * 80];
  uint32_t i;
  for(i = 0; i < sizeof(cbuf) / sizeof(cbuf[0]); i++) {
      cbuf[i] = bg_color;
  }

  /*Create the canvas object*/
  lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);

  /*Assign the buffer to the canvas*/
  lv_canvas_set_buffer(canvas, cbuf, 100, 80, LV_IMG_CF_TRUE_COLOR);

  lv_canvas_draw_circle(canvas, 25, 54, 10, LV_COLOR_BLACK);
  lv_canvas_flood_fill(canvas, 25, 54, bg_color, LV_COLOR_RED); // Fill circle does not take effect
}
```

seyyah, 2019-03-12:
We discuss this issue at https://github.com/littlevgl/lv_examples/pull/19#issuecomment-458432459

> @kisvegabor I tried to make similar to obj_mem_leak_tester. But I got only empty canvas area. Consecutive drawings did not make.

chenshijianworkgit, 2019-03-12:
@seyyah 
Test the example you provided above as follows，fill does not take effect：
[image]

seyyah, 2019-03-12:
> ```c
> lv_canvas_flood_fill(canvas, 25, 54, bg_color, LV_COLOR_RED); // Fill circle does not take effect
> ```

Parameter order is wrong. The corrected version is below

```c
lv_canvas_flood_fill(canvas, 25, 54, LV_COLOR_RED, bg_color);
```

Result is

[image]

chenshijianworkgit, 2019-03-12:
This can be done after the parameters have been changed. This was the case in my previous example.

seyyah, 2019-03-12:
Test code part is

```c
lv_point_t point1 = {.x=50, .y=10}, point2 = {.x=30, .y=50}, point3 = {.x=70, .y=50};
lv_canvas_draw_line(canvas, point1, point2, LV_COLOR_RED);
lv_canvas_draw_line(canvas, point1, point3, LV_COLOR_RED);
lv_canvas_draw_line(canvas, point3, point2, LV_COLOR_RED);
lv_canvas_flood_fill(canvas, 50, 30, LV_COLOR_RED, bg_color); 
```

Result is

[image]

chenshijianworkgit, 2019-03-12:
It is ok to fill a circle and a triangle separately on a blank canvas. In the example, some areas are superimposed, and the picture cannot be drawn at this time.

seyyah, 2019-03-12:
> Parameter order is wrong. The corrected version is below
> 
> ```c
> lv_canvas_flood_fill(canvas, 25, 54, LV_COLOR_RED, bg_color);
> ```

This part is resolved by https://github.com/littlevgl/lv_examples/commit/990fee6c35843294b3b0562c80fcdecb41a17633

chenshijianworkgit, 2019-03-12:
You can change the parameters in the above example where there are comments in the examples that can't be displayed to see if there is any effect.

seyyah, 2019-03-12:
> It is ok to fill a circle and a triangle separately on a blank canvas. In the example, some areas are superimposed, and the picture cannot be drawn at this time.

Surely. The flood fill algorithm is based on the connected nodes have bg_color.

> Flood fill (also known as seed fill) is an algorithm that determines the area connected to a given node in a multi-dimensional array.

## :clipboard: Reference
1. https://www.techiedelight.com/flood-fill-algorithm/

kisvegabor, 2019-03-13:
FYI, in `v6.0` you will be able to draw LittlevGL object directly to the canvas with (opacity, anti-aliasing etc). Although LittlevGL doesn't support polygon drawing yet.

kisvegabor, 2019-03-24:
It seems @seyyah explained well this issue occurs so I close this issue.
If you have further comments feel free to reopen it.

closed 2019-03-24 by kisvegabor