lvgl/lvgl #933: Tileview object related questions

chenshijianworkgit, 2019-03-07:
Hi，
The tileview object depends on the page object implementation, but there is no macro judgment in the lv_tileview.h header file.

In the lv_test_tileview_1 test function, without changing the contents of the vp array, using #if0 #endif to expand the created objects one by one will find many interesting problems. I don't know if this is a usage problem or a bug.

For example, when only p01 and p11 objects are created, you can switch from p01 to p11, but you cannot switch from p11 to p01.

After creating the p01, p11, list12 objects, after switching to the list12 object interface, you can slide to the left blank page.

After creating the p01, p11, list12, p10 objects, you cannot slide to the left blank page after switching to the list12 object interface. However, when sliding from p01 to the right for the first time, it will slide to p10 instead of p11, and then switch between p01 and p11 when switching back and forth.

After creating the p01, p11, list12, p10, p21 objects, you can slide to the blank page on the right after switching to the list12 object interface or slide down to the blank page after switching to p21, afterwards whether it is left or up Can't recover.

chenshijianworkgit, 2019-03-07:
The problem is in the latest master branch。

If you mask out the last line of lv_tile_tileview_1 function lv_tileview_set_tile_act, you can drag to the top blank page or the blank page below for the first time.

embeddedt, 2019-03-07:
@kisvegabor

kisvegabor, 2019-03-08:
It's because when you create e.g. only the first two tiles `vp` doesn't make sense. It describes the whole tileview with the 6 tiles of the example. For example, if you return after the creation of the second tile, the tileview will think it has only one row and will interpret `vp` accordingly.

I see that it's not intuitive but I have a possible improvement in my mind. [set_valid_drag_dirs](https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_tileview.c#L559) can be improved to check the children of the tileview and if there is a child on a neighbor tile then allow scrolling there.

chenshijianworkgit, 2019-03-12:
Yes, in the complete case, the last line of lv_tileview_set_tile_act function is masked, then you can drag to the space above or below.

kisvegabor, 2019-03-13:
If you don't call `lv_tileview_set_tile_act` LittlevGL will think that you are on the `(0;0)` tile. So you should set valid tile when you create the tileview.

chenshijianworkgit, 2019-03-13:
But there is no explicit (0,0) in the array vp. Is this a bug?

kisvegabor, 2019-03-14:
> But there is no explicit (0,0) in the array vp. Is this a bug?

When you create the `tileview`, in lack of better idea, LittlevGL sets the (0;0) tile as active. But when you set the valid positions LittlevGL can see that an invalid tile is selected but still doesn't know which tile to choose. So you need to set tile explicitly with   `lv_tileview_set_tile_act`.

chenshijianworkgit, 2019-03-14:
That is to say, the lv_tileview_set_tile_act function must be called when using the tileview object.

It would be more reasonable to let the lvgl library internally select a valid tile page without the user calling the lv_tileview_set_tile_act function function.

chenshijianworkgit, 2019-03-14:
Still let the user call it himself, this is more clear

chenshijianworkgit, 2019-03-14:
Ok, thank you for your answer.

chenshijianworkgit, 2019-03-14:
For the novice to use tileview two points, if the array does not match the actual creation of the object may be wrong, the other is necessary to call lv_tileview_set_tile_act function to activate a page.

embeddedt, 2019-03-14:
@chenshijianworkgit For future reference, it's easier to read one comment rather than four. Please edit additional information into your existing comment.

kisvegabor, 2019-03-15:
@chenshijianworkgit 
I fixed it in `dev-6.0`. In `lv_tileview_set_valid_positions` if an invalid tile is shown then the first element of the `valid_positions` array is selected.

https://github.com/littlevgl/lvgl/blob/dev-6.0/lv_objx/lv_tileview.c#L164

closed 2019-03-15 by kisvegabor