lvgl/lvgl #934: Character/text on canvas

Pablo2048, 2019-03-07:
When looking at lv_canvas example here https://github.com/littlevgl/lv_examples/blob/master/lv_tests/lv_test_objx/lv_test_canvas/lv_test_canvas.c I found that we can not put characters/text to canvas. It seems like there are no functions like `lv_canvas_letter()`. Is this for some reason or there is plan to allow writting text to the canvas with another method?

embeddedt, 2019-03-07:
@Pablo2048 I think the canvas is meant for displaying raw images. Use a label if you want to display text.

Pablo2048, 2019-03-07:
Unfortunately I can't use just label. If the image rotation will be implemented only on canvas there is no way to do rotated text or char glyph.

embeddedt, 2019-03-07:
@Pablo2048 I see. In that case let's get @kisvegabor's opinion about adding this feature.

kisvegabor, 2019-03-08:
Hi,

I found a great trick to solve it. IT will leverage the multi-display support of `v6.0`. The concept:
1. Create a small e.g. `50x50` secondary display.
2. Create a label on it and set its text
3. Force LittelvGL to redraw the screens (`lv_refr_now()`)
4. Delete the display. The display buffer will still contain the rendered image
5. Rotate/modify the buffer with the letter as you need
6. Assign the buffer to canvas 

It might seem complicated at first but it's a very powerful tool. This way use LittlevGL built-in drawing functions to create an image run-time. 

Code example:
```c
    static lv_disp_buf_t disp_buf2;
    lv_disp_buf_init(&disp_buf2, dbuf2, NULL, 100*50);

   /*Create an other display*/
    lv_disp_drv_init(&disp_drv);            /*Basic initialization*/
    disp_drv.hor_res = 100;
    disp_drv.ver_res = 50;
    disp_drv.buffer = &disp_buf2;
    disp_drv.flush_cb = my_disp_flush;   
    lv_disp_t * disp2 = lv_disp_drv_register(&disp_drv);

    lv_disp_set_default(disp2);

    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    lv_obj_set_pos(label, 10, 10);

    lv_refr_now();
    lv_disp_remove(disp2);

    lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);
    lv_canvas_set_buffer(canvas, dbuf2, 100, 50, LV_IMG_CF_TRUE_COLOR);
```

Pablo2048, 2019-03-08:
Great. Thank You - I'll give it a try...

kisvegabor, 2019-03-24:
@Pablo2048 Had you got a chance to try it? Or did you find an other solution to your "rotating letters" issue?

Pablo2048, 2019-03-25:
@kisvegabor Unfortunately I didn't have free time to test it until now. I'll try this until end of this week and let you know the result.

kisvegabor, 2019-03-25:
Great, thank you!

closed 2019-05-02 by stale[bot]