lvgl/lvgl #935: Python bindings

rreilink, 2019-03-07:
Next to the Micropython bindings, there should also be bindings for 'regular' Python. The goal should be to have the same API for Micropython and regular Python.

embeddedt, 2019-03-07:
cc @kisvegabor

kisvegabor, 2019-03-08:
As discussed earlier @rreilink would like to rely more on naming conventions. For example it should be known that a paramter is saved by LittelvGL or not. What other places of the library should be improved to aid Python binding generation?

rreilink, 2019-03-10:
One other issue that I have come across is the inconsistent naming of enum types / members `lv_hal_indev_type_t` and `lv_preloader_type_t` . If we want to annotate setters/getters with the enum type that they expect/return, this consistency is desirable.

**lv_hal_indev_type_t**

```c
enum {
    LV_INDEV_TYPE_NONE,        /*Show uninitialized state*/
    LV_INDEV_TYPE_POINTER,     /*Touch pad, mouse, external button*/
    LV_INDEV_TYPE_KEYPAD,      /*Keypad or keyboard*/
    LV_INDEV_TYPE_BUTTON,      /*External (hardware button) which is assinged to a specific point of the screen*/
    LV_INDEV_TYPE_ENCODER,     /*Encoder with only Left, Right turn and a Button*/
};
typedef uint8_t lv_hal_indev_type_t;
```
This should be renamed to `lv_indev_type_t` or the members renamed to `LV_HAL_INDEV_TYPE_xxx` (the former has the advantage of not requiring changes to code that uses lvgl).

**lv_preloader_type_t**
```c
enum {
    LV_PRELOAD_TYPE_SPINNING_ARC,
    LV_PRELOAD_TYPE_FILLSPIN_ARC,
};
typedef uint8_t lv_preloader_type_t;
```
This should be renamed to `lv_preload_type` or the members renamed to `LV_PRELOADER_xxx`.

rreilink, 2019-03-11:
@kisvegabor two other issues that IMHO could be improved:
1. lv_btnm_set_map should be changed to take a char* instead of char**. (Terminating each item with '\x00' and terminating the list with '\x00\x00' as discussed earlier in #557)

2. lv_ddlist_get_selected_str should have a way to specify the buffer size to prevent buffer overflow (this is not Python specific). Could be like `snprintf`: provide the buffer size in the call, and return the length of the item as return value.

kisvegabor, 2019-03-12:
@rreilink do you have a script to list all the enum incompatibilities?

> lv_btnm_set_map should be changed to take a char* instead of char**. (Terminating each item with '\x00' and terminating the list with '\x00\x00' as discussed earlier in #557)

I'm considering it for a while and will try it soon in v6.0.

> lv_ddlist_get_selected_str should have a way to specify the buffer size to prevent buffer overflow (this is not Python specific). Could be like snprintf: provide the buffer size in the call, and return the length of the item as return value.

It seems like an easy and reasonable improvemnt.

amirgon, 2019-03-12:
@rreilink @kisvegabor @embeddedt   

On #557 we started a discussion about Python bindings and how to integrate Python and Micropython bindings.   
I agree, as @rreilink says, that the goal should be to have the same API for Micropython and regular Python. But I don't think this should be done "at all cost".  
The question is how to achieve this and also keep the bindings **General** and **Forward Compatible**.  

My opinion is that the bindings script should know **as little as possible** about the **specifics** of lvgl: which specific functions/enums/globals/structs it contains and what is the meaning of each.  
Instead, the binding script should build the bindings solely according to **conventions**. Such conventions would allow the script to know, for example, what are the "classes", their hierarchies, their members, etc. They would allow the script to know what a "struct" is, but not care it this struct represents a style, a theme or a driver handler.    
That way, when lvgl changes, the binding is kept compatible. Even if lvgl forks, or users decide to make their own changes to their local version of lvgl, the binding should still work.  
In fact, if someone decides to use the same conventions and build an entirely different library, the binding should, at least theoretically, still work for his new library, as long as he sticks the same conventions.

Today, micropython binding is designed that way. I think the Python binding is not, and contains specifics of lvgl hard coded in the binding script itself, instead of relying solely on conventions.  

Some examples:

- `lv_style_t` is predefined as `Style_Object` instead of deducing it from the struct definition.
- Fonts are predefined instead of deducing them from the struct definition.
- `lv_coord_t` is predefined as `short int` instead of deducing it from the typedef
- Only 16 bits color depth is supported
- Callback handling is predefined, for example btn_set_action, instead of handling generic callback
- Display and input drivers are predefined

I think it makes sense to continue this discussion now, when we consider having the same API for Python and Micropython, and maybe even having the same script to generate them both.

So here are some questions I have for @rreilink in this context:

- Do you plan to support generic structs and globals? This would cover styles fonts etc. without the need to hard-code them into the binding script. This is supported in Micropython binding today.
- Do you plan to support generic callbacks? I have a limited support for this today (one callback per object), and planning full support for v6.0.
- Do you plan to support generic driver registration? This is supported in Micropython today and I have several "plug-in" drivers ready. I'm also planning to support pure-python drivers.
- We had a discussion about preprocessor. Do you plan to use a standard preprocessor or rely on something else? Are you performing custom parsing of the sources or only parse the preprocessed code? As we previously discussed, I'm against source level parsing since it's not a generic solution, and prefer parsing prerpocessesd code.
- How do you support symbols in a generic way? On Micropython I'm using numstr technique to convert enum values to 32bit long string values.  
- How is memory managed? When are structs allocated and deallocated? To solve this for Micropython, I've enable micropython's Garbage-Collection mechanism on lvgl so it becomes very easy and "pythonic" for the user to create and use structs, without the need to specifically allocate/deallocate them, and without the need to keep track of reference counting.

rreilink, 2019-03-12:
>So here are some questions I have for @rreilink in this context:

> * Do you plan to support generic structs and globals? This would cover styles fonts etc. without the need to hard-code them into the binding script. This is supported in Micropython binding today.

Yes.
> * Do you plan to support generic callbacks? I have a limited support for this today (one callback per object), and planning full support for v6.0.

Yes. The current implementation is like this to support more than one callback per object (which I needed in a project). If v6.0 of lvgl can support registering an additional `void *` or the like when registering an action, all this stuff is no longer needed, and both MicroPython and Python bindings can use the same solution.

> * Do you plan to support generic driver registration? This is supported in Micropython today and I have several "plug-in" drivers ready. I'm also planning to support pure-python drivers.

Yes

> * We had a discussion about preprocessor. Do you plan to use a standard preprocessor or rely on something else? Are you performing custom parsing of the sources or only parse the preprocessed code? [...]

I plan to use the 'standard' preprocessor as that was the consensus in #557. I have already switched to it (not yet committed/pushed). I did run into an issue that `cpp` [misbehaves in Mac OS X](https://bugs.llvm.org/show_bug.cgi?id=31886), but `cc -E`  surely worked.

> * How do you support symbols in a generic way? On Micropython I'm using the numstr technique to convert enum values to 32bit long string values.

I believe for the colors and symbols, custom parsing of those specific include files outweighs the loopholes that one needs to jump through if only using the preprocessed source. The numstr technique increased lv_symbols.h from to 140 to 207 lines, having two lines on different places for each line, and [introduced a compile issue](https://github.com/littlevgl/lvgl/commit/19ea8ac55f79cd6b557348dc0001e3e3fbf4e6f8).

As discussed in #557, and as @kisvegabor stated:
> The new evidence was that we couldn't find a good way to add the colors and the related macros without source code parsing. In light of that, it seemed to me, the source code parsing need to introduced for colors. And once we have source code parsing it should be easy to add it for the symbols too which are quite similar. 

> * How is memory managed? When are structs allocated and deallocated? To solve this for Micropython, I've enabled micropython's Garbage-Collection mechanism on lvgl so it becomes very easy and "pythonic" for the user to create and use structs, without the need to specifically allocate/deallocate them, and without the need to keep track of reference counting.

They will have to be reference-counted. That's the way things are in 'regular' Python. The good new is, with generic structs, this only needs to be implemented once.

Regarding your other remarks:
> lv_style_t is predefined as Style_Object instead of deducing it from the struct definition.
> Font type is predefined instead of deducing it from the struct definition.
> lv_coord_t is predefined as short int instead of deducing it from the typedef

Style_Object and the Font type can go away when the generic structs are implemented

> Only 16 bits color depth is supported

This is only a quirck of the way I hacked the framebuffer in there, this should obviously go away with the support for registering a display driver.

> Callback handling is predefined, for example btn_set_action, instead of handling generic callback

As mentioned, if lvgl 6.0 supports an additional argument for the callbacks, this can be made generic.

> Display and input drivers are predefined

As mentioned, will be made generic. I think for desktop-applications it makes sense to (maybe as a compile-time option) supply a framebuffer implementation that can be registered as the display via the lvgl driver registration way.

kisvegabor, 2019-03-13:
> Yes. The current implementation is like this to support more than one callback per object (which I needed in a project). If v6.0 of lvgl can support registering an additional void * or the like when [...]

The current implementation of generic events (it replaces the obejct specific actions) for `v6.0` look like [this](https://github.com/littlevgl/lvgl/blob/dev-6.0/lv_core/lv_obj.h#L204). Just to be sure, is it good for Python and Micropython too?

amirgon, 2019-03-13:
> The current implementation of generic events (it replaces the obejct specific actions) for `v6.0` look like [this](https://github.com/littlevgl/lvgl/blob/dev-6.0/lv_core/lv_obj.h#L204). Just to be sure, is it good for Python and Micropython too?

@kisvegabor Looks good to me.
We would need to define `LV_USE_USER_DATA_MULTI` to 1 and single to 0 to make it really generic (user_data that can be deduced from callback name).  

Do you plan to change all callbacks in that manner? (a quick search shows [`get_bitmap`](https://github.com/littlevgl/lvgl/blob/e8fdac41a96f052e7cb557dd9dce0d24292af123/lv_misc/lv_font.h#L56) and `get_width`, for example, don't have `user_data` members on `lv_font_t`.  
`lv_log_register_print` doesn't even receive a struct as the first parameter)

kisvegabor, 2019-03-14:
> Do you plan to change all callbacks in that manner? (a quick search shows get_bitmap and get_width, for example, don't have user_data members on lv_font_t.

I'm trying to add the most difficult and critical things of v6.0` first. So I'm planning to update the other function too but just a little bit later. Do the missing parts block you in updating the Micropython binding?

amirgon, 2019-03-14:
> Do the missing parts block you in updating the Micropython binding?

@kisvegabor No, it's not blocking me.  
Just wanted to make sure other callbacks will be handled the same way. For example, in case of `lv_log_register_print` I don't have a clear understanding how the same convention can be used, as there is no struct to hold the user_data.

kisvegabor, 2019-03-15:
@amirgon 
I remember that you already explained why you think that a ser_data would be useful for `lv_log_register_print` too but I can't find that conversation. Can you tell it again?

amirgon, 2019-03-15:
@kisvegabor I think that logging callback can be useful on (micro)python. To let the user redirect the logs either to screen, to file, to UART etc. in pure (micro)python.
To create a callback on micropython, we need an associated micropython object which is set when the callback is registered and used when it is called. This micropython object is used when the user register a callback which is an object method or a closure as I explained [here](https://github.com/littlevgl/lvgl/issues/316#issuecomment-466408224).
To make this callback generic we need to follow come naming convention, but this is a bit different because there is no existing struct to hold the user_data.

The callback registration is defined today like this:

```c
void lv_log_register_print(void f(lv_log_level_t, const char *, uint32_t,  const char *));
```

The convention we discussed assumes that the first argument of the callback would be a pointer to **struct** which has a `void *<callback_name>_user_data` member.  
In this case the first argument is a `lv_log_level_t` which is `int8_t`, and not a struct, so this doesn't match our convention and we need to decide how to handle such case.  

One option is to add the `user_data` to the callback and callback registration:

```c
void lv_log_register_print(void f(lv_obj_user_data_t, lv_log_level_t, const char *, uint32_t,  const char *), lv_obj_user_data_t);
```

Another option is to leave it as is, and let the micropython define its user_data internally.  This means that we could have only one global callback associated with `lv_log_register_print`, but this is the case anyway. So the convention would be extended for the case that the first argument is not `void*`.  
  
My opinion is that users may find the user_data useful (to keep track of output file descriptor or whatever) so adding `user_data` to the callback makes sense. However, it can work both ways.

kisvegabor, 2019-03-17:
> To create a callback on micropython, we need an associated micropython object which is set when the callback is registered and used when it is called

Clear now, thank you!

Could it work to use a global `user_data` for logging in Python? We will have only one callback so don't need to store an arbitrary number of `user_data` in this case.

amirgon, 2019-03-17:
> Could it work to use a global `user_data` for logging in Python? We will have only one callback so don't need to store an arbitrary number of `user_data` in this case.

Yes. As I said, we don't have to add the user_data to the callback, if it's global. In such case the binding script could create and use the user_data internally.  
We can only do that in case of a "global" callbacks, where there is **only one** instance of such callback across all lvgl. So this may be valid for this case, but not for driver or object callbacks.

kisvegabor, 2019-03-18:
> Yes. As I said, we don't have to add the user_data to the callback, if it's global. In such case the binding script could create and use the user_data internally.

Then I vote for using the "internal user data" in this case. It seems more reasonable to redirect LOG/WARN/ERROR types to the console/file etc. inside the callback than registering separate callbacks for each of them.

kisvegabor, 2019-03-29:
@rreilink I fixed:
- lv_hal_indev_type_t
- lv_preloader_type_t
- lv_ddlist_get_selected_str (`buf_size` argument added)

Is it not possible or "just" difficult to handle `const char * x[]` variables in the python binding? I'm asking because it's quite convenient in C and I want bo sure it's really worth to change it.

rreilink, 2019-03-30:
Thanks! Using `const char **` variables is surely possible, I had already implemented it before I started some major overhauling to get things aligned with MicroPython. I'll put that code back.

> If you also have time these days, please let me know what topics need to be discussed yet.

I'll try to free up some time in the evenings the coming week :-)

kisvegabor, 2019-03-31:
> Using const char ** variables is surely possible, I had already implemented it before I started some major overhauling to get things aligned with MicroPython. I'll put that code back.

Great! @amirgon what is your view about `const char **`?

> I'll try to free up some time in the evenings the coming week :-)

Awesome! :)

amirgon, 2019-03-31:
@kisvegabor Is the purpose to represent an array of strings? Can we assume `char ** ` will **always** represent an array of strings? (in C, in the general case, `char ** ` could represent other things as well. For example, `char **` passed as a function argument could represent a single string that is accessed indirectly so the called function could set it as a result value)  
If we can assume it always represents an array of strings and nothing else, then we can automatically convert between `char **` and a python array of strings.

embeddedt, 2019-03-31:
> If we can assume it always represents an array of strings and nothing else, then we can automatically convert between `const char **` and a python array of strings.

The problem with assumptions like these in the C language is that the assumptions could later be forgotten.

What about taking advantage of the fact that arrays and pointers are *somewhat* interchangeable and specify the parameter as `const char *[] some_arg` instead? That makes it clear that it is being treated like an array, and allows us to clearly distinguish between the two cases you mentioned. We could add that kind of requirement to the coding standard.

kisvegabor, 2019-04-02:
@embeddedt 
I agree it would be a clear solution (even clearer for the user). But the definition looks like this, doesn't it? `const char * some_arg[]`
Reference: https://www.tutorialspoint.com/cprogramming/c_passing_arrays_to_functions.htm

embeddedt, 2019-04-02:
@kisvegabor Whoops, I was thinking of C# and Java syntax when I typed it. Your definition is correct.

@amirgon What do you think about it?

amirgon, 2019-04-02:
> @amirgon What do you think about it?

What we need is a strict set of rules how to convert between C types and Python types.  
If we agree that `const char * some_arg[]` would **always** represent an array of strings, then we could convert it automatically to python array of strings in the [micro]python bindings.

embeddedt, 2019-04-02:
@amirgon I think that it would be safe to enforce that rule. The syntax makes it obvious as well.

kisvegabor, 2019-04-03:
I've updated `dev-6.0` accordingly: https://github.com/littlevgl/lvgl/commit/1d986e13df0468d4d90cdc0d3fd770b2dcf083d1

amirgon, 2019-04-03:
> I've updated `dev-6.0` accordingly: [1d986e1](https://github.com/littlevgl/lvgl/commit/1d986e13df0468d4d90cdc0d3fd770b2dcf083d1)

@kisvegabor What about return values? For example:
```c
const char ** lv_btnm_get_map(const lv_obj_t * btnm);
```

kisvegabor, 2019-04-04:
@amirgon I find that the same syntax is not supported for return values. 

If really this is the case, probably a naming convention inforthe parameter, like `map_array`, would be more reasonable.

amirgon, 2019-04-04:
> @amirgon I find that the same syntax is not supported for return values.
> 
> If really this is the case, probably a naming convention for the parameter, like `map_array`, would be more reasonable.

A naming convention could help. Without it, the [micro]python user would have to "cast" explicitly between the pointer and the array.

For example:

Without naming convention:
```python
map = lv.btnm.get_map()
arr = lv.cast_to_string_array(map)
```
vs. with naming convention:
```python
arr = lv.btnm.get_map()
```

`cast_to_string_array` does not exist today, this is just an example of how this can be solved without naming convention.

If we decide to use naming convention, how would we define it for return values?
We could, for example, require the usage of a `typedef`:

```c
const char lv_string_array**;
lv_string_array lv_btnm_get_map(const lv_obj_t * btnm);
```

kisvegabor, 2019-04-05:
> If we decide to use naming convention, how would we define it for return values?

It can work. An other solution could be to add the hint to function name: `lv_btnm_set/get_map_array`.

Both are fine for me. We can implement which you prefer more.

rreilink, 2019-04-05:
What do you think of the following two solutions to differentiate between `char *` arrays as input argument vs. indirect access to a `char*`

1. use the `_out` suffix to denote return values passed via pointers:
```c 
lv_btnm_set_map(lv_obj_t *obj, char ** map)
char ** lv_btnm_get_map(lv_obj_t *obj)
void some_other_function(char ** arg_out)
```
If `some_other_function` is were store a `char*` in arg_out

2. use `const` to differentiate between a `char*[]` input argument and a `char*` output argument:
```c 
lv_btnm_set_map(lv_obj_t *obj, char * const * map)
char * const * lv_btnm_get_map(lv_obj_t *obj)
void some_other_function(char ** arg)
```

amirgon, 2019-04-05:
> 1. use the _out suffix to denote return values passed via pointers:

If this is the default, we might not need to use any special prefix at all. If you see `char **` - it means "out arg" and nothing else.

> 2. use `const` to differentiate between a `char*[]` input argument and a `char*` output argument

I think it feels more intuitive to use C array notation (`char*[]`) than const (`char * const *`).  
We really want to represent an array here, not a constant value, so the usage of `const` to represnt an array is  confusing.  

So I prefer this:
```c
lv_btnm_set_map(lv_obj_t *obj, const char * map[])
```

over this:
```c
lv_btnm_set_map(lv_obj_t *obj, const char * const * map)
```

kisvegabor, 2019-04-08:
The probéem I see with `_out` is we can't use it for return values in `get` functions i.e. still there is no way to know if the return value is an array or pointer.

I feel `char * const * map` difficult to read. Something more descriptive would be better. But it has the advantage to can be used in return values too.

What about my earlier suggestion: `lv_btnm_set/get_map_array`?

rreilink, 2019-04-09:
> What about my earlier suggestion: `lv_btnm_set/get_map_array`?

Sounds to me like this is the best solution of the suggestions that we have discussed.

amirgon, 2019-04-09:
> > Sounds to me like this is the best solution of the suggestions that we have discussed.
> 
> Great! If @amirgon also agrees, I'll add it.

Sure. It's good for me.

kisvegabor, 2019-04-10:
Done. I also update
 `void lv_chart_set_points_array(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t * y_array);`

It might be tricky because it has 3 parameters.

amirgon, 2019-04-10:
> Done. I also update
> `void lv_chart_set_points_array(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t * y_array);`
> 
> It might be tricky because it has 3 parameters.

@kisvegabor That's the solution for array return value. But what about array arguments?  
(`lv_chart_series_t[] ser, lv_coord_t[] y_array`)

kisvegabor, 2019-04-10:
I meant `..._array` can be used for both `set` and `get` functions. However, it still possible to denote array arguments with `[]`
E.g.:
```c
void lv_chart_set_points_array(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t y_array[]);
void lv_btnm_set_map_array(const lv_obj_t * btnm, const char map[]);
void lv_btnm_set_ctrl_map_array(const lv_obj_t * btnm, const lv_btnm_ctrl_t ctrl_map[]);
```

amirgon, 2019-04-10:
> I meant `..._array` can be used for both `set` and `get` functions. However, it still possible to denote array arguments with `[]`

@kisvegabor I think it's easier and clearer to specify that a function argument is an array **in the argument itself** (using `[]`, or naming convention) than in the function name (for "set" functions).  
When you specify this in the function name, there is no "automatic" way to know which parameter is the array, in case there are more than one, like in `lv_chart_set_points_array`.  

For "get" functions, since a function has only one return value, specifying that the return value is an array on the function name makes more sense.

kisvegabor, 2019-04-10:
@amirgon 
That makes sense. Updated accordingly. See 3d4c784

kisvegabor, 2019-04-14:
@rreilink Is there anything else to improve?

rreilink, 2019-04-17:
@kisvegabor I am currently looking in how to manage data that is passed to LittlevGL by reference. This will require some way to manage when those references are used by LittlevGL and when they are released. I am still figuring out how to do this exactly, what I am currently thinking of is the following:

Everywhere where a reference that is passed as function argument is retained (stored) by LittlevGL, this is marked.
`ext->styles[xxx] = style;` --> `LV_REF_REPLACE(ext->styles[xxx], style);`

When not using Python, this macro would just expand to the original assignment. When compiling for Python, this would be something like:
```
do {
  void *tmp = ext->styles[LV_BTN_STATE_REL]
  pylv_incref(style);
  ext->styles[LV_BTN_STATE_REL]=style;
  pylv_decref(tmp);
} while (0);
```
(The temporary pointer is required since the old object should only be freed after the new one has been assigned, since the decref could trigger code that requires a valid object in `ext->styles`)

`pylv_incref` and `pylv_decref` would be support functions that ensure that the underlying data will not be freed while it is used.

Furthermore, there needs to be some additional cleanup when an lvgl object is deleted, it should call something like `LV_REF_RELEASE(ext->styles[xxx]);`

As said, this still does require some figuring out on how it would work in total, so I'll try some stuff out first.

rreilink, 2019-04-17:
> Are you interested in hosting the Python binding in LittlevGL organization? If so, I can create an `lv_binding_python` and give Write or even Admin access .

That might help in keeping the binding-specific discussions, issues etc centralized, so I would say yes.

kisvegabor, 2019-04-20:
I see the problem.

I have to concerns:
1. It's easy to miss an assignment.
2. It's difficult to read. 

How does it work in the current Python binding?

amirgon, 2019-04-20:
> How does it work in the current Python binding?

In micropython it's solved by the fact that we use Garbage Collection. Micropython does not require or support reference counting, in order to save RAM and simplify things. This actually makes porting libraries to micropython much easier, if they can be adjusted to support gc (like we did for lvgl!)  

I know that Python [also has gc](https://rushter.com/blog/python-garbage-collector/) that can be enabled optionally. I wonder if it could be used **instead** of reference counting, making lvgl support much simpler.

rreilink, 2019-04-21:
The current (non-micro) Python binding just leaks references, i.e. if a structure is passed to lvgl once, it will never be freed. This makes sure nothing will ever crash, but is not the desired behavior I would say.

Unfortunately, when creating C extensions in Python, one has to use reference counting. The gc is only used to break reference cycles (i.e. object a referencing b and b referencing a), but reference counting still has to be used.

I share the concern that it is easy to miss an assignment. Therefore, I created a Python script that parses the lvgl source code and looks for assignments that store pointer arguments to a function. It also handles the case when pointer arguments are passed on to a function that is called, which in turn stores the pointer. The result of that is [found here](https://pastebin.com/1RUvpUnu).

As shown, the majority of the cases is related to all `xxx_set_style` functions. Only those lines that do an actual assignment (e.g. `ext->styles_btn[LV_BTN_STATE_REL]`) would need to be marked; calls to `yyy_set_style(...)` would disappear from this list once `yyy_set_style` is properly marked.

kisvegabor, 2019-04-22:
I thought there are more reference assignments. Looking at the list it seems doable to add the macros as you suggested.

Can you narrow the list to only to the assignments which really needs to be marked?

rreilink, 2019-04-22:
I will add the macros in my fork; this enables me to check that it is indeed correct. Will keep you posted.

rreilink, 2019-04-22:
See [this commit](https://github.com/rreilink/lvgl/commit/a90739a7e0177bbd5eee9ed7416799c8c9443464)

for what I think would be required to make this work. Would this be acceptable? Obviously, the used macro's need to be defined somewhere, what would be the best place to put a default definition for them (that does nothing but to create the original behavior)?

embeddedt, 2019-04-22:
@amirgon @canardos @kisvegabor 

It's not at all @rreilink's fault (I think it's just the way Python and reference counting are), but if this macro trickery is necessary to make the Python binding work, I'm not sure if it's a good idea to support this. From what I can tell, even with the script, we could end up allowing a reference assignment into the code tree that isn't guarded in this manner. That would later result in a memory leak.

The MicroPython binding is much simpler... there are a few places where `LV_GC_ROOT` is used, but those are rare, and aside from that, the only other requirement is to use a custom heap allocator (the functionality for which was already present anyway).

What do you think? Like @canardos pointed out [here](https://github.com/littlevgl/lvgl/issues/1036#issuecomment-485242287), this is looking quite complicated, and I'm not sure it's worthwhile. I do understand why having Python as well as MicroPython support would be a bonus, but is it worth the extra awkwardness in the code?

kisvegabor, 2019-04-23:
@rreilink Thank you for the update. Now we can clearly how would it look like.

Unfortunately, It really makes LittlevGL more difficult to read and maintain. So we need to discuss the things mentioned by @embeddedt  and @canardos.

I believe Python binding is important as it could be the core of a drag and drop GUI builder once. But we need to figure out something to leave LittlevGL simple.

As it was already said it's mainly about the styles, would it be possible to have functions to create/release styles? For example (probably it's not the best approach just an example)
```python
style1 = lv.style_create;
style1.init(lv.style_plain);  #Inc. reference count as well
style1.delete();
```

It seems with Python inited styles like this we would solve the majority of the issues.

rreilink, 2019-04-23:
> ... but is it worth the extra awkwardness in the code?

I would say, no, actually. Looking at the changeset, let's just not add this to LittlevGL. I agree with @canardos , we should be reluctant to changing too much to LittlevGL. I made the first version of the Python bindings without any changes to LittlevGL.

What I propose to do in the bindings is the following:

- once a reference is passed to LittlevGL, this is marked, and the refcount is increased
- when it is no longer used, the user *can* call a `style.free()` or the like. If the item was marked as being used by LittlevGL, refcount is decreased and the item is marked as not in use by LittlevGL. When subsequently the last reference in Python goes out of scope, the item is cleaned up.
    - If the user chooses not to call `style.free`, the code will work but the object will stay alive. This is no big deal as long as there are no styles created and set within a loop or similar
    - If the users calls `style.free()` while it is still in use in LittlevGL --> **boom** as would be the case when using C and freeing a style that is still in use

A proposed approach would be to create just the number of styles that are required, and then modifying those styles as opposed to creating a new style for each required appearance change.

embeddedt, 2019-04-23:
@rreilink That idea could work. Considering that styles aren't likely to be created over and over, imposing "manual" garbage collection would be fine.

kisvegabor, 2019-04-24:
It sounds good to me as well!

kisvegabor, 2019-05-22:
@rreilink Are there any news here?

rreilink, 2019-05-24:
Yes, I did not have too much time for a while, but picked it up again.

Upon updating to the latest dev-6.0, I noticed that `lv_obj_get_style` now returns a `const *style_t` instead of previous non-const.

So, in C, you can no longer do e.g. `obj_get_style(ob)->body.radius = 0`. Should I also enforce this behavior in Python? 

Is there a particular reason that styles are now const?

embeddedt, 2019-05-24:
@rreilink See #999. The API was updated because in many cases styles are constant and don't need to change.

kisvegabor, 2019-05-25:
> So, in C, you can no longer do e.g. obj_get_style(ob)->body.radius = 0. Should I also enforce this behavior in Python?

Yes, if possible it should be enforced in Python too to avoid writing into `const` data.

kisvegabor, 2019-06-05:
@rreilink We will announce a feature freeze for `v6.0` in a few days. Is there anything "API breaking" to do here?

rreilink, 2019-06-05:
None that I currently foresee

[1 more reply]

closed 2019-07-23 by stale[bot]