lvgl/lvgl #938: Shape of round buttons not round

harteware07, 2019-03-11:
I designed some round buttons, all the used styles have the  ` .body.radius = LV_RADIUS_CIRCLE;` argument. 
I added a screenshot of the simulator, the behaviour is the same on the hardware. Version is 5.3

[image]

`btnCalDays[i] = lv_btn_create(bg_edit, NULL);
		lv_obj_set_size(btnCalDays[i], 40, 40);
		lv_btn_set_style(btnCalDays[i], LV_BTN_STYLE_REL, &bg_style);
		lv_btn_set_style(btnCalDays[i], LV_BTN_STATE_PR, &indic_style);
		lv_btn_set_style(btnCalDays[i], LV_BTN_STATE_TGL_REL, &indic_style);
		lv_btn_set_style(btnCalDays[i], LV_BTN_STYLE_TGL_PR, &indic_style);
		lv_btn_set_style(btnCalDays[i], LV_BTN_STYLE_INA, &knob_style);
		lv_btn_set_toggle(btnCalDays[i], 1);
		lv_obj_set_free_num(btnCalDays[i], (1<<i));
		lv_btn_set_action(btnCalDays[i], LV_BTN_ACTION_CLICK, gui_scenes_edit_calendar);
`

You can see that the circles are not drawn smoothly but have "stairs" in them. Have you observed this behaviour and is there a solution for it?

Thanks again!

embeddedt, 2019-03-11:
I think that this behavior is not abnormal (and I've seen it in other places), but @kisvegabor knows more about how the antialiasing works than I do.

harteware07, 2019-03-11:
imho it's not the antialiasing, it' the algorithm that calculates the roundness that produces the edges

embeddedt, 2019-03-11:
@harteware07 Try turning off antialiasing and see whether the circle looks round.

kisvegabor, 2019-03-27:
I've played with a little and improved the circle's anti-aliasing. 

It's still not perfect but it's the limitation of the curernt algorithm because due to rounding errors some parts of the circle results in lines:

[image]
[image]

If you want to imrove it, the relevant code is here: https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_draw/lv_draw_rect.c#L1422

Feel free to reopen this issue if you have found better paramters.

closed 2019-03-27 by kisvegabor