lvgl/lvgl #940: Enhance keyboard to allow switching to user defined maps

fovea1959, 2019-03-12:
According to the docs, we can use lv_kb_set_map to switch a keyboard to a user defined map.

In that map, though, I want to define keys that switch to other user-defined maps.

Can we do a an enhancement request to allow some form of table lookup in lb_kb_def_action so that we can override what map gets used for "abc" "ABC" "1#", as well as add new keys that take the keyboard to a different user defined map?

embeddedt, 2019-03-12:
It seems like a reasonable improvement to me. @kisvegabor, what do you think?

kisvegabor, 2019-03-13:
It sounds like a reasonable update. We can create a simple struct like:
```c
typedef struct {
  const char * id;   //E.g. "abc" or "ABC"
  const char ** map;
  const lv_btnm_ctrl_t *ctrl_bits;
}lv_kb_map_t;
```
And add `lv_kb_map_t maps[LV_KB_MAX_MAP_NUM]` to `lv_kb_ext_t`.

In `dev-6.0` the control bits of the buttons are moved to separate array to increase flexibility.

fovea1959, 2019-03-13:
An alternative could be provide an optional callback where app code could be passed the key and return the map to, returning NULL if the default is desired.

kisvegabor, 2019-03-14:
It also seems possible, however, for `v6.0` we are trying to make callback as generic as possible and remove object specific callback.

We are working on Python and Micropython bindings and I'd like to know which way would be prefered in bindings.

@amirgon @rreilink can you comment on this?

amirgon, 2019-03-14:
> It also seems possible, however, for `v6.0` we are trying to make callback as generic as possible and remove object specific callback.

@kisvegabor Why not add a callback in a generic way, with a canonical `user_data`? What is the difference in this case?  
I don't think there is an issue if some objects have their own callbacks, as long as the script can automatically find and use the associated `user_data` in order to save (micro)python-specific internal data.

kisvegabor, 2019-03-15:
@amirgon Thank you!

@fovea1959 
In `dev-6.0` [this function](https://github.com/littlevgl/lvgl/blob/dev-6.0/lv_objx/lv_kb.c#L410) is called directly when a button is pressed or long pressed. There can be a callback in `lv_kb_ext_t` which default values is this function. 
Can you sand a PR?

kisvegabor, 2019-03-26:
I found a good way to implement it. I added `lv_kb_def_event_cb` which is the default event handler of the keyboard. It adds the characters to the Text Area and changes the map.
1. You can overwrite the deafut `event_cb` by `lv_obj_set_event_cb(kb, my_event_cb)`
2. If you need an `event_cb` for Keyboard for any other purposes you can just call `lv_kb_def_event_cb` in your `event_cb` to let the Keyboard to handle the button presses in the default way.

I also updated the `demo` application to work this way.

If its not working or have some remarks, feel free to reopen this issue.

closed 2019-03-26 by kisvegabor