lvgl/lvgl #941: BATTERY_SYMBOL_EMPTY macro not expanded properly

rreilink, 2019-03-13:
When including both `Python.h` (Python3.6) and `lvgl.h`, the symbol for BATTERY_SYMBOL_EMPTY (and only that particular symbol) does not expand properly.

I made this minimal example:
```c
#include "Python.h"
#include "lvgl/lvgl.h"

SYMBOL_BATTERY_EMPTY
_SYMBOL_VALUE3(EF,A0,B0)
_SYMBOL_VALUE3(EF,A0,b0)
_SYMBOL_CHAR(EF)_SYMBOL_CHAR(A0)_SYMBOL_CHAR(B0)
_SYMBOL_STR(\xEF\xA0\xB0)
_SYMBOL_STR_(\xEF\xA0\xB0)
```

`gcc -E -I/Users/rob/miniconda3/include/python3.6m -c test3.c`

```c
...
# 3 "test3.c" 2

"\xEF\xA0\x0"
"\xEF\xA0\x0"
"\xEF\xA0\xb0"
\xEF\xA0\xB0
"\xEF\xA0\xB0"
"\xEF\xA0\xB0"
```

Note how the  `_SYMBOL_VALUE3` macro using upper case `B0` incorrectly expands to `\x0` whereas the lower case `b0` correctly expands to `\xb0`.

rreilink, 2019-03-13:
cc @kisvegabor  @amirgon

amirgon, 2019-03-13:
Probably Python.h defined `B0` to 0 somewhere.
If this is true, that's called global namespace hijacking and is a serious problem on the Python side. If they define such general macros without prefixing them properly, be prepared for more trouble in the future.  

If this is really the problem, you can `#define B0 B0` after including Python.h as a workaround.

rreilink, 2019-03-13:
> Probably Python.h defined `B0` to 0 somewhere.

I checked, B0 is indeed defined.

> be prepared for more trouble in the future.

Well... yes, but we all should. Since that B0 is defined in [sys/termios.h](https://github.com/lattera/glibc/blob/895ef79e04a953cac1493863bcae29ad85657ee1/bits/termios.h#L288)

Seriously, this is exactly why I think this solution with macro trickery in the lvgl source was a bad idea, even more so since it also affects those users that don't use (Micro)Python.

I do have another suggestion, though, that might serve the purpose equally and not require custom parsing. All the MicroPython binding needs to know are the *names* of the symbols, not so much the symbols themselves (since you can do MP_DEFINE_STR_OBJ(mp_LV_SYMBOL_CLOSE,LV_SYMBOL_CLOSE);

So what if we just add:

`static const char _all_lv_symbols="SYMBOL_AUDIO SYMBOL_VIDEO SYMBOL_LIST" ...;`

As it is not used, that adds no RAM or ROM size (as far as I know). It survives preprocessing. Sure, it requires having the same list of symbols twice, which is a risk for forgetting to add something, but that is as it is with the current solution as well.

That is, assuming that the consensus is still that duplicating all symbols outweighs [these 7 lines of Python code that do the trick](https://github.com/rreilink/pylvgl/blob/96e2eeb6ef7741942b4128b942d4760eb1604fc2/sourceparser.py#L135). I'd be more than happy to write a PR to include that into `gen_mpy` (and to show that it does not alter the generated code, if that would be a requirement).

amirgon, 2019-03-14:
> Well... yes, but we all should. Since that B0 is defined in [sys/termios.h](https://github.com/lattera/glibc/blob/895ef79e04a953cac1493863bcae29ad85657ee1/bits/termios.h#L288)

`sys/termios.h` header is used for terminal I/O interfaces. I don't think it is supposed to be included on public library headers or on `python.h`. I would expect it to be included only on **C files** whose purpose is to handle terminal IO.  
On that context, it makes sense to define these macros. **On global context, it does not**.

> So what if we just add:
> 
> `static const char _all_lv_symbols="SYMBOL_AUDIO SYMBOL_VIDEO SYMBOL_LIST" ...;`

Wouldn't an enum be much cleaner?

```c
enum {
    SYMBOL_AUDIO,
    SYMBOL_VIDEO,
    SYMBOL_LIST,
    ...
} _all_lv_symbols;
```

> That is, assuming that the consensus is still that duplicating all symbols outweighs [these 7 lines of Python code that do the trick](https://github.com/rreilink/pylvgl/blob/96e2eeb6ef7741942b4128b942d4760eb1604fc2/sourceparser.py#L135).

It's not a matter of the number of lines it takes. It's not the work you do now that counts. It's the property of Forward Compatibility and making the code Generic, as I explained [here](https://github.com/littlevgl/lvgl/issues/935#issuecomment-472193278).

rreilink, 2019-03-14:
> Wouldn't an enum be much cleaner?

Either would work for me. I thought of it too but would choose the char* approach since it adds less identifiers to the global namespace.

amirgon, 2019-03-14:
> Either would work for me. I thought of it too but would choose the char* approach since it adds less identifiers to the global namespace.

I don't think we should worry too much about adding identifiers to the global namespace, as long as they are prefixed correctly.  
However, some compiler *might* not optimize out an unused const char string, or on the other hand might give an annoying warning about unused const char global variable.  
This will not happen with an enum.

kisvegabor, 2019-03-15:
I've just one question: is it really enough to know the name of the SYMBOLs? This way `label.set_text(SYMBOL_OK + "Apply")` will work?

rreilink, 2019-03-15:
> is it really enough to know the name of the SYMBOLs?

Yes it is, for the bindings generator. For example, for SYMBOL_AUDIO, it could generate:
```c
PyModule_AddString(module, "SYMBOL_AUDIO", SYMBOL_AUDIO);
```
So the name appears in the generated code both as a string literal, and as a macro. The latter expands to the value at compile time of the binding. Obviously, the resulting compiled code will then contain both the name and the expanded value, allowing the use as you mentioned.

This approach has the added benefit that the definition of SYMBOL_AUDIO can contain any arithmetic, further macros etc... and the bindings generator does not need to care about it. This is also why my opinion is that it would be no problem to do custom parsing: the parser would not need to expand the macros, only to detect which macros exist.

As a side note, Python has a macro:
```c
PyModule_AddStringMacro(module, SYMBOL_AUDIO);
```
That expands to the code above. I am using that now in pylvgl, and it works.

kisvegabor, 2019-03-15:
@rreilink Thank you for the explanation. It's much clearer now.

Then @amirgon 's suggestion to list the symbols in an enum can be the compromise solution. What about `_LV_STR_SYMBOL_AUDIO` names. The scripts can convert all `_LV_STR_XYZ` to `"XYZ"`.

kisvegabor, 2019-03-30:
@amirgon  @rreilink 
If you agree on using enums can you send a PR with this update? I'm afraid of doing it myself because you know much better how it should look like.

rreilink, 2019-03-30:
If @amirgon agrees, I can send a PR

amirgon, 2019-03-30:
@rreilink Let's try the enum approach. `_LV_STR_` prefix as @kisvegabor suggested.

kisvegabor, 2019-04-09:
The PR is merged so I close it!

Thank you!