lvgl/lvgl #945: Notification from text area when the Enter key is hit – 6.0 version

manison, 2019-03-15:
I'm trying to rewrite the solution from #754 from signal based to the new event based interface introduced in 6.0 version, but I'm hitting some issues.

My original signal handler looks like this:

```c
static lv_res_t TextArea_OnSignal(lv_obj_t *obj, lv_signal_t sign, void *param)
{
    if (sign == LV_SIGNAL_CONTROLL) {
        uint32_t c = *((uint32_t *)param);
        if (c == '\n' || c == '\r')
            TextArea_OnEnterKeyPress(obj);
    }
    return TextAreaOriginalSignalFunc(obj, sign, param);
}
```

Now the first change in 6.0 is that Enter key no longer generates `LV_SIGNAL_CONTROLL` signal but `LV_SIGNAL_PRESSED` is raised instead. `LV_SIGNAL_PRESSED` can be also generated by encoder or pointer devices but I only need to respond to keypad devices. I tried to take advantage of the new event based mechanism and thought that the event handler below does the trick:

```c
lv_res_t TextArea_OnEvent(lv_obj_t *obj, lv_event_t e)
{
    lv_indev_t *indev;
    if (e == LV_EVENT_PRESSED &&
       (indev = lv_indev_get_act()) != NULL &&
       lv_indev_get_key(indev) == LV_GROUP_KEY_ENTER)
    {
        TextArea_OnEnterKeyPress(obj);
    }
    return LV_RES_OK;
}
```

The problem with this is that `lv_indev_get_key` returns the **previously** pressed key instead the actual key that generated the event.

The comment for `lv_indev_get_key` really says "*return the last pressed key (0 on error)*" so this is probably intentional behavior. But I can't imagine this behavior could be useful. IMHO it should be changed to 
* returns the actual key being pressed when called in the context of LV_EVENT_PRESSED event handler
* returns the last pressed key if called outside the event handler
* returns 0 on error

What do you think?

kisvegabor, 2019-03-17:
Independently I've also realized that sending ENTER in `LV_SIGNAL_CONTROLL` is still useful. It's already updated in `dev-6.0`.

I've also fixed `lv_indev_get_key` in `dev_6.0`.

manison, 2019-03-18:
> I've also fixed lv_indev_get_key in dev_6.0.

Seems to work as suggested, thanks.

At the end I realized I cannot use events in this case. I need to eat the Enter key and prevent it to reach the object. In event it is too late, the `\n` char is already added to the text area. So it seems I'm stuck with the signals.

OT: I always wondered why the signal is called LV_SIGNAL_CONTROL**L** Does the second _L_ mean anything or is it a typo?

kisvegabor, 2019-03-19:
> Seems to work as suggested, thanks.

Great!

> At the end I realized I cannot use events in this case. I need to eat the Enter key and prevent it to reach the object. In event it is too late, the \n char is already added to the text area. So it seems I'm stuck with the signals.

I updated `LV_EVENT_VALUE_CHANGED` to tell the inserted char/text of Text Area as the `data` of the event (`lv_event_get_data()`). Once you see that that ENTER is inserted you can delete it and do you what you need.

> OT: I always wondered why the signal is called LV_SIGNAL_CONTROLL Does the second L mean anything or is it a typo?

In Hungarian (my native language) it's "kontroll" and I thought it's similar in English too. :sweat_smile: 
I've fixed it.

manison, 2019-03-19:
> I updated LV_EVENT_VALUE_CHANGED to tell the inserted char/text of Text Area as the data of the event (lv_event_get_data()). Once you see that that ENTER is inserted you can delete it and do you what you need.

I would like to prevent the operation altogether rather than trying to undo it. So for now I am probably staying with the signal based approach, where one have the opportunity to pass the signal to the original signal handler or not. Moreover deleting the char (`lv_ta_del_char`) would also send `LV_EVENT_VALUE_CHANGED` which spins the opportunity for potential infinite loop.

There is one possible solution for the event based approach though. We could use the event data to communicate back the result from the handler and modify the data in-place:

```c
lv_res_t ta_event_handler(lv_obj_t *obj, lv_event_t event)
{
  …
  char *data = lv_event_get_data();
  if (*data == '\n' || *data == '\r') *data = 0;
  …
}
```

> In Hungarian (my native language) it's "kontroll" and I thought it's similar in English too. I've fixed it.

Oh, it's the "Hungarian notation" 😃

kisvegabor, 2019-03-19:
> There is one possible solution for the event based approach though. We could use the event data to communicate back the result from the handler and modify the data in-place:

As data can be `const` you can't assign a value directly. However, `LV_EVENT_INSERT` can be added. It's `data` is the string which should be added. But you can overwrite it with a function like `lv_ta_set_insert_replace(ta, "new text")`. If you use `""` no text will be added. You can also format the text area this way. For example:
- if you need an IP address  you can insert an extra `.` where required. 
- Make all letter uppercase
- Handle complex formats like a license plate

I checked it quickly and it seems easy to implement. The only drawback would be `LV_EVENT_INSERT` will be sent with modified text too. What do you think?

manison, 2019-03-20:
> As data can be const you can't assign a value directly.

It does not have to be `const` 😃

> However, LV_EVENT_INSERT can be added. It's data is the string which should be added. But you can overwrite it with a function like lv_ta_set_insert_replace(ta, "new text"). If you use "" no text will be added.
…
I checked it quickly and it seems easy to implement. The only drawback would be LV_EVENT_INSERT will be sent with modified text too. What do you think?

Interesting idea, I quite like it. It could probably solve another problem related to events I am right now facing. When you implement this feature I'm happy to test.

kisvegabor, 2019-03-20:
> It does not have to be const smiley

But it could be :smile: Especially in the case of a Text area with `lv_ta_add_text(ta, "abc")`

> When you implement this feature I'm happy to test.

Okay, thanks! I'll let you know.

kisvegabor, 2019-03-20:
It was quicker than I thought. I pushed the updates to `dev-6.0`.

I tested it like this to make all letter uppecase
```c
        case LV_EVENT_INSERT:
            {
                const char * ins = lv_event_get_data();
                if(ins) {
                    static char buf[32];
                    strcpy(buf, ins);
                    uint8_t i;
                    for (i = 0; buf[i] != '\0'; i++) {
                        if(buf[i] >= 'a' && buf[i] <= 'z') buf[i] -= 'a' - 'A';
                    }
                    lv_ta_set_insert_replace(obj, buf);
                }
            }
            break;
```

manison, 2019-03-21:
> I pushed the updates to dev-6.0.

It was really quick. I test it and it works well, thank you.

Unfortunately the other issue I mentioned is not solved :(
The problem is with the event flow:
1. _Enter_ key press raises the new `LV_EVENT_INSERT` event
2. I catch it in the _ta_ event handler and display confirmation _mbox_
3. Release of the _Enter_ key generates the `LV_EVENT_SELECTED` event
4. This event closes the _mbox_

The result is that the _mbox_ is immediately closed. I don't know how to solve this.

kisvegabor, 2019-03-22:
> It was really quick. I test it and it works well, thank you.

Great! :)

> Unfortunately the other issue I mentioned is not solved :(
For POINTER indevs there is the `lv_indev_wait_release` function which should be used in this case too. 

I updated `dev-6.0` to make support `lv_indev_wait_release()` wotk with KEYPAD and ENCODER too.

I testis it like
```c
     case LV_EVENT_XXX:
            lv_indev_wait_release(lv_indev_get_act());
            break;

```

manison, 2019-03-22:
> I updated dev-6.0 to make support lv_indev_wait_release() wotk with KEYPAD and ENCODER too.

It works, thanks.

But I also had to fix the error #971.

closed 2019-03-26 by manison

kisvegabor, 2019-03-26:
Great! Thank you for the testing.
Good to know that now `dev-6.0` is tested with KEYPAD too.

manison, 2019-03-26:
> Good to know that now dev-6.0 is tested with KEYPAD too.

Yes, our application is to be controlled by keyboard mainly, so I'm testing it with lvgl configured with keypad input drivers ⌨️