lvgl/lvgl #946: Bug when deleting object from within nested event handler

manison, 2019-03-15:
When deleting and creating objects from within nested events, the `obj_act_event_deleted` flag gets rewritten and further events in chain touches deleted object which results in crash.

This line in `lv_obj_send_event` is responsible that the flag is lost in event chain:
https://github.com/littlevgl/lvgl/blob/2819b30716e4bb76f36a90abd9958933966f072e/lv_core/lv_obj.c#L1186

Minimum test case (modified lv_test_group):

```c
static lv_obj_t *prev_focused_obj;
static lv_group_t *group;

static lv_res_t mbox_event(lv_obj_t *obj, lv_event_t e)
{
    if (e != LV_EVENT_CLICKED)
    {
        return LV_RES_OK;
    }

    lv_group_focus_freeze(group, false);
    lv_obj_t *blanket = lv_obj_get_parent(obj);
    lv_obj_del(blanket);

    lv_group_focus_obj(prev_focused_obj);

    lv_obj_t *o = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_text(o, "Label");

    return LV_RES_INV;
}

void test_crash(lv_group_t *g)
{
    lv_obj_t *scr = lv_scr_act();
    lv_obj_t *blanket = lv_obj_create(scr, NULL);
    lv_obj_set_size(blanket, lv_obj_get_width(scr), lv_obj_get_height(scr));
    lv_obj_t *mbox = lv_mbox_create(blanket , NULL);
    lv_obj_align(mbox, blanket, LV_ALIGN_CENTER, 0, 0);
    lv_mbox_set_text(mbox, "Test");
    static const char *map[] = { "Yes", "No", "" };
    lv_mbox_add_btns(mbox, map);
    prev_focused_obj = lv_group_get_focused(g);
    lv_obj_t *btnm = lv_mbox_get_btnm(mbox);
    lv_group_add_obj(g, btnm);
    lv_group_focus_obj(btnm);
    lv_group_focus_freeze(g, true);
    lv_obj_set_event_cb(mbox, mbox_event);
    group = g;
}
```

embeddedt, 2019-03-15:
I'm not familiar with the new event system yet, so I think @kisvegabor should handle this if he doesn't mind.

kisvegabor, 2019-03-15:
@manison 
You have pointed out a very important issue. I've fixed it in `dev-6.0`. 
`padding.top/bottom/left/right` is also added to `dev-6.0` to replace `padding.hor/ver`. The `lv_examples` repo is updated accordingly.

manison, 2019-03-18:
@kisvegabor I'm testing the latest version at the moment (abfdfd1b20340ae75e0e6929ea085880eec7c304) and unfortunately the test code posted above still crashes.

Here's the backtrace:

>	lvgl_sim.exe!lv_obj_send_event(_lv_obj_t * obj, lv_event_t event) Line 1190	C
 	lvgl_sim.exe!lv_obj_send_event(_lv_obj_t * obj, lv_event_t event) Line 1201	C
 	lvgl_sim.exe!indev_keypad_proc(_lv_indev_t * i, lv_indev_data_t * data) Line 492	C
 	lvgl_sim.exe!indev_proc_task(void * param) Line 329	C
 	lvgl_sim.exe!lv_task_exec(lv_task_t * lv_task_p) Line 322	C
 	lvgl_sim.exe!lv_task_handler() Line 125	C
 	lvgl_sim.exe!WinMain(...) Line 100	C
 	…

kisvegabor, 2019-03-19:
@manison Finally, I found the issue. Should work now.

commit 4d8ec838 (2019-03-19): fix(event): in lv_obj_del get the focused obj before removing the obj from the group

fix #946
closed 2019-03-19 by manison