lvgl/lvgl #949: Crash in lv_btnm_signal

embeddedt, 2019-03-16:
I'm verifying that the Windows drivers are working properly under `v6.0`, and I encountered this crash immediately after I click the screen using `lv_test_theme_1`. 

This is a tricky one; it doesn't occur every time I run the program; only sometimes. However the crash behavior is consistent. Here's the backtrace:

```
#0  0x0042b32e in lv_area_copy (dest=0x28fcd4, src=0x76139c) at lvgl/lv_misc/lv_area.h:67
#1  0x0042d493 in invalidate_button_area (btnm=0x6e10c4, btn_idx=65535) at lvgl\lv_objx\lv_btnm.c:1098
#2  0x0042cac8 in lv_btnm_signal (btnm=0x6e10c4, sign=8 '\b', param=0x6e0f64) at lvgl\lv_objx\lv_btnm.c:831
#3  0x00413bf4 in indev_proc_press (proc=0x6e0f74) at lvgl\lv_core\lv_indev.c:739
#4  0x00412de9 in indev_pointer_proc (i=0x6e0f64, data=0x28fde8) at lvgl\lv_core\lv_indev.c:366
#5  0x00412c9f in indev_proc_task (param=0x0) at lvgl\lv_core\lv_indev.c:327
#6  0x00428023 in lv_task_exec (lv_task_p=0x6e2afc) at lvgl\lv_misc\lv_task.c:319
#7  0x00427bab in lv_task_handler () at lvgl\lv_misc\lv_task.c:125
```

embeddedt, 2019-03-16:
@kisvegabor

kisvegabor, 2019-03-17:
I've fixed it, can you try it?

embeddedt, 2019-03-17:
@kisvegabor It's working on Windows. Thanks!

I had a similar issue on STM32 so I'll check that platform and reopen this issue if it's still not working.

closed 2019-03-17 by embeddedt

kisvegabor, 2019-03-17:
Great, thank you!

embeddedt, 2019-03-18:
It works properly on STM32 as well. Thanks again!