lvgl/lvgl #950: [suggestion] lv_tick_inc(x) - provide support for absolute values

puzrin, 2019-03-17:
**Problem**

Current signature of `lv_tick_inc(x)` cause additive errors. With all appropriate limitations.

**Suggestion**

Provide `lv_tick(x_absolute_ms)`

Special suggestion for [6.0] - just drop relative support to hell :).

---

Why is this convenient:

- Can call `lv_tick()` with any frequency, without delay at all.
- (?) Can provide internal timer API-s for drivers/extentions.
- (?) May be later can join with `lv_task_handler()` for simplified use (via internal micro-scheduler).

embeddedt, 2019-03-17:
@puzrin What is the problem with the existing `lv_tick_inc`?

>  Can call `lv_tick()` with any frequency, without delay at all.

I think that `lv_tick_inc(0)` will do nothing, and you can already call it with whatever frequency you want.

I'm sure that I haven't understood something, because I don't understand exactly what should be changed.

(cc @kisvegabor)

puzrin, 2019-03-17:
> What is the problem with the existing `lv_tick_inc`?

It can not be used as exact time source in many cases. At every call, when you do:

```c
while (1) {
  delay(1);
  lv_tick_inc(1);
}
```

real delay is `1 + execution time`. For 10000 calls, error is `10000 * execution time` and so on.

All system have something like [SDL_GetTicks()](https://wiki.libsdl.org/SDL_GetTicks?highlight=%28%5CbCategoryTimer%5Cb%29%7C%28CategoryEnum%29%7C%28CategoryStruc%29). When you pass absolute value, local errors are not summarized.

kisvegabor, 2019-03-17:
On embedded systems usually you call `lv_tick_inc` form a Timer interrupt. This way you wont't have an additive error (or at least minimal because of long-lasting higher priority interrupts)

If you already have a "tick_get " function you can config LittlevGL to use it here:   
https://github.com/littlevgl/lvgl/blob/master/lv_conf_templ.h#L147

This way you can use the tick interface in either relative or absolute mode.

embeddedt, 2019-03-17:
The tick is used to keep track of things like animations, not to drive real-time hardware. My argument is that it doesn't need to have perfect precision - I doubt the very minimal execution time is going to add up very quickly. In the event that it does, what effect would it really have on the user?

puzrin, 2019-03-17:
> On embedded systems usually you call `lv_tick_inc` form a Timer interrupt. This way you wont't have an additive error (or at least minimal because of long-lasting higher priority interrupts)

This can work, but have additional requirements on usage way (interrupts only).

Then second question is - why should user care about `lv_task_handler()` calls, if `lv_tick_inc()` already exists?

> In the event that it does, what effect would it really have on the user?

User will not be affected directly. Indirectly - may be we eliminate need of manual `lv_task_handler()` call after some discuss.

kisvegabor, 2019-03-18:
> Then second question is - why should user care about lv_task_handler() calls, if lv_tick_inc() already exists?

LittlevGL needs to know the current time with good accuracy. E.g to know that: it's a 200 ms long animation and we are at 120 ms, so the current value is ... All the animation, rendering, input device handling happen in `lv_task_handler` which can last even for 100 ms. Therefore `lv_tick_inc()`  should be called in an interrupt to "count" the time independently.

puzrin, 2019-03-18:
Why not call `lv_task_handler()` automatically from `lv_tick_inc()`. Looks like now user have to manually reinvent simplified scheduler.

Is it a problem, if nested interrupt happens when previous is not complete?

puzrin, 2019-03-18:
As far, as i understand the root of problem, there are 2 "conflicting" things:

1. some heavy calculations may happens (i mean, NOT wait loops)
2. time should be tracked properly

Solution => move critical part of time tracking out of scope. Use `lv_tick(absulite_ticks)` as i suggested.

If users make "wait loops" in drivers (porting) - explain they should not do so (use states + events approach instead).

embeddedt, 2019-03-18:
We already have a method for getting the elapsed number of milliseconds in `lv_conf.h`:

https://github.com/littlevgl/lvgl/blob/98d43daa5d580f241856b88f0286dac784dbeeb9/lv_conf_templ.h#L144-L148

puzrin, 2019-03-18:
Yes. But API promotes relative values use, and that prevents joining 2 methods into one. I'd say, current API promotes "poor man mutitasking use". Because combination of [ timer interrupt, main loop ] is simplified equivalent of 2 tasks.

Having 2 tasks support is not mandatory requirement for lvgl work. Fix requires switch to absolute ticks. I think, difference is not principle, but allows more flexible use.

embeddedt, 2019-03-18:
@puzrin I don't think I understand. If you configure `LV_TICK_CUSTOM` in `lv_conf.h`, you no longer need to call `lv_tick_inc`, which therefore means you only need 1 task.

> Fix requires switch to absolute ticks.

Isn't `LV_TICK_CUSTOM` using absolute ticks?

puzrin, 2019-03-18:
Goal is to not switch to absolute ticks "just for fun", but to get something useful. For example - more simple integration (1 unified method call, instead 2 methods).

Currently, i can switch to absolute ticks, but that will not give anything useful.

puzrin, 2019-03-18:
Imagine, LVGL is black box. It communicating with outer world via several interfaces:

- `lv_tick_inc()` (~ event)
- `lv_task_handler()` (~ event)
- methods in "porting" files.

I just apply formal logic to check if every method is necessary or something can be simplified.

kisvegabor, 2019-03-20:
We really have these 3 interfaces but I think all three is required as you need a way to know the current time because `lv_task_handler` execution time can be long. I can't imagine anything else to know the current time than an independent Timer.

By switching to absolute time, `lv_tick_inc()` is not required anymore so it simplifies the interface.

puzrin, 2019-03-20:
> I think all three is required as you need a way to know the current time because lv_task_handler execution time can be long. I can't imagine anything else to know the current time than an independent Timer.

I suspect, updating timestamp while `lv_task_handler` incomplete is the way to tearing effects.

The only case when you really need more exact timer inside `lv_task_handler` - if it contains internal scheduler to split long things into subtasks.

kisvegabor, 2019-03-20:
> I suspect, updating timestamp while lv_task_handler incomplete is the way to tearing effects.

I can't see why is it better to update a timestamp inside `lv_task_handler` than getting the current time when required. The user still needs to provide a `tick/time_get` function.

puzrin, 2019-03-20:
> I can't see why is it better to update a timestamp inside `lv_task_handler` than getting the current time when required. The user still needs to provide a tick/time_get function.

I mean a bit different thing. Not "time should be updated inside `lv_task_handler`". But "time should not be mandatory updated in `lv_tick_inc()`". Current requirement to update time inside `lv_tick_inc()` is not necessary overcomplication - add restrictions, but not give any benefits.

After that, updating inside `lv_task_handler` is logical, to keep API simple.

kisvegabor, 2019-03-21:
> I mean a bit different thing. Not "time should be updated inside lv_task_handler". But "time should not be mandatory updated in lv_tick_inc()". 

If time is not updated independently then a task inside itself couldn't see the elapsed time. 
For example, the renderer task measures the rendering time of a display. So time needs to be updated meanwhile. 

So not updating the time continuously and independently would introduce restrictions. 

If you don't want to always update `lv_tick_inc` you can add your custom `tick_get` function with `LV_TICK_CUSTOM_SYS_TIME_EXPR` as mentioned above.

puzrin, 2019-03-21:
> For example, the renderer task measures the rendering time of a display. So time needs to be updated meanwhile.

Now understand. Thanks.

closed 2019-03-21 by puzrin