lvgl/lvgl #953: [6.0] Slider is clipped at edges

embeddedt, 2019-03-18:
My guess is that this is related to the recent padding changes. This screenshot is from `lv_test_theme_1` with the material theme, but the problem is also present on another theme, so I don't think it's necessarily theme-specific.

[image]

embeddedt, 2019-03-18:
@kisvegabor

kisvegabor, 2019-03-18:
It was an issue in `lv_test_theme1`. I fixed by adjusting the paddings. 
See: https://github.com/littlevgl/lv_examples/commit/c1d44cc9f199f7a8878d84e54493f6328c66d078

Is it working well there too?

embeddedt, 2019-03-18:
@kisvegabor It's working properly now. Thanks!

closed 2019-03-18 by embeddedt

kisvegabor, 2019-03-18:
Great! Thank you! :)