lvgl/lvgl #954: [6.0] Button matrix toggling behavior has changed

embeddedt, 2019-03-18:
This is with the horizontal button matrix in `lv_test_theme_1`.

In `v5.3`, the button matrix was set up more like radio buttons; i.e. you could toggle individual buttons and only that button would be selected.

Now the toggling seems to be broken - clicking on each button has no effect on what is toggled. It seems to be permanently set:

https://github.com/littlevgl/lv_examples/blob/7c73ee6a6bf9089132a3f18279f5c18d96e5d464/lv_tests/lv_test_theme/lv_test_theme_1.c#L118

My initial guess is that this is related to the changes made in #825.

embeddedt, 2019-03-18:
(cc @kisvegabor)

(off-topic) Does GitHub notify you when I request your review on a pull request?

kisvegabor, 2019-03-19:
I improved `lv_btnm` to enable individual button toggling. In `v5.3` only one button could be toggled at once. I was considering to add a function like `lv_btnm_set_one_toggle` (or so) but didn't do it yet. Would you be interested in adding it?

> Does GitHub notify you when I request your review on a pull request?

Yes, it does. Once we are talking about it: for me, the new "workflow" with the forum and the modified notifications work well. Is it good for you this way as well?

embeddedt, 2019-03-19:
> Would you be interested in adding it?

Sure!

> For me, the new "workflow" with the forum and the modified notifications work well. Is it good for you this way as well?

Yes.

kisvegabor, 2019-03-20:
Both are great! :)

embeddedt, 2019-03-20:
Implemented in #966.

closed 2019-03-20 by embeddedt