lvgl/lvgl #960: Improve triangle drawing in case of right angled triangles

kisvegabor, 2019-03-18:
Reminder issue from #799 

In case of right-angled triangles, only one side needs to be calculated.

kisvegabor, 2019-05-31:
I added the optimization. If the triangle is "tall" then it is drawn from vertical lines, if "flat" it is drawn from horizontal lines to minimize the number of lines.

Unfortunately, with the optimization, there are overdrawn pixels on the common edges of two adjacent triangles. So the optimized drawing is used only if the triangle has no opacity.

If you experience some unexpected behavior please comment here and reopen the issue.

closed 2019-05-31 by kisvegabor

embeddedt, 2019-05-31:
@kisvegabor Can you please share an example that shows where the overdraw occurs?

kisvegabor, 2019-06-02:
@embeddedt 
I created the `tri_err` branch where the triangle drawing is modified to always use the optimized version and it always uses `LV_OPA_50`.

To test it used this code:
```c
    lv_obj_t * chart = lv_chart_create(lv_scr_act(), NULL);
    lv_chart_set_type(chart, LV_CHART_TYPE_AREA);
    lv_obj_set_size(chart, 300, 200);
    lv_chart_series_t * ser = lv_chart_add_series(chart, LV_COLOR_RED);

    while(1) {
        /* Periodically call the lv_task handler.
         * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        usleep(5 * 1000 * 100);

        lv_chart_set_next(chart, ser, rand() % 100);
}
```

The result:
[image]

embeddedt, 2019-06-02:
@kisvegabor I see, thank you.

It looks like it might be an off-by-one error to me. I will investigate it further.

embeddedt, 2019-06-02:
@kisvegabor I did some debugging. The problem is significantly reduced by setting `corr_value` to 1. Unfortunately, that can occasionally cause artifacts like this:

```c
        lv_obj_t * chart = lv_chart_create(lv_scr_act(), NULL);
	lv_chart_set_type(chart, LV_CHART_TYPE_AREA);
	lv_obj_set_size(chart, 300, 200);
	lv_chart_series_t * ser = lv_chart_add_series(chart, LV_COLOR_RED);

	static int series[] = { 50 ,24, 25, 12, 34, 58, 48, 92 };
	for(int i = 0; i < 8; i++) {
		lv_chart_set_next(chart, ser, series[i]);
	}
```

[image]

I'll keep investigating.

kisvegabor, 2019-06-02:
Try to draw the flat and tall triangles with a different color. The artifact is caused when a flat and tall triangle "meets". 

The basic problem is that when you draw two adjacent triangles there will be a common edge. The internal lines, which builds up the triangles, are started from the same "edge line" therefore they have common points.  
The `corr_value` is used to reduce y2 coordinates of the vertical lines of the tall triangles. This way the most bottom pixels of the tall triangles are not drawn hence there won't be common pixels. This way it works well is only tall triangles are used. However, the flat triangles are built up from horizontal lines and it's not easy to reduce their y2 coordinates.