lvgl/lvgl #961: swipe left/right support for images

Masi2017, 2019-03-18:
Hi All,
I have a gallery menu in my application so user can see the stored photos. Does image object support swipe left/right?

embeddedt, 2019-03-18:
@Masi2017 No, I think you would need to extend it's functionality with a signal handler.

@kisvegabor wrote a great post about that: https://blog.littlevgl.com/2018-12-13/extend-lvgl-objects

Masi2017, 2019-03-19:
@embeddedt Thank you I'll have a look.

kisvegabor, 2019-03-19:
@Masi2017 
You can even use `lv_tabview` where swiping and `tab_load_action`s are supported.

---

We started a [Forum](https://forum.littlevgl.com/) to discuss questions there. GitHub issue should be rather about development related topics. See this [announcement](https://github.com/littlevgl/lvgl/issues/906).

If you have further questions about image swiping please open a topic in the Forum and let's continue there.

closed 2019-03-19 by kisvegabor