lvgl/lvgl #962: What is the minimum resolution that this library supports? Can I use this library to support 64*64 LED full-color dot matrix screen

chenyang8094, 2019-03-19:
RT

kisvegabor, 2019-03-19:
Yes, you can. There is no minimum resolution ( > 1px)

---

We use GitHub issue's for development related discussions. Please use the[Forum](https://forum.littlevgl.com) to ask questions.

closed 2019-03-19 by kisvegabor

xudong8, 2024-06-05:
> RT

可以交流学习下吗[image]

kisvegabor, 2024-06-05:
If each byte is mapped to 8 pixels be sure that the bit order is correct.

Also be sure that you set a `rounder_cb` (in v8, example [here](https://docs.lvgl.io/8.4/porting/display.html)) or `LV_EVENT_INVALIDATE_AREA` (in v9, more info [here](https://docs.lvgl.io/9.1/porting/display.html#events)) to be sure that LVGL always redraws an area which start on 8 px boundary.