lvgl/lvgl #963: How to eliminate border and padding of a list?

qianpei69, 2019-03-20:
Hi !

I use page and container to make a name list. which has a little icon-img and string.
It is running on a low density LCD (128x128) and we have big font (16x16 pixels, Chinese). 
Because of stortage of display space, I want to eliminate the border of the background page, but failed.
I tried style the page : lv_page_set_style(page, LV_PAGE_STYLE_SCRL, &style1); with 
    style1.body.border.width = 0;
    style1.body.border.part = LV_BORDER_NONE;
and no effects. It seems style.body.border 's elements does not applied to page. (border.opa either)
If I use LV_PAGE_STYLE_BG or _SB there are paddings between items which I do not want.
My question is: Is there a way to remove all the borders and paddings in a list ? or minimize them (eg. 1 pixel) ?

Thanks.

qianpei69, 2019-03-20:
Hi !

I found out what i missed after examing the documents.
there are 2 container in a 'page', I set style of a page in wrong way.

Thanks.

embeddedt, 2019-03-20:
Glad to hear you figured it out!
----------------------
We use GitHub issues for development related discussions.
In the future, please use the forum to ask questions.

closed 2019-03-20 by embeddedt